EPFL-LAP/dynamatic #79: Wrong result used in integration tests

chatpion, 2024-05-16:
Hi,

The `gcd` example of the integration tests seems to expect wrong values.

It is supposed to compute the GCD of 7966496 and 314080416, which is 32. Running the C code returns the correct result, but the generated circuit returns instead 7966496. However, it seems that the integration test expects this value, instead of the correct one.

I'm working in the VM provided in the README.

lucas-rami, 2024-05-20:
Hi! Thanks for the bug report :)

It looks like there is in fact an issue with our `gcd` algorithm. Both the C code and the circuit return 7966496 when running on my machine. I will soon push a fix.

closed 2024-05-20 by lucas-rami

chatpion, 2024-05-22:
Hi, does the circuit generated from your fixed code manage to return a value? On my end, the simulation was still not done after running for a few hours. This also happens when using smaller values for `a` and `b`.

commit e8a28e57 (2024-05-20): [integration-test] Fix `gcd` kernel implementation

The implementation of the `gcd` kernel, which computes the GCD of two
signed integers using the "binary GCD" algorithm, was incorrect,
yielding an incorrect result on the test input set in the kernel's main
function. This fixes the mistakes in the algorithm's implementation. The
kernel now returns the correct result for the test input.

Fixes #79.