EPFL-LAP/dynamatic #80: [integration-test] expecting channel<f32> but get channel<i32>

Jiahui17, 2024-05-23:
I hit this error on this benchmark (https://github.com/Jiahui17/dynamatic-mlir/blob/iwls24-sharing/integration-test/gsum/gsum.c):

```
.../handshake_export.mlir:23:11: error: 'hw.instance' op operand type #0 must be '!handshake.channel<f32>', but got '!handshake.channel<i32>'
    %18 = oehb [1] %8 {bb = 1 : ui32, name = #handshake.name<"oehb1">} : i32
          ^
.../handshake_export.mlir:23:11: note: see current operation: %20 = "hw.instance"(%9, %arg2, %arg3) {argNames = ["ins", "clk", "rst"], instanceName = "oehb1", moduleName = @handshake_oehb_1_32, parameters = [], resultNames = ["outs"]} : (!handshake.channel<i32>, i1, i1) -> !handshake.channel<f32>
/home/jxu/Repositories/dhls-utils/./dynamatic/integration-test/gsum/out/comp/handshake_export.mlir:14:11: note: module declared here
    %10 = oehb [1] %trueResult_10 {bb = 1 : ui32, name = #handshake.name<"oehb15">} : f32
          ^
[FATAL] Failed to lower to HW
```

lucas-rami, 2024-05-25:
I did not get the error in exactly the same place (though I did get an identical one at some other point in the IR) but this is not so surprising as I was actively working on the pass, so we may have slightly different versions. In any case, let me know if the issue persists!

closed 2024-05-25 by lucas-rami

Jiahui17, 2024-05-26:
great! thanks for the fix

commit a0adf4db (2024-05-25): [HandshakeToHW] Fix incorrect type in mem. converter

A data type was used in place of an address type.

Fixes #80.