EPFL-LAP/dynamatic #81: Compile error by switching order of two for loops,failing to generate LSQs

jirheee, 2024-06-04:
I am trying dynamatic for my personal project, and very strange compile error occurs. I changed the order of two for loops, but former successfully compiles while latter fails. I thought that dynamatic is capable of transforming arbitrary CFGs into circuits, but maybe there is a misunderstanding. Could you please check what I have done wrong? Thank you in advance!

### successful code snippet
```c
unsigned simple_mul(inout_int_t a[N]) {
  int tmp = 0;

  for (unsigned i = 0; i < N; ++i) {
    tmp += a[i];
  }

  for (unsigned i = 0; i < N; ++i) {
    a[i] = a[i] * a[i] * a[i];
  }

  return tmp;
}
```

### failed code snippet
```c
unsigned simple_mul(inout_int_t a[N]) {
  int tmp = 0;

  for (unsigned i = 0; i < N; ++i) {
    a[i] = a[i] * a[i] * a[i];
  }

  for (unsigned i = 0; i < N; ++i) {
    tmp += a[i];
  }

  return tmp;
}
```

### error message
```
dynamatic> compile             
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
[INFO] Compiled cf to handshake
[INFO] Applied transformations to handshake
[INFO] Built kernel for profiling
[INFO] Ran kernel for profiling
[INFO] Profiled cf-level
[INFO] Running smart buffer placement
[INFO] Placed smart buffers
[INFO] Canonicalized handshake
[INFO] Lowered to HW
[INFO] Created visual DOT
[INFO] Converted visual DOT to PNG
[INFO] Created simple_mul DOT
[INFO] Converted simple_mul DOT to PNG
[INFO] Compilation succeeded

dynamatic> write-hdl
[INFO] Converted DOT to VHDL
Error: Unable to access jarfile /home/ubuntu/jungin/dynamatic/bin/generators/lsq-generator.jar
[FATAL] Failed to generate LSQs

dynamatic> simulate
[INFO] Built kernel for IO gen.
[INFO] Ran kernel for IO gen.
[INFO] Launching Modelsim simulation
```

lucas-rami, 2024-06-04:
Hi! Thanks for reaching out.

> Error: Unable to access jarfile /home/ubuntu/jungin/dynamatic/bin/generators/lsq-generator.jar

I fixed this error [on the last commit](56de9ee). Can you pull and try again to see if the error still occurs?

lucas-rami, 2024-06-13:
Closing because this seems resolved. Feel free to re-open if the problem persists.

closed 2024-06-13 by lucas-rami