EPFL-LAP/dynamatic #110: Problem in the --handshake-minimize-lsq-usage pass

AyaElAkhras, 2024-08-10:
The new cfToHandshake pass implementing fast token delivery exposes an error in the --handshake-minimize-lsq-usage pass. A screenshot of the error message along with one handshake.mlir triggering the error are attached for reference. The mlir file is attached as a zip file because mlir files seem to not be supported.
[image]
[handshake.zip](https://github.com/user-attachments/files/16569867/handshake.zip)

AyaElAkhras, 2024-08-22:
There are cases (such as the one attached earlier) where the --handshake-minimize-lsq-usage pass fails to minimize the usage of LSQs for fast token delivery circuits.

pcineverdies, 2025-01-27:
This gets solved with the current FTD implementation, I'd re-open it in case we see new integration problem later on.

commit 617889a5 (2024-08-19): [CFG] Do not emit error on BB control id. failure

Our CFG analysis emits an error when it cannot identify the control
value of one of the "basic block" of a Handshake function. This is
overkill; client code may simply wish to skip an optimization in the
case where this identification is unsuccessful. This commit makes the
method simply return a logical failure without producing an erorr.

Fixes #110.
closed 2025-01-27 by pcineverdies