EPFL-LAP/dynamatic #111: Problem in lsq-generator.jar from within write-hdl --experimental

AyaElAkhras, 2024-08-10:
Sometimes, we get the following error (attached screenshot) that seems to be triggered from the lsq-generator. The error appears when we use the write-hdl --experimental. I'm attaching the mlir files (as zip files because mlir files seem to not be supported) of one example that triggers this error.
[hw.zip](https://github.com/user-attachments/files/16569946/hw.zip)
[handshake_export.zip](https://github.com/user-attachments/files/16569947/handshake_export.zip)
[handshake.zip](https://github.com/user-attachments/files/16569948/handshake.zip)
[image]

lucas-rami, 2024-08-19:
Thanks for the report. I was able to replicate the problem on the HEAD of `main` using the `hw.mlir` file your provided. 

After investigating the issue, it looks like the Chisel LSQ component cannot represent LSQs that have 0 load ports (`"numLoadPorts": 0` in the JSON configuration file). In particular, the component attempts to compute the base-2 logarithm of the number of load ports; when the latter is 0, it understandably fails and generates this error.

I tried to fix this quickly but unfortunately failed to do so (I fixed the error at the top-level but things didn't work in nested components either) due to my limited knowledge of Chisel and of the LSQ architecture. It would be great if someone more knowledgeable than me on these topics could try to tackle this. I suspect this is not too difficult of a problem to solve when one knows what they are doing.

lucas-rami, 2024-08-20:
> Could the problem be in the generation of the JSON configuration file? I'm suggesting so because all of the input C++ codes that trigger this problem contain load operations so they should result in the addition of load ports to the LSQ.

If it is it is not showcased by the example you provide. The `handshake.mlir` IR (that your pass spits out supposedly) does not contain any load port. So the LSQ generator is correctly instructed to have 0 load ports according to your circuit.

AyaElAkhras, 2024-08-20:
I tried running one example on the old and new Dynamatic and found out that the generated json files are different. Specifically, the "numLoadPorts" is 2 in the file generated by the old flow and is 0 in the file generated by the new flow. I'm attaching the two json files for reference.
[new.json](https://github.com/user-attachments/files/16678050/new.json)
[old.json](https://github.com/user-attachments/files/16678055/old.json)

I'm also attaching the C++ example file and the generated MLIR files, in case they are needed. 
[example.zip](https://github.com/user-attachments/files/16678093/example.zip) 
[mlir_files.zip](https://github.com/user-attachments/files/16678135/mlir_files.zip)

lucas-rami, 2024-08-20:
I just checked with the C example you provide and the HEAD of `main`. While there are loads in the source code these are optimized away by MLIR---one can see by looking at the code that they are indeed not necessary to guarantee correct execution)---therefore the LSQ is correctly configured to be generated with 0 load ports.

The problem does not happen with legacy Dynamatic because LLVM does not optimize away these loads (I have seen other similar examples where MLIR can better optimize the source code than LLVM). I guess this situation never presents itself in any of our other integration tests, so nobody ever noticed that the Chisel does not work when 0 load ports are requested (which I think is still a sane use case of the LSQ).

AyaElAkhras, 2024-08-21:
I see, then it is truly a limitation in the Chisel code.

Jiahui17, 2024-08-22:
Benchmarks failed to compile due to this issue (in #120 ):

`covariance`,
`covariance_float`,

A-Alosius, 2025-06-16:
I may be a little late on this, but I faced the same issue while testing the benchmarks mentioned by @Jiahui17 and also noticed the number of load ports issue @AyaElAkhras found. While I don't yet have much experience with this I came up with a speculation and tweaked the covariance benchmark to spot if the issue could be from the C code rather than from dynamatic. The covariance code that seems to cause the issue is this part

 /* Calculate the m * m covariance matrix. */
  for (j1 = 0; j1 < 30; j1++)
    for (j2 = j1; j2 < 30; j2++) {
      int x = 0;
      for (i = 0; i < 30; i++)
        x += data[i][j1] * data[i][j2];

      symmat[j1][j2] = x;  
      symmat[j2][j1] = x;
    }
}

The issue seems to come from the two assignments of the symmat variable. When one of them is commented out, the compilation is successful. My hypothesis is that the assignment might be creating simultaneous memory accesses that the compiler struggles to disambiguate because the indexes are variables and, in some cases, will be equal causing a conflict in writing. When the variables j1, and j2 are replaced with constants, compilation is also successful.
I also tried creating a separate array named symmat2 to receive the second assignment as is and it also worked fine.  I don't know if this helps in anyway, but I am open to learn. I will test @AyaElAkhras example and see if I find anything

AyaElAkhras, 2025-06-16:
Thanks @A-Alosius for experimenting here! 

Your analysis and conclusions make sense to me. However, I'm not aware of anyone currently working with the Chisel LSQ generator. I think the Python LSQ generator is the one actively used by everyone.

A-Alosius, 2025-06-16:
Hi @AyaElAkhras. I am using the most recent main branch for dynamatic (I cloned it 2 weeks ago. Not sure if any major updates have been made since then). It uses the python LSQ generator if I am not mistaken. It still gives me issues for the covariance integration test.

AyaElAkhras, 2025-06-16:
> It uses the python LSQ generator if I am not mistaken. It still gives me issues for the covariance integration test.
>
I see. I did not know that the problem exists there too.

closed 2025-11-05 by Jiahui17