EPFL-LAP/dynamatic #114: [Backend] Incorrect port naming in hls_verify_atax.vhd vs atax_wrapper.vhd

Jiahui17, 2024-08-20:
Benchmark: atax

Behavior: the port list in generated testbench is incompatible with the kernel

`hls_verify_atax.vhd`:

```vhdl
duv: 	 entity work.atax_wrapper
		port map (
			clk => tb_clk,
			rst => tb_rst,
			end_valid => tb_end_valid,
			end_ready => '1',
			A_address0 => A_mem_address0,
			A_ce0 => A_mem_ce0,
			A_we0 => A_mem_we0,
			A_din0 => A_mem_dout0,
			A_dout0 => A_mem_din0,
			A_address1 => A_mem_address1,
			A_ce1 => A_mem_ce1,
			A_we1 => A_mem_we1,
			A_din1 => A_mem_dout1,
			A_dout1 => A_mem_din1,
			x_address0 => x_mem_address0,
			x_ce0 => x_mem_ce0,
			x_we0 => x_mem_we0,
			x_din0 => x_mem_dout0,
			x_dout0 => x_mem_din0,
			x_address1 => x_mem_address1,
			x_ce1 => x_mem_ce1,
			x_we1 => x_mem_we1,
			x_din1 => x_mem_dout1,
			x_dout1 => x_mem_din1,
			y_address0 => y_mem_address0,
			y_ce0 => y_mem_ce0,
			y_we0 => y_mem_we0,
			y_din0 => y_mem_dout0,
			y_dout0 => y_mem_din0,
			y_address1 => y_mem_address1,
			y_ce1 => y_mem_ce1,
			y_we1 => y_mem_we1,
			y_din1 => y_mem_dout1,
			y_dout1 => y_mem_din1,
			tmp_address0 => tmp_mem_address0,
			tmp_ce0 => tmp_mem_ce0,
			tmp_we0 => tmp_mem_we0,
			tmp_din0 => tmp_mem_dout0,
			tmp_dout0 => tmp_mem_din0,
			tmp_address1 => tmp_mem_address1,
			tmp_ce1 => tmp_mem_ce1,
			tmp_we1 => tmp_mem_we1,
			tmp_din1 => tmp_mem_dout1,
			tmp_dout1 => tmp_mem_din1,
			start_ready => tb_start_ready,
			start_valid => tb_start_valid
		);

```

`atax.vhd`:

```vhdl
entity atax_wrapper is
  port (
    -- inputs
    start_valid : in std_logic;
    clk : in std_logic;
    rst : in std_logic;
    end_ready : in std_logic;
    -- outputs
    start_ready : out std_logic;
    end_valid : out std_logic;
    A_ce0 : out std_logic;
    A_we0 : out std_logic;
    A_address0 : out std_logic_vector(31 downto 0);
    A_ce1 : out std_logic;
    A_we1 : out std_logic;
    A_address1 : out std_logic_vector(31 downto 0);
    x_ce0 : out std_logic;
    x_we0 : out std_logic;
    x_address0 : out std_logic_vector(31 downto 0);
    x_ce1 : out std_logic;
    x_we1 : out std_logic;
    x_address1 : out std_logic_vector(31 downto 0);
    y_ce0 : out std_logic;
    y_we0 : out std_logic;
    y_address0 : out std_logic_vector(31 downto 0);
    y_ce1 : out std_logic;
    y_we1 : out std_logic;
    y_address1 : out std_logic_vector(31 downto 0);
    tmp_ce0 : out std_logic;
    tmp_we0 : out std_logic;
    tmp_address0 : out std_logic_vector(31 downto 0);
    tmp_ce1 : out std_logic;
    tmp_we1 : out std_logic;
    tmp_address1 : out std_logic_vector(31 downto 0)
  );
end entity;

```

lucas-rami, 2024-08-22:
Thanks for the report. I am unable to replicate this on the HEAD of `main` (currently 88aa87c). I have tried the frontend script (minus smart buffer placement i.e. I ran `compile --simple-buffers`) you provide on `atax` and `kernel_3mm_float` and the RTL compiles and simulates correctly.

Jiahui17, 2024-08-22:
Thanks for checking, I will double check if I am building the whole thing correctly before getting back to you

Jiahui17, 2024-08-22:
Ok, I think the problem is that I was on an outdated branch

Everything is fine now