EPFL-LAP/dynamatic #115: [handshake] failed to legalize operation 'arith.sitofp'

Jiahui17, 2024-08-20:
Benchmark: `dct.c`

Behavior:

```
[INFO] Applied Dynamatic transformations to cf
/home/jxu/Repositories/dhls-utils/dynamatic/./integration-test/dct/out/comp/cf_dyn_transformed.mlir:8:10: error: failed to legalize operation 'arith.sitofp'
    %2 = arith.sitofp %1 {handshake.name = "sitofp0"} : i32 to f64
         ^
/home/jxu/Repositories/dhls-utils/dynamatic/./integration-test/dct/out/comp/cf_dyn_transformed.mlir:8:10: note: see current operation: %19 = "arith.sitofp"(%18) {handshake.bb = 1 : ui32, handshake.name = "sitofp0"} : (i32) -> f64
[FATAL] Failed to compile cf to handshake
```

lucas-rami, 2024-08-22:
This is related to #102. I did not create Handshake equivalent for these two operations because I am not sure that we have working RTL components for them. If we do then equivalent operations can be added to Handshake super easily, then they may be trivially converted in the `CfToHandshake` pass, and finally be added to the backend support.

Jiahui17, 2024-08-26:
Closed this because it is a duplicate of #102

closed 2024-08-26 by Jiahui17