EPFL-LAP/dynamatic #116: [Backend] Incompatible lsq port naming in top-level module

Jiahui17, 2024-08-20:
Benchmark: `triangular.c`

Behavior:

In `triangular.vhd`

```vhdl
  lsq0 : entity work.handshake_lsq_lsq0
    port map(
      clock => clk,
      io_ctrl_0_ready => fork17_outs_2_ready,
      io_ctrl_0_valid => fork17_outs_2_valid,
      io_ldAddrToMC_bits => lsq0_ldAddrToMC,
      io_ldAddrToMC_ready => lsq0_ldAddrToMC_ready,
      io_ldAddrToMC_valid => lsq0_ldAddrToMC_valid,
      io_ldAddr_0_bits => lsq_load0_addrOut,
      io_ldAddr_0_ready => lsq_load0_addrOut_ready,
      io_ldAddr_0_valid => lsq_load0_addrOut_valid,
      io_ldDataFromMC_bits => mem_controller1_ldData_1,
      io_ldDataFromMC_ready => mem_controller1_ldData_1_ready,
      io_ldDataFromMC_valid => mem_controller1_ldData_1_valid,
      io_ldData_0_bits => lsq0_ldData_0,
      io_ldData_0_ready => lsq0_ldData_0_ready,
      io_ldData_0_valid => lsq0_ldData_0_valid,
      io_stAddrToMC_bits => lsq0_stAddrToMC,
      io_stAddrToMC_ready => lsq0_stAddrToMC_ready,
      io_stAddrToMC_valid => lsq0_stAddrToMC_valid,
      io_stAddr_0_bits => lsq_store0_addrOut,
      io_stAddr_0_ready => lsq_store0_addrOut_ready,
      io_stAddr_0_valid => lsq_store0_addrOut_valid,
      io_stDataToMC_bits => lsq0_stDataToMC,
      io_stDataToMC_ready => lsq0_stDataToMC_ready,
      io_stDataToMC_valid => lsq0_stDataToMC_valid,
      io_stData_0_bits => lsq_store0_dataToMem,
      io_stData_0_ready => lsq_store0_dataToMem_ready,
      io_stData_0_valid => lsq_store0_dataToMem_valid,
      reset => rst
    );
```

In `handshake_lsq_lsq0.v`:
```vhdl
module handshake_lsq_lsq0(	// src/main/scala/lsq/LSQBRAMExperimental.scala:20:7
  input         clock,	// src/main/scala/lsq/LSQBRAMExperimental.scala:20:7
                reset,	// src/main/scala/lsq/LSQBRAMExperimental.scala:20:7
  output [31:0] io_storeData,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
                io_storeAddr,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_storeEn,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input  [31:0] io_loadData,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output [31:0] io_loadAddr,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_loadEn,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
                io_ctrl_0_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_ctrl_0_valid,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_ldAddr_0_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_ldAddr_0_valid,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input  [31:0] io_ldAddr_0_bits,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_ldData_0_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_ldData_0_valid,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output [31:0] io_ldData_0_bits,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_stAddr_0_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_stAddr_0_valid,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input  [31:0] io_stAddr_0_bits,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_stData_0_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_stData_0_valid,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input  [31:0] io_stData_0_bits,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  input         io_memDone_ready,	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
  output        io_memDone_valid	// src/main/scala/lsq/LSQBRAMExperimental.scala:23:14
);
```

The signal `io_ldDataFromMC_bits` appears in the port list of the module instantiation but not in the port list of the LSQ itself.

lucas-rami, 2024-08-22:
Similarly to #114 I am unable to replicate the issue on the HEAD of `main`. The port `io_ldDataFromMC_bits` exists in my LSQ's interface, see below.

```verilog
module handshake_lsq_lsq0(	// src/main/scala/lsq/LSQSlave.scala:20:7
  input         clock,	// src/main/scala/lsq/LSQSlave.scala:20:7
                reset,	// src/main/scala/lsq/LSQSlave.scala:20:7
                io_ldAddrToMC_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_ldAddrToMC_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output [31:0] io_ldAddrToMC_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_ldDataFromMC_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_ldDataFromMC_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input  [31:0] io_ldDataFromMC_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_stAddrToMC_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_stAddrToMC_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output [31:0] io_stAddrToMC_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_stDataToMC_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_stDataToMC_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output [31:0] io_stDataToMC_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_ctrl_0_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_ctrl_0_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_ldAddr_0_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_ldAddr_0_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input  [31:0] io_ldAddr_0_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_ldData_0_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_ldData_0_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output [31:0] io_ldData_0_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_stAddr_0_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_stAddr_0_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input  [31:0] io_stAddr_0_bits,	// src/main/scala/lsq/LSQSlave.scala:23:14
  output        io_stData_0_ready,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input         io_stData_0_valid,	// src/main/scala/lsq/LSQSlave.scala:23:14
  input  [31:0] io_stData_0_bits	// src/main/scala/lsq/LSQSlave.scala:23:14
);
```

Can you make sure that the LSQ generator is built correctly by the build script? It was changed semi-recently and it looks like the LSQ you get does not reflect these changes.

Jiahui17, 2024-08-22:
Everything is fine now

closed 2024-08-22 by Jiahui17