EPFL-LAP/dynamatic #117: [Backend] Incorrect/incomplete MC port naming and mem_to_bram in module instantiation

Jiahui17, 2024-08-21:
Benchmark: `gsum`

Script:

```
set-dynamatic-path .;     set-src ./integration-test/gsum/gsum.c;     set-clock-period 4;     compile;     write-hdl --hdl vhdl;     simulate;     synthesize;     exit
```

# Issue 1:

`gsum.vhd`:

```vhdl
  mem_controller0 : entity work.mem_controller_storeless(arch) generic map(1, 32, 32)
    port map(
      clk => clk,
      ldAddr(0) => mc_load0_addrOut,
      ldAddr_ready(0) => mc_load0_addrOut_ready,
      ldAddr_valid(0) => mc_load0_addrOut_valid,
      ldData(0) => mem_controller0_ldData_0,
      ldData_ready(0) => mem_controller0_ldData_0_ready,
      ldData_valid(0) => mem_controller0_ldData_0_valid,
      loadAddr => mem_controller0_loadAddr,
      loadEn => mem_controller0_loadEn,
      memDone_ready => mem_controller0_memDone_ready,
      memDone_valid => mem_controller0_memDone_valid,
      rst => rst,
      storeAddr => mem_controller0_storeAddr,
      storeEn => mem_controller0_storeEn
    );
```

In `mem_controller_storeless.vhd`

```vhdl
entity mem_controller_storeless is
  generic (
    NUM_LOADS : integer;
    DATA_WIDTH : integer;
    ADDR_WIDTH : integer
  );
  port (
    clk, rst : in std_logic;
    -- load address input channels
    ldAddr       : in  data_array (NUM_LOADS - 1 downto 0)(ADDR_WIDTH - 1 downto 0);
    ldAddr_valid : in  std_logic_vector(NUM_LOADS - 1 downto 0);
    ldAddr_ready : out std_logic_vector(NUM_LOADS - 1 downto 0);
    -- load data output channels
    ldData       : out data_array (NUM_LOADS - 1 downto 0)(DATA_WIDTH - 1 downto 0);
    ldData_valid : out std_logic_vector(NUM_LOADS - 1 downto 0);
    ldData_ready : in  std_logic_vector(NUM_LOADS - 1 downto 0);
    --- memory done channel
    memDone_valid : out std_logic;
    memDone_ready : in  std_logic;
    -- interface to dual-port BRAM
    loadData  : in  std_logic_vector(DATA_WIDTH - 1 downto 0);
    loadEn    : out std_logic;
    loadAddr  : out std_logic_vector(ADDR_WIDTH - 1 downto 0);
    storeEn   : out std_logic;
    storeAddr : out std_logic_vector(ADDR_WIDTH - 1 downto 0);
    storeData : out std_logic_vector(DATA_WIDTH - 1 downto 0)
  );
end entity;
```

The port `loadData` is in the port list of the MC_storeless module, but not in the instantiation.

# Issue 2:

In `gsum_wrapper.vhd`:
```vhdl
  mem_to_bram_converter_a : entity work.mem_to_bram(arch) generic map(32, 32)
    port map(
      address0 => mem_to_bram_converter_a_address0,
      address1 => mem_to_bram_converter_a_address1,
      ce0 => mem_to_bram_converter_a_ce0,
      ce1 => mem_to_bram_converter_a_ce1,
      loadAddr => gsum_wrapped_a_loadAddr,
      loadEn => gsum_wrapped_a_loadEn,
      storeAddr => gsum_wrapped_a_storeAddr,
      storeEn => gsum_wrapped_a_storeEn,
      we0 => mem_to_bram_converter_a_we0,
      we1 => mem_to_bram_converter_a_we1
    );
```

In `mem_to_bram.vhd`:

```vhdl
entity mem_to_bram is
  generic (
    DATA_WIDTH : integer;
    ADDR_WIDTH : integer
  );
  port (
    -- from circuit
    loadEn    : in std_logic;
    loadAddr  : in std_logic_vector(ADDR_WIDTH - 1 downto 0);
    storeEn   : in std_logic;
    storeAddr : in std_logic_vector(ADDR_WIDTH - 1 downto 0);
    storeData : in std_logic_vector(DATA_WIDTH - 1 downto 0);
    -- to BRAM
    ce0      : out std_logic;
    we0      : out std_logic;
    address0 : out std_logic_vector(ADDR_WIDTH - 1 downto 0);
    dout0    : out std_logic_vector(DATA_WIDTH - 1 downto 0);
    ce1      : out std_logic;
    we1      : out std_logic;
    address1 : out std_logic_vector(ADDR_WIDTH - 1 downto 0);
    dout1    : out std_logic_vector(DATA_WIDTH - 1 downto 0);
    -- back from BRAM
    din0 : in std_logic_vector(DATA_WIDTH - 1 downto 0);
    din1 : in std_logic_vector(DATA_WIDTH - 1 downto 0);
    -- back to circuit
    loadData : out std_logic_vector(DATA_WIDTH - 1 downto 0)
  );
end entity;
```

The ports `storeData`, `din0`, and `din1` are missing in the instantiation in `gsum_wrapper.vhd`.

lucas-rami, 2024-08-22:
Same as #114 and #116, I'm unable to replicate the issue with the current HEAD of `main`.

Unrelated sidenote, I realize that simulation will fail for floating-point based benchmarks because of floating point inaccuracies. I guess it is possible and normal that there are small differences between our CPUs FP accuracy and the one of our RTL? On `gsum` I get this token mismatch adter simulation for example.

```
[ERROR COMPARE] Token mismatch: [0x5889794e] and [0x58895cb4] are not equal (at transaction id 0).
```

Jiahui17, 2024-08-22:
Everything is fine now

(I will open another issue for the FP mismatch)

closed 2024-08-22 by Jiahui17