EPFL-LAP/dynamatic #152: [Handshake] Branch Canonicalization

kb29x37, 2024-09-21:
Sometimes, both Branch outputs connect to the same CMerge / same Multiplexer. In such cases, the Branch and the component it connects to can be removed.

For example, in the binary_search benchmark, the situation occurs between cond_br19 and control_merge6, cond_br16 and mux17, cond_br2 and mux23, etc.

See the example attached.

[image]
[binary_search.txt](https://github.com/user-attachments/files/17084988/binary_search.txt)

Jiahui17, 2024-09-21:
Maybe we put this in #148?

lucas-rami, 2024-09-21:
I think these should be separate; the other one can be completed independently of this one (even if I mention over there that we should think of new optimizations it is mostly focused on the mechanical refactoring of the canonicalization patterns). I also made the other one as an a "good first issue" so I would like to keep it simple if possible.

AyaElAkhras, 2024-09-22:
Such situations and more are systematically tackled by a set of transformations packaged in a term rewriting system that I'm currently working on.

closed 2026-10-31 by Jiahui17