EPFL-LAP/dynamatic #154: Exception during compile commande on Introduction/Ch1

leothaud, 2024-09-25:
I built dynamatic from sources and installed its dependencies by following the readme. Then, I tried to use dynamatic on Ch1 of the Introduction example, but the compile command threw a GRBException. Do you know how to solve the problem? You can find below the complete trace of my dynamic session.

```
================================================================================
============== Dynamatic | Dynamic High-Level Synthesis Compiler ===============
======================== EPFL-LAP - v2.0.0 | March 2024 ========================
================================================================================

dynamatic> set-src tutorials/Introduction/Ch1/loop_multiply.c 
dynamatic> compile
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
[INFO] Compiled cf to handshake
[INFO] Applied transformations to handshake
[INFO] Built kernel for profiling
[INFO] Ran kernel for profiling
[INFO] Profiled cf-level
[INFO] Running smart buffer placement with CP = 4.000
terminate called after throwing an instance of 'GRBException'
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /home/dleothau/work/dynamatic/bin/dynamatic-opt /home/dleothau/work/dynamatic/tutorials/Introduction/Ch1/out/comp/handshake_transformed.mlir --handshake-set-buffering-properties=version=fpga20 "--handshake-place-buffers=algorithm=fpl22 frequencies=/home/dleothau/work/dynamatic/tutorials/Introduction/Ch1/out/comp/frequencies.csv timing-models=/home/dleothau/work/dynamatic/data/components.json target-period=4.000 timeout=300 dump-logs"
 #0 0x000056af37655126 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0xd8126)
 #1 0x000056af37652d7e llvm::sys::RunSignalHandlers() (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0xd5d7e)
 #2 0x000056af37655b54 SignalHandler(int) Signals.cpp:0:0
 #3 0x00007c024ec4c1d0 (/usr/lib/libc.so.6+0x3d1d0)
 #4 0x00007c024eca53f4 __pthread_kill_implementation /usr/src/debug/glibc/glibc/nptl/pthread_kill.c:44:76
 #5 0x00007c024ec4c120 raise /usr/src/debug/glibc/glibc/signal/../sysdeps/posix/raise.c:27:6
 #6 0x00007c024ec334c3 abort /usr/src/debug/glibc/glibc/stdlib/abort.c:81:7
 #7 0x00007c024ee97b2c __gnu_cxx::__verbose_terminate_handler() (.cold) /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/vterminate.cc:75:10
 #8 0x00007c024eeadf3a __cxxabiv1::__terminate(void (*)()) /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48:15
 #9 0x00007c024ee9752a std::unexpected() /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:71:1
#10 0x00007c024eeae1f6 (/usr/lib/libstdc++.so.6+0xae1f6)
#11 0x000056af37639a96 GRBEnv::start() (.cold.25) Env.o:0:0
#12 0x000056af376ae055 dynamatic::buffer::extractCFDFC(dynamatic::handshake::FuncOp, llvm::SetVector<dynamatic::experimental::ArchBB*, llvm::SmallVector<dynamatic::experimental::ArchBB*, 0u>, llvm::DenseSet<dynamatic::experimental::ArchBB*, llvm::DenseMapInfo<dynamatic::experimental::ArchBB*, void>>, 0u>&, llvm::SetVector<unsigned int, llvm::SmallVector<unsigned int, 0u>, llvm::DenseSet<unsigned int, llvm::DenseMapInfo<unsigned int, void>>, 0u>&, llvm::SetVector<dynamatic::experimental::ArchBB*, llvm::SmallVector<dynamatic::experimental::ArchBB*, 0u>, llvm::DenseSet<dynamatic::experimental::ArchBB*, llvm::DenseMapInfo<dynamatic::experimental::ArchBB*, void>>, 0u>&, unsigned int&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, int*) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x131055)
#13 0x000056af3769d597 dynamatic::buffer::HandshakePlaceBuffersPass::getCFDFCs(dynamatic::buffer::FuncInfo&, dynamatic::Logger*, llvm::SmallVector<dynamatic::buffer::CFDFC, 1u>&) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x120597)
#14 0x000056af3769be77 dynamatic::buffer::HandshakePlaceBuffersPass::placeBuffers(dynamatic::buffer::FuncInfo&, dynamatic::TimingDatabase&) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x11ee77)
#15 0x000056af3769a4a9 dynamatic::buffer::HandshakePlaceBuffersPass::placeUsingMILP() (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x11d4a9)
#16 0x000056af37698c8f dynamatic::buffer::HandshakePlaceBuffersPass::runDynamaticPass() (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x11bc8f)
#17 0x000056af37693c28 dynamatic::DynamaticPass::runOnOperation() (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x116c28)
#18 0x000056af37f68e1f mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x9ebe1f)
#19 0x000056af37f6c718 mlir::PassManager::run(mlir::Operation*) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x9ef718)
#20 0x000056af37be7cee performActions(llvm::raw_ostream&, std::shared_ptr<llvm::SourceMgr> const&, mlir::MLIRContext*, mlir::MlirOptMainConfig const&) MlirOptMain.cpp:0:0
#21 0x000056af37be6de0 mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&)::$_0>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) MlirOptMain.cpp:0:0
#22 0x000056af37c039e2 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x6869e2)
#23 0x000056af37be3117 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x666117)
#24 0x000056af37be35ac mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&) (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0x6665ac)
#25 0x000056af37642bf9 main (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0xc5bf9)
#26 0x00007c024ec34e08 __libc_start_call_main /usr/src/debug/glibc/glibc/csu/../sysdeps/nptl/libc_start_call_main.h:74:3
#27 0x00007c024ec34ecc call_init /usr/src/debug/glibc/glibc/csu/../csu/libc-start.c:128:20
#28 0x00007c024ec34ecc __libc_start_main /usr/src/debug/glibc/glibc/csu/../csu/libc-start.c:347:5
#29 0x000056af376427e5 _start (/home/dleothau/work/dynamatic/bin/dynamatic-opt+0xc57e5)
/home/dleothau/work/dynamatic/tools/dynamatic/scripts/compile.sh: line 169: 60091 Aborted                 (core dumped) "$DYNAMATIC_OPT_BIN" "$F_HANDSHAKE_TRANSFORMED" --handshake-set-buffering-properties="version=fpga20" --$BUFFER_PLACEMENT_PASS="algorithm=fpl22 frequencies=$F_FREQUENCIES timing-models=$DYNAMATIC_DIR/data/components.json target-period=$TARGET_CP timeout=300 dump-logs" > "$F_HANDSHAKE_BUFFERED"
[FATAL] Failed to place smart buffers
```

Jiahui17, 2024-09-26:
Hi, thank you for reporting the issue! 

For now, I suggest to check out [this link](https://github.com/Jiahui17/dynamatic-scripts?tab=readme-ov-file#prerequisite-get-your-gurobi-license) to properly configure gurobi before doing another clean build. 

After setting up gurobi, you could run the command below. If everything is fine, you will see output like what we have below:

```
$ /opt/gurobi1003/linux64/bin/gurobi_cl
Set parameter Username
Set parameter LogFile to value "gurobi.log"
Academic license - for non-commercial use only - expires 20XX-XX-XXX
Using license file /opt/gurobi1003/linux64/gurobi.lic

Usage: gurobi_cl [--command]* [param=value]* filename
Type 'gurobi_cl --help' for more information.
```

I will add a section on the main README on how to configure gurobi.

leothaud, 2024-09-27:
Thank you. The problem was that I needed a gurobi license.

closed 2024-09-27 by Jiahui17