EPFL-LAP/dynamatic #155: [Bitwidth] Array indices are always inefficiently set to 32-bit

Jiahui17, 2024-09-26:
## Commit: 
[58fa4c6](https://github.com/EPFL-LAP/dynamatic/commit/58fa4c62567f5dd006759fb1aaeade95ca739291)

## Commands: 
```
dynamatic> set-dynamatic-path ./dynamatic;   set-src ./dynamatic/integration-test/fir/fir.c;   set-clock-period 4;   compile  --sharing;   write-hdl --hdl verilog;   simulate;   exit
```

## Issue:

Let's consider the file `integration-test/fir/out/comp/handshake_export.mlir` and the memory controller/LSQ of an array with 1000 elements:

```mlir
    %outputs_0, %memEnd_1 = mem_controller[%arg0 : memref<1000xi32>] %arg2 (%addressResult_2) %46#1  \
        {connectedBlocks = [1 : i32], handshake.bufProps = #handshake<bufProps{"0": [0,0], [0,0], 0.000000e+00, 0.000000e+00, 0.000000e+00, "1": [0,0], [0,0], 0.000000e+00, 0.000000e+00, 0.000000e+00, "2": [0,0], [0,0], 0.000000e+00, 0.000000e+00, 0.000000e+00}>, \
        handshake.name = "mem_controller3"} : (!handshake.channel<i32>) -> !handshake.channel<i32>
```

Although the maximum array index (1000) can be represented using 10 bits, the memory controller above uses 32-bit addresses, which causes unnecessary resource overhead.

Jiahui17, 2024-09-26:
@lucas-rami Thank you so much for the quick answer! I will take a quick look to see if the verifier needs a fundamental change to support it.

lucas-rami, 2024-09-26:
Oh my bad then! Then let's make the default behavior yes. I can make that change now, will push a PR soon.

Jiahui17, 2024-09-26:
Sounds good, thank you!

lucas-rami, 2024-09-26:
Ah it looks like the rewrite patterns no longer work as they used to (there were most likely memory interface changes which broke them sometime in the last year). It's probably a small fix but I'm a bit too tired to attempt this now. Will look over this and push it through the finish line during the weekend.

PR #157 (2024-09-30): [HandshakeOptimizeBitwidths] Optimize width of address-carrying channels
Currently, all our address-carrying channels in the IR and in the final circuit are hardcoded so that their data signal is 32-bits. Since Dynamatic enforces the size of all memories to be known at compile-time, we have the opportunity to thin out these signals to the bare minimum number of bits necessary to index into each particular memory region.

This commit adds this optimization to the default compilation flow so that all Dynamatic-generated circuits can benefit from it. Main changes are detailed below.

- The `--han [... cut]
commit 2b5761a7 (2024-09-30): [Handshake] Optimize width of address-carrying channels (#157)

Currently, all our address-carrying channels in the IR and in the final
circuit are hardcoded so that their data signal is 32-bits. Since
Dynamatic enforces the size of all memories to be known at compile-time,
we have the opportunity to thin out these signals to the bare minimum
number of bits necessary to index into each particular memory region.

This commit adds this optimization to the default compilation flow so
that all Dynamatic-generated circuits can benefit from it. Main changes
are detailed below.

- The `--handshake-op [... cut]
commit 282c5990 (2024-10-15): [Handshake] Optimize width of address-carrying channels (#157)

Currently, all our address-carrying channels in the IR and in the final
circuit are hardcoded so that their data signal is 32-bits. Since
Dynamatic enforces the size of all memories to be known at compile-time,
we have the opportunity to thin out these signals to the bare minimum
number of bits necessary to index into each particular memory region.

This commit adds this optimization to the default compilation flow so
that all Dynamatic-generated circuits can benefit from it. Main changes
are detailed below.

- The `--handshake-op [... cut]
closed 2024-09-30 by lucas-rami