EPFL-LAP/dynamatic #161: Casting floats to double issue

murphe67, 2024-10-07:
In the ADMM integration test, the line:
    float u_bound = vdc * 0.5;
causes the error:
./dynamatic;   set-src ./dynamatic/integration-test/admm/admm.c;   set-clock-period 4;   compile --simple-buffers ;   write-hdl --hdl vhdl;   simulate; #  synthesize;   exit
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
/home/emurphy/dynamatic-scripts/./dynamatic/integration-test/admm/out/comp/cf_dyn_transformed.mlir:5:10: error: failed to legalize operation 'arith.extf'
    %0 = arith.extf %arg0 {handshake.name = "extf0"} : f32 to f64
         ^
/home/emurphy/dynamatic-scripts/./dynamatic/integration-test/admm/out/comp/cf_dyn_transformed.mlir:5:10: note: see current operation: %14 = "arith.extf"(<<UNKNOWN SSA VALUE>>) {handshake.bb = 0 : ui32, handshake.name = "extf0"} : (f32) -> f64
[FATAL] Failed to compile cf to handshake

which I interpret as extending vdc (a float) to a double is not supported.

changing the 0.5 (implicit double) to 0.5f (explicit float) fixed the error, e.g.:
    float u_bound = vdc * 0.5f;

murphe67, 2024-10-07:
I'm seeing the same issue you're seeing on the other branch ("admm_circular_integration_test"), if that helps

Jiahui17, 2024-11-08:
@murphe67 Could you check this again in the current main? This bug is not reproduced on my end

murphe67, 2024-11-25:
Just ran it on my end, worked perfectly

closed 2024-11-25 by murphe67