EPFL-LAP/dynamatic #164: [MLIR] [LLVM] ninja check-dynamatic failed

PigBrainOverflow, 2024-10-09:
[0/1] Running the Dynamatic regression tests
FAIL: Dynamatic :: Dialect/Handshake/invalid.mlir (26 of 26)
******************** TEST 'Dynamatic :: Dialect/Handshake/invalid.mlir' FAILED ********************
Exit Code: 139

Command Output (stderr):
--
RUN: at line 1: ~/envs/dynamatic/build/bin/dynamatic-opt ~/envs/dynamatic/test/Dialect/Handshake/invalid.mlir --split-input-file --verify-diagnostics
+ ~/envs/dynamatic/build/bin/dynamatic-opt ~/envs/dynamatic/test/Dialect/Handshake/invalid.mlir --split-input-file --verify-diagnostics
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ~/envs/dynamatic/build/bin/dynamatic-opt ~/envs/dynamatic/test/Dialect/Handshake/invalid.mlir --split-input-file --verify-diagnostics
1.      MLIR Parser: custom op parser 'handshake.func'
#0 0x000055b50b5cb277 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (~/envs/dynamatic/build/bin/dynamatic-opt+0xc6277)
#1 0x000055b50b5c90ce llvm::sys::RunSignalHandlers() (~/envs/dynamatic/build/bin/dynamatic-opt+0xc40ce)
#2 0x000055b50b5cbbfa SignalHandler(int) Signals.cpp:0:0
#3 0x00007fad1b4cf520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
~/envs/dynamatic/build/test/Dialect/Handshake/Output/invalid.mlir.script: line 1:  6162 Segmentation fault      (core dumped) ~/envs/dynamatic/build/bin/dynamatic-opt ~/envs/dynamatic/test/Dialect/Handshake/invalid.mlir --split-input-file --verify-diagnostics

--

********************
********************
Failed Tests (1):
  Dynamatic :: Dialect/Handshake/invalid.mlir

Testing Time: 0.13s
  Passed: 25
  Failed:  1
FAILED: test/CMakeFiles/check-dynamatic ~/envs/dynamatic/build/test/CMakeFiles/check-dynamatic 
cd ~/envs/dynamatic/build/test && /usr/bin/python3.10 ~/envs/dynamatic/polygeist/llvm-project/build/./bin/llvm-lit -sv ~/envs/dynamatic/build/test
ninja: build stopped: subcommand failed.

Jiahui17, 2024-11-08:
Hi @PigBrainOverflow thank you for reporting this issue! 

I played with different settings and I discovered that: 
1. The "expect-error" thing only works if we build dynamatic in Debug mode.
2. If we remove the "expect-error" comment line, dynamatic-opt will not throw a seg fault but a normal error message.

Maybe we can put this caveat in the [GettingStarted](https://github.com/Jiahui17/dynamatic/blob/main/docs/GettingStarted.md#testing) guide.

I will close this issue for now; feel free to reopen it if the issue persists after building the project in the Debug mode.

ebosnjak, 2024-11-11:
As @paolo-ienne suggested, I am reopening this, since this behavior should be considered a bug, for someone to fix at some point.

PR #248 (2025-02-03): [Test] Skip invalid test when building in release mode
Currently, when building in release mode, the unit test `invalid.mlir` fails, due to mlir rising a segmentation fault rather than the expected error message. 

On the one hand, we cannot get rid of this test, since it is actually useful to validate the handshake dialect, and it still runs correctly in debug mode.
On the other hand, it's important to have a working test suite, to be integrated in the CI/CD pipeline. 

This PR fixes the problem by relying on a CMAKE macro to know whether the project was built in release or debug mode, and po [... cut]
commit ae581a34 (2025-02-03): [Test] Skip invalid test when building in release mode (#248)

Currently, when building in release mode, the unit test `invalid.mlir`
fails, due to mlir rising a segmentation fault rather than the expected
error message.

On the one hand, we cannot get rid of this test, since it is actually
useful to validate the handshake dialect, and it still runs correctly in
debug mode.
On the other hand, it's important to have a working test suite, to be
integrated in the CI/CD pipeline.

This PR fixes the problem by relying on a CMAKE macro to know whether
the project was built in release or debug mode,  [... cut]
closed 2025-02-03 by pcineverdies