EPFL-LAP/dynamatic #165: Failed to export RTL (vhdl) with exp-sharing-wrapper-generator

udareechk, 2024-10-11:
Hi, 

I built your latest code on Ubuntu 22.04 from the source code and tested the tutorial example in Ch1 to be working.
I'm trying to run the script in the PR [Sharing][hdl] Resource sharing support in VHDL backend #85. I get an error in exporting to RTL for benchmark gsum.c (given in the script).

The output I get is below:

[image]

Is there anything I have missed in the build process to generate the exp-sharing-wrapper-generator? How can I solve this?

Thanks.

Jiahui17, 2024-10-11:
Thanks for reporting the issue! 

I think the problem is that the binary is not yet symlinked to `dynamatic/bin` in `build.sh`. I will try to fix this soon. 

For the time being, you could try to do this manually:

```sh
ln -sf ./dynamatic/build/bin/exp-sharing-wrapper-generator ./dynamatic/bin/exp-sharing-wrapper-generator
```

closed 2024-10-11 by Jiahui17

udareechk, 2024-10-11:
Hi, Thanks for the quick support and the fix provided!

commit 88e56a68 (2024-10-11): [build.sh] Update build script for sharing

The binary was not symlinked from dynamatic/build/bin to dynamatic/bin

Fixes #165
commit 4bba542f (2024-10-15): [build.sh] Update build script for sharing

The binary was not symlinked from dynamatic/build/bin to dynamatic/bin

Fixes #165