EPFL-LAP/dynamatic #166: Using the bash script to test Resource Sharing: Simulation Error for float_basic.c

udareechk, 2024-10-11:
Hi, 

I am trying to use the provided bash script in merged PR [Sharing][hdl] Resource sharing support in VHDL backend #85.
When I test the script with "polyn_mult.c" & "float_basic.c" given as benchmarks in the script, the generated DOT diagrams do not show any operations being shared. Is this the expected behaviour for these two applications or should they exhibit a resource-sharing output? (For "gsum.c" the output DOT diagram shows a resource-sharing wrapper)

Also, with the float_basic.c application, after compilation and RTL generation completes, I get an error in simulation with the application with the below error message:
```
[INFO] Launching Modelsim simulation
** Warning: setting ADDR NO RANDOMIZE failed - Success.
[ERROR COMPARE] Token mismatch: [oxffc00000] and [0x7f800001] are not equal (at transaction id 0).
FATAL Simulation failed
```
What is the cause of the simulation failure and how to solve this?

Thanks.

Jiahui17, 2024-10-11:
>  Just to clarify, do you mean polyn_mult & basic_float are not relevant example applications that will show sharing?

They are---I think we also share some integer multipliers there

udareechk, 2024-10-15:
Hi, 

Thank you for this information. How can I get information or find out which units are being shared with which operations in these two applications? In my compilation, I do not see a sharing wrapper generated for these two applications as seen in the gsum application. 

Thanks.

Jiahui17, 2024-10-15:
Oh i see, probably because the algorithm discovered that sharing muls in these apps will penalize performance. Could you change `SIMPLE_BUFFERS="--simple-buffers"` in the script? This will tell the optimization tool to ignore throughput and share all the resources

udareechk, 2024-10-15:
Hi, Thanks for your reply and the given information. 

I'm sorry, but isn't this what's already being used in the script? I'm using your script in PR [Sharing][hdl] Resource sharing support in VHDL backend https://github.com/EPFL-LAP/dynamatic/pull/85. So, it seems to be already using `SIMPLE_BUFFERS="--simple-buffers"` in the compilation. 

Can you kindly verify this on your side and let me know if I am missing anything else in the tool configuration/ compilation ?

Thanks!

Jiahui17, 2024-10-15:
Sorry, it was my bad, forgot that I removed integer multipliers from the list of sharable ops before doing the PR. It is now supported in main

udareechk, 2024-10-18:
Hi, Thanks for checking out the issue and for the support. The other two example applications show sharing now (except for the floating point comparison issue in simulation for float_basic.c).

May I know why only integer multipliers are shared and integer adders are not shared in these two applications?

Thanks!

Jiahui17, 2024-10-20:
> May I know why only integer multipliers are shared and integer adders are not shared in these two applications?

Sharing adders is possible; we are not doing it because adders are very cheap and it is not worth sharing them..

Jiahui17, 2024-10-20:
In any case, I will close the issue here :)

closed 2024-10-20 by Jiahui17