EPFL-LAP/dynamatic #173: [Speculation] Ready bit of the speculator's signals are always 1, causing bug when predecessor stalls

shundroid, 2024-10-21:
In Haoran's implementation, the two ready bits of the predictor (internal module in the speculator) are always set to 1, which makes the speculator's all ready signals always 1.
It may cause a bug in the following situation.

In the circuit of the figure below, when the data channel of the cond_br stalls, the output data of the speculator also stalls. If the two inputs of the speculator don't stall, it loses some control tokens that should be passed to predecessor units and causes the matching problem.

[image]

shundroid, 2024-10-21:
To fix this, I made the predictor's ready bits depend on the ready signal from the internal predecessor unit.
[image]

Note that the predictor is connected to the speculator's inputs like this way:
[image]

Fixed in #172

murphe67, 2025-03-12:
Iis this fixed with the new backend?

shundroid, 2025-03-12:
yes

closed 2025-03-12 by shundroid