EPFL-LAP/dynamatic #174: [Speculation] SaveCommit units don't handle NoCmp signal correctly

shundroid, 2024-10-21:
Background (from Haoran's thesis):
- When the speculator receives real data before it speculates, the speculator sends a NoCmp signal to SaveCommit units through SCSaveCtrl channel.
- When SaveCommit units receive a NoCmp signal, it should perform `resend`. That is, SaveCommit units should send out the oldest data token and drop it from the internal buffer.
  - Note that SaveCommit units also perform `resend` when the speculator finds out its oldest prediction is wrong (imagine that Save units need to resend non-spec tokens in this case).

In Haoran's implementation, NoCmp signal seems to be handled incorrectly and causes a bug that SaveCommit units cannot accept any control signals after NoCmp signal is sent to them.

In detail as follows:
1. NoCmpEn signal is sent to the save-commit unit. That is, pValidArray(1) = '1' and controlInArray(0) = "100".
2. Then, NoCmpEn becomes 1 and other something-En signals are 0 (en_proc process in the save_commit_control arch)
3. readyArray(1) becomes 0 because NoCmpEn is not handled in the ready_proc process
4. It never accepts the NoCmpEn signal and stalls permanently

shundroid, 2024-10-21:
Apart from the thesis, NoCmpEn signal is defined separately from ResendEn signal and NoCmpEn signal is not handled correctly.
I guess that this NoCmpEn signal is probably old and unnecessary. I removed NoCmpEn signal and replaced it with ResendEn signal and it worked.

Fixed in #172

murphe67, 2025-03-12:
Is this fixed with the new backend?

shundroid, 2025-03-12:
Yes, it's not made as a PR or merged though

shundroid, 2025-03-12:
But currently speculation RTL code is not included in the first place, so maybe it shouldn't be an issue for this repo