EPFL-LAP/dynamatic #183: Compilation Error with Resource Sharing: C Program with Function Calls

udareechk, 2024-11-06:
Hi, I'm trying to compile a C application with a top function that includes another function call within it. 
I am getting the following error during compilation with resource sharing.

[image]

May I know whether there is a fix or a work-around for this issue?

Thanks!

Jiahui17, 2024-11-06:
@udareechk thanks for reporting the issue! Please also attach the source file you used here.

Also, external function calls are not yet supported (but let's see the kernel first)

pcineverdies, 2025-01-24:
I'd close this issue as function calls are not supported yet. 
Feel free to reopen it if the problem is strictly related to the sharing mechanism!

closed 2025-01-24 by pcineverdies