EPFL-LAP/dynamatic #195: **Ideas for The Dynamatic Logo**

Jiahui17, 2024-11-26:
A project logo increases recognition (e.g., CIRCT, MLIR, etc); Dynamatic has been around for many years but doesn't have an official logo. I came up with 5 drafts to start the discussion:

[image]

paolo-ienne, 2024-11-26:
Fantastic idea, @Jiahui17!   🥇

ETHZ may not be that happy with this EPFL-centric view--or would they be ok....  ;^)

Should we not go for something less textual? We should try one of those AI things. I would see well something based on a whirlwind or so. Maybe we can still use the EPFL red (well it was some fancy red but now it is a simple #FF0000)....

paolo-ienne, 2024-11-26:
These are all copyrighted stuff from Adobe Stock, but would something like this be nice? In #FF0000 red?
[image]
[image]
[image]
[image]
[image]
[image]

paolo-ienne, 2024-11-27:
Cute!