EPFL-LAP/dynamatic #196: [TypeSystem] Supporting UnbundleOp/BundleOp in HW/RTL

Jiahui17, 2024-11-27:
We plan to use [the handshake type system](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/TypeSystem.md) to support a variety of optimization strategies like multi-threaded or speculative execution, and channel rigidification.

As a first step to support them in HW dialect and RTL lowering, we propose the following changes:

- Discrimination logic in `HandshakeToHW` pass for UnbundleOp/BundleOp.
- `tools/backend/unbundle-generator` that generates VHDL/Verilog implementation according to the complex types handled by the unbundle op. 
- `tools/backend/bundle-generator` that generates VHDL/Verilog implementation according to the complex types handled by the bundle op.

What do you think? I also wanna ask the speculation experts (@shundroid, @murphe67 ): have you already implemented something similar or do you need it at all?

murphe67, 2024-11-27:
We are just moving to create a proper backend for speculation now, @shundroid is investigating exactly what we need. For me, I think we will avoid using bundle-unbundle to begin with, instead creating two new operations and then handwrite the VHDL that does the required rewiring. Once we have that working with the new type system, it would be more elegant to then use the bundle-unbundle system, but we haven't really looked at it yet.

murphe67, 2024-11-27:
Explicitly writing that down though, it does make me wonder what the benefit of having the rewiring information in the IR is- would it be scalable to create a custom operation each time we want to build a new channel out of the wires of an old channel, and the simply write an rtl generator it? Mapping operation to RTL seems easier than mapping MLIR code to RTL

rpirayadi, 2024-11-27:
I just wanted to mention that what I am currently doing includes extracting dataless signals out of channels so I can synchronize memory accesses using them. To do so, I am using bundle/unbundle, so, I would benefit from this change!

murphe67, 2024-11-27:
Yes if we take this as an example (maybe I have the details wrong but) 
[image]
If we just add an extract handshake op, and keep the specifics of what happens inside the op out of the IR. This means we can write a (probably simple) RTL generater. Would this be easier than trying to convert the MLIR unbundle-bundle code to RTL?

shundroid, 2024-11-27:
For speculation, we made some breaking changes to the current backend just to experiment with how the frontend works, so we hadn’t focused on the type system until now. As Emmet mentioned, we decided today to start implementing the backend using the type system, so I’ve just started looking into it.

We were initially planning to handle the entire rewiring directly in the RTL code without relying on IR features like bundle/unbundle, but it’s still a bit abstract. I’ll discuss it further with Emmet.

Jiahui17, 2024-11-27:
@rpirayadi and our use cases of the type system don't involve functional changes so I guess bundle and unbundle make sense here (they are just syntax sugars for packing/unpacking signals); but the speculation / tagging stuff is functional (adding information), so maybe adding new unit is more suitable

murphe67, 2024-11-27:
Ours is also syntactic sugar, which is my point- how much work is building a generalizable conversion if all of these units end up being 2 lines of RTL. If it's a lot of work, maybe it is not worth it.

Jiahui17, 2024-11-28:
> Ours is also syntactic sugar, which is my point- how much work is building a generalizable conversion if all of these units end up being 2 lines of RTL. If it's a lot of work, maybe it is not worth it.

I am more convinced now. For instance for setting the ready signal to constant we could just invent a new op called "Rigidifier"

[image]

It simply cuts the ready and bypasses valid/data.

One limitation might be how to represent merged valid signals (one optimization that we tried in FPGA23).

lucas-rami, 2024-11-29:
Thanks for bringing up the topic and for the interesting thoughts from everyone. I think it should be pretty easy to add full HW/RTL support for bundle/unbundle, in fact I don't mind giving this a few hours this weekend.

My opinion on the bundle/unbundle VS new operations discussion is the following. When possible, I think people should use bundle/unbundle instead of adding a new operation that is just a "black box" for a well-defined set of existing dataflow components (I'm including bundle/unbundle as "existing" here, even if they are more akin to "MLIR-level type manipulation operations" rather than functional components). In @Jiahui17's last message for example, I would create the IR representing the graph on the left instead of the one on the right.

My rationale is that adding new operations yields scalability issues. Some passes need to iterate over the whole CFDG and to some extent need to know what each operation "does", even if only superficially. Every time we add an operation we have to teach these passes what to do upon encountering them (in addition to adding lowering to HW and RTL implementations). Of course, sometimes we have to add new operations because no combination of the existing ones can express the same behavior, and in that case we have no choice.

A legitimate concern with my preferred solution might be that, taking the `Rigidifier` as an example again, out-of-the-box it would be harder to iterate over all `Rigidifer`s in the IR than if we just had a dedicated operation (same argument applies to creating/erasing `Rigidifier`s). That is true; however it is also easy to write a small reusable piece of code that would identify the IR structure on the left of the picture and return some object containing the structure's "operand" and "result", as in the IR structure on the right of the picture.

paolo-ienne, 2024-11-29:
For what it matters, I tend to strongly agree with @lucas-rami: a component should only be added if (i) it cannot be expressed as a composition of more elementary components or, perhaps, if (ii) its direct implementation is much better than the implementation of a composition of elementary components (inclusive of natural logic optimizations). The implementation of generators of arbitrary bundle/unbundle operators does not seem at all hard to me--but maybe I did not think it through....

murphe67, 2024-11-29:
At the in-person discussion we settled on passes requiring bundle-unbundle being separated in 2 halves: the first pass would be pre-buffering, and would insert a black-box operation, in order to maintain the buffering assumption that each variable is a channel. The second pass would be post-buffering, and would convert the operation to explicit bundle-unbundle. 

Depending on whether the whole CDFGs passes are pre-buffering or post-buffering, I would say we possibly have the scalability issues anyway, unless we want to revisit that decision.

For me, the bundle-unbundle solution only makes sense to use if there is optimization done based on that information, but the post-buffering limitation means there is probably little room to optimize. If we are adding this information to the IR purely to generate the RTL from it, we are over-complicating the RTL generation for no reason. 

There is a possible argument that even though we may not optimize based on it now, we may do so later, but it still seems unnecessary to me. 

If we can add this information pre-buffering and extract timing information from it, we have an obvious use case, but are we able to do that without breaking the buffering algorithms?

closed 2025-03-12 by murphe67