EPFL-LAP/dynamatic #199: [LSQ] Throughput problem in LSQ / LSQStore

Jiahui17, 2024-12-05:
(This issue is related to #191 and #193)

Pinging the LSQ experts: @lucas-rami @paolo-ienne @rpirayadi @lana555 @qianxu1998 

# Background

In #193 we decided to unify LSQLoad/MCLoad and LSQStore/MCStore. While the logic behind the load ports is the same for LSQ/MC, it is not the same for the store ports.

Consider the `histogram` example in commit [815081c](https://github.com/Jiahui17/dynamatic/commit/815081c554b974debfefba96177a949e18e8c844). After buffering we get the following circuit:

[image]

There is a pair of reconvergent paths from fork -> ST (highlighted in colors) where the left side has 9 cycles latency and the right side has 0 cycle latency. 

This decision is **correct** since store data and address are allowed to arrive out-of-sync when connected to LSQ (in other words, the store entry arguments can be given at different times). In the performance model, I think the LSQ store port is treated as two separate sinks.

# Issue

Yet, after unifying the HDL implementations of LSQStore and MCStore, this model is no longer valid. Now, this store port acts as a join, therefore the reconvergent paths in the example are not latency-balanced and the II is penalized.

# Discussion

Some solutions:

## 1. Does allowing supply arguments at different times bring performance benefits?

One possible solution is to keep the store port as it is, and always model it as a join. I couldn't think of a case where this penalizes performance.

## 2. Split the HDL implementation of the store ports

We partially roll-back #193 and split the store ports.

paolo-ienne, 2024-12-05:
Ouch! Thanks @Jiahui17 for the simple and clear problem statement! There is always something weird one does not think of....   :-(

My understanding, from other discussions, is that even if the store is unified, there is always a nonambiguous and possibly explicit way to tell if the port will require an LSQ. My understanding (expressed in [this comment](https://github.com/EPFL-LAP/dynamatic/issues/185#issuecomment-2500280769) but never confirmed by anyone) is that dependency annotations evolve (and probably decrease monotonically) in the flow--either because someone realizes that they are already enforced (FPT'19) or because someone enforces them by adding new components and paths to the circuit (@rpirayadi's future stuff). If some conditions are met (maybe as simple as "there are no dependency annotations", but maybe it is more complex), then the LSQ is unneeded. Am I fundamentally mistaken?

If not, whatever part of the code builds the timing model, should call a function `isThisStoreAnLsqStore` every time it meets a store and act accordingly (much as today the same code handles differently LSQStore vs. MCStore). Or?!

Jiahui17, 2024-12-05:
I agree with the flow (always call `isThisStoreAnLSQStore` to decide which timing model / HDL implementation to use). So should we go with this solution?

---

> My understanding, from other discussions, is that even if the store is unified, there is always a nonambiguous and possibly explicit way to tell if the port will require an LSQ. My understanding (expressed in https://github.com/EPFL-LAP/dynamatic/issues/185#issuecomment-2500280769 but never confirmed by anyone) is that dependency annotations evolve (and probably decrease monotonically) in the flow--either because someone realizes that they are already enforced (FPT'19) or because someone enforces them by adding new components and paths to the circuit (@rpirayadi's future stuff). If some conditions are met (maybe as simple as "there are no dependency annotations", but maybe it is more complex), then the LSQ is unneeded. Am I fundamentally mistaken?

I agree with the dependency annotation system and I just want to add one small thing that we might want to remember:

One effect of buffering is to decide the time when the load/store requests reach the memory interfaces (i.e., placing buffers "schedules" operations, which might honor the dependency), this might remove the need for an LSQ.

paolo-ienne, 2024-12-05:
Is there some documentation anywhere on these dependency attributes? If not, it would be nice to have, among other things to confirm or refute my assumptions.

Jiahui17, 2024-12-06:
> Is there some documentation anywhere on these dependency attributes?

I don't think we have it (also not sure how this is managed in the memory analysis pass).

paolo-ienne, 2024-12-06:
Maybe @rpirayadi could volunteer?...  ;^)   I think he plans to be quite active on that front.

lucas-rami, 2024-12-08:
> My understanding, from other discussions, is that even if the store is unified, there is always a nonambiguous and possibly explicit way to tell if the port will require an LSQ. My understanding [...]

At this point dependency attributes are just carried down the flow without modifications from the affine IR level. The motivation for this design was to allow every pass at any point in the flow to identify all dependencies by just looking at those attributes, which worked fine till now. For example, the current LSQ analysis/minimization pass reads them to identify all RAW/WAR/WAW dependencies in the kernel. I think here may be an argument for removing those as they become enforced through various methods (e.g., LSQ, added paths between ports), but in that case the LSQ analysis pass will need to be slightly updated.

> If not, whatever part of the code builds the timing model, should call a function isThisStoreAnLsqStore every time it meets a store and act accordingly (much as today the same code handles differently LSQStore vs. MCStore). Or?!

Yes, it is basically [this function](https://github.com/EPFL-LAP/dynamatic/blob/788a3b353bdb06199df838e9d61e997226d7fcc0/include/dynamatic/Dialect/Handshake/HandshakeOps.h#L572-L576).

> Is there some documentation anywhere on these dependency attributes? If not, it would be nice to have, among other things to confirm or refute my assumptions.

Yes, [here](https://github.com/EPFL-LAP/dynamatic/blob/788a3b353bdb06199df838e9d61e997226d7fcc0/include/dynamatic/Dialect/Handshake/HandshakeAttributes.td#L154-L251).

paolo-ienne, 2024-12-08:
Thanks @lucas-rami for the clarifications.

> At this point dependency attributes are just carried down the flow without modifications from the affine IR level. The motivation for this design was to allow every pass at any point in the flow to identify all dependencies by just looking at those attributes, [...] I think here may be an argument for removing those as they become enforced through various methods (e.g., LSQ, added paths between ports), but in that case the LSQ analysis pass will need to be slightly updated.

I have some sympathy for the idea of leaving all dependency annotations around--who knows who might want to know something about that, at some point down the flow.... It is less clear to me how trivial it is to understand the _active_ ones at a particular moment, after further analyses, circuit modifications, etc., and I suspect totally trivial it is not (the FPT'19 pass is an example and I guess we would not define it completely trivial). So, maybe a good idea could be to keep all dependency information there but have a flag each (_active_) that is always initialized to _true_ and gets converted to _false_ if a pass either discovers that the dependency is naturally honoured (e.g., FPT'19) or adds circuitry to the same effect (what @rpirayadi plans). What is marked _active_ is what should be used by `HandshakeReplaceMemoryInterfaces`, if I understand things right.

lucas-rami, 2024-12-08:
> So, maybe a good idea could be to keep all dependency information there but have a flag each (active) that is always initialized to true and gets converted to false if a pass either discovers that [...]

I think it's a good compromise!

> What is marked active is what should be used by `HandshakeReplaceMemoryInterfaces`, if I understand things right.

Just to make sure we agree, the attribute used there is slightly different. `HandshakeReplaceMemoryInterfaces` does not look at dependencies directly but instead at the `MemInterfaceAttr` attribute (as opposed to `MemDependenceAttr`) on memory ports, which says explicitly which port should connect to which kind of interface. The FPT'19 pass basically transforms the existing `MemDependenceAttr`s into `MemInterfaceAttr`s for the `HandshakeReplaceMemoryInterfaces` pass, which runs after.

rpirayadi, 2024-12-08:
> So, maybe a good idea could be to keep all dependency information there but have a flag each (active) that is always initialized to true and gets converted to false if a pass either discovers that [...]

I also agree with this idea. And, I can start implementing it. 

> Just to make sure we agree, the attribute used there is slightly different. HandshakeReplaceMemoryInterfaces does not look at dependencies directly but instead at the MemInterfaceAttr attribute (as [...]

In order to solve this issue, we can introduce a new pass that simply checks every memory port and if there is one that all of its dependencies are inactive, then change the `MemInterfaceAttr` from LSQ to MC. 

Another solution could be that after changing how LSQ analysis pass works, move this logic into `HandshakeReplaceMemoryInterfaces`. That way the LSQ analysis pass will just inactive some dependencies and rely on `HandshakeReplaceMemoryInterfaces`. However, this may need some more serious modifications on LSQ analysis pass.

lucas-rami, 2024-12-08:
> In order to solve this issue, we can introduce a new pass that simply checks every memory port and if there is one that all of its dependencies are inactive, then change the MemInterfaceAttr from LSQ to MC.

This is not an issue to me, just good separation of concern. One attribute expresses a dependency between two ports, the other the interface (if any) we want to (re)connect the port to.

In my mind `HandshakeReplaceMemoryInterfaces` should be as dumb as possible. Read `MemInterfaceAttr`s, replace memories according to them (with some sanity verifications at most), done. I'm against it taking any "decision" of its own. If some present or future memory analysis pass realizes that all of a port's dependencies are somehow already handled, then it can directly "set the `MemInterfaceAttr` to MC" itself, which is what FPT'19 does in some cases. I don't see a justification for a separate pass to do this simple thing.

paolo-ienne, 2024-12-08:
I think we violently agree....  :-)

- Initial analysis places all dependencies there and all active.
- Anyone, by whatever means and for whatever reason, may assert that some dependencies are no longer to be considered and deactivates them (with the evident responsibility of maintaining correctness).
- Whenever this happens, for the correctness/completeness argument mentioned elsewhere by @lucas-rami, one calls a function that
  - sets `MemInterfaceAttr` to the appropriate value, given the active dependencies, and 
  - calls `HandshakeReplaceMemoryInterfaces` to implement such decision in a correct circuitry.

I agree with @lucas-rami that this seems a clean separation of concerns.

It seems to me that this directly addresses the original issue that this comment belongs to: call the function that sets `MemInterfaceAttr` before buffering and use `MemInterfaceAttr` to distinguish LSQ ports from MC ports.

lana555, 2025-01-20:
@rpirayadi @paolo-ienne I think the discussion above is not really capturing the problem: having a join in the lsq store limits the throughput of the circuit to (essentially) the latency it takes to compute the store data. Isn't this the source of the II issues we discussed? Can we just remove the join from lsq store and problem solved? I understand the beauty of having a single store port for the MC and LSQ, but the price is not worth it...

paolo-ienne, 2025-01-23:
Thanks @lana555 for attracting back our attention to this one. Indeed, I think you are perfectly right and this bug has never been addressed and the subsequent discussion (important to me to understand how Dynamatic works, unfortunately obscured the original problem that went forgotten!  :cry:

I have added the `bug` label to try to avoid this to be lost again.

I believe the proper solution is still what I suggested [here](https://github.com/EPFL-LAP/dynamatic/issues/199#issuecomment-2520615075):
> If not, whatever part of the code builds the timing model, should call a function `isThisStoreAnLsqStore` every time it meets a store and act accordingly (much as today the same code handles differently `LSQStore` vs. `MCStore`). Or?!

Probably the implementation details are imprecise or wrong but the spirit is correct, I think: the point is that building the model and interrogating the database about the timing characteristics of the store, the fact that a port will be implemented as LSQ or MC should be used rather than ignored (as it appears the case to be today).

1 - Does anyone disagree with this solution? Shall we go ahead?

2 - Can someone implement it? @Jiahui17 and @rpirayadi may be the best ones to implement this, perhaps cooperatively?!

Jiahui17, 2025-01-23:
@paolo-ienne @lana555 

> Probably the implementation details are imprecise or wrong but the spirit is correct, I think: the point is that building the model and interrogating the database about the timing characteristics of [...]

I agree (we evaluate `isThisStoreAnLsqStore` and do things differently). 

@rpirayadi 

On the implementation side, how about the following:

1. **HDL side**: Keep the storePort the same for LSQ/MC (both without a join inside) and add join logic to the memory controller. 
2. **Buffer side**: No changes now (we already have `isThisStoreAnLsqStore` and the throughput model is the same for LSQ/MC).
3. **Memory dependence side**: I am not an expert on this part but I guess everything is decoupled from the first two points.

What do you think?

pcineverdies, 2025-01-23:
> About this, I would argue that keeping the two different RTL implementations, and then deciding which one to use, is more convenient. However, if there are advantages to doing it this way, we can consider this.

However, if it's just a matter of **always** having a join for the _address_ and _data_ (we can discuss if this is the case or not) inputs of a store, I don't really see the disadvantages in moving the `join` to the memory controller, as Jiahui suggested! This is a very clean solution and it avoids you (or whoever involved) the hassle of modifying too many things at too many levels of abstraction.

PR #246 (2025-01-23): [Verilog][VHDL] Removing Join from HDL Implementation of Store.{vhd,v}
Fix the throughput problem (see #199) related to the store port and LSQ. Now the memory interface (LSQ|MC) synchronizes the data and address tokens.

Changes:
- [store.{vhd,v}]: Remove the join (excessive synchronization for LSQ).
- [mem_controller_loadless.{vhd,v}]: Add join to make sure that store is issued when both data and address are ready.

fixes #199
commit 5d58dc0c (2025-01-23): [Verilog][VHDL] Removing Join from HDL Implementation of Store.{vhd,v} (#246)

Fix the throughput problem (see #199) related to the store port and LSQ.
Now the memory interface (LSQ|MC) synchronizes the data and address
tokens.

Changes:
- [store.{vhd,v}]: Remove the join (excessive synchronization for LSQ).
- [mem_controller_loadless.{vhd,v}]: Add join to make sure that store is
issued when both data and address are ready.

fixes #199
closed 2025-01-23 by Jiahui17