EPFL-LAP/dynamatic #201: [integration-test] cnn benchmark

AyaElAkhras, 2024-12-06:
The [cnn](https://github.com/EPFL-LAP/dynamatic/tree/main/integration-test/cnn) benchmark is currently deadlocking in the `main` branch using the simple buffer placement (`on-merges`).

I did not manage to debug myself much, but playing with the `#define`s, I realized that the benchmark works fine if `kKernel` is set to 1 or to 2, which might suggest that it is not actually deadlocking but just taking tooooo long to complete execution.

Two questions:
- Did it successfully complete execution with anyone?
- If no for the previous question, I wonder if we can change `kKernel` to 1 or 2, which would mean supporting convolution with a kernel size of 1x1 or 2x2 instead of 4x4?

Any thoughts?

paolo-ienne, 2024-12-06:
> - I wonder if we can change kKernel to 1 or 2, which would mean supporting convolution with a kernel size of 1x1 or 2x2 instead of 4x4?

I would not think that changing a benchmark because of a bug or a serious compiler limitation is a good idea, especially if it denatures the benchmark (what exactly is a 1×1 convolution?!).

AyaElAkhras, 2024-12-06:
> I would not think that changing a benchmark because of a bug or a serious compiler limitation is a good idea, especially if it denatures the benchmark (what exactly is a 1×1 convolution?!).
>
I agree with you in general, but the point here is that, from my side, this benchmark never completes simulation, which could be either because there is a bug that makes it deadlock, or it is really requiring a very long simulation time. It has a loop-nest with depth 6 so it is likely that it demands a very long simulation time. To understand what is going on, I played with the different loop bounds decreasing their value, and found out that changing `kKernel`, which is the loop bound of the two innermost loops in the nest, from 4 to 1 makes it finish in a few minutes, and from 4 to 2 makes it finish in ~1 hour and from 4 to 3 makes it finish in ~1h30. This made me conclude that it is likely that it is not deadlocking and just requiring a lot of simulation time.

My suggestion of changing the `kKernel` to 1 or 2 was mainly for practical reasons--all of our benchmarks typically take up to a few minutes to complete simulation but never hours. A 1x1 convolution exists and is useful https://medium.com/analytics-vidhya/talented-mr-1x1-comprehensive-look-at-1x1-convolution-in-deep-learning-f6b355825578 so it does not really denature the benchmark. 

Maybe a compromise would be to add a new cnn-1x1 benchmark, instead of modifying the existing benchmark? And, of course, anyone could play with the parameters to try out larger kernel sizes, knowing that the simulation time is likely to be significantly long.

Did anyone experience a faster simulation time than what I get?

Jiahui17, 2024-12-06:
> Did it successfully complete execution with anyone?
Yes, it finishes as it is now, but also I have no clue why the simulation is that slow...

Have you tried also the LU decomposition one? I remember that one is also quite slow despite operating on 3x3 matrices...

paolo-ienne, 2024-12-06:
Ok, this must be AI lingo--in my book 1×1 convolution is a multiplication by a constant....   ;^)