EPFL-LAP/dynamatic #204: [Handshake] Should verifyIndexWideEnough be performed in MergeLikeOpInterface?

shundroid, 2024-12-07:
`verifyIndexWideEnough` is a function that checks whether the `index` bitWidth is sufficient to cover all inputs of a merge-like operation. Currently, it is called in the `verify` methods of the following operations:

- `MuxOp::verify`
- `ControlMergeOp::verify`

However, there are some issues with this approach:

- The check does not cover `MergeOp`.
- The `MergeLikeOpInterface` verifies other conditions separately, and this check should also be included in the interface.
  - **Note**: `MergeLikeOpInterface` is currently implemented by `MuxOp`, `ControlMergeOp`, and `MergeOp`.

@lucas-rami What do you think?

Jiahui17, 2024-12-07:
I don't think a merge has an index though :D

shundroid, 2024-12-07:
Wow that's true, sorry 😅

closed 2024-12-07 by shundroid