EPFL-LAP/dynamatic #208: [Buffers] Integrating New Buffer Algorithms and Documentation on Buffering Passes

Jiahui17, 2024-12-10:
# Introduction

Buffer placement is the core of the entire Dynamatic HLS tool. Its role is similar to scheduling in Vitis HLS. Over the years, many buffer placement algorithms have been proposed; here is a short list:

- **Pre-characterized delay**: Using pre-characterized delay to estimate critical paths.
- **Technology-mapping-aware**: Using subject graph to estimate critical paths after mapping.
- **Stall-eliminating**: Eliminating stalls as much as possible for circuit redundancy removal.
- **Cost aware**: Some buffer types are cheaper and can be chosen strategically. 

(WIP: the papers)

Yet, the current implementation is designed primarily with the first two papers in mind. In this issue, we classify the constraints of different buffer placement formulations and propose a plan for integrating the newer algorithms into Dynamatic. Here, we are mainly concerned with the algorithms based on modeling the problem as a MILP.

# Overview

A buffer placement algorithm has three main goals:

- **Timing closure.** The produced circuit should respect a certain clock frequency target.
- **High throughput.** A higher throughput is always desired.
- **Lower area.** Practical systems are usually resource-constrained; a higher area often means higher power.

Following this classification, we can visualize the relationship between 5 different approaches as the following:

[image]

For each of the three goals, we have to choose one and exactly one set of constraints (i.e., one big box) or one objective. Also, although the graph structure is nearly identical across all 5 algorithms (i.e., CFDFCs), the constraint formulation are fundamentally different.

# Implementation

## What to change in the existing codebase?

TL;DR: not much

(WIP)

## What to add?

(WIP)

AyaElAkhras, 2024-12-16:
Thanks @Jiahui17 for the nice presentation of the prior work and the limitations!

> - Technology-mapping-aware: Using subject graph to estimate critical paths after mapping.
> - Cost aware: Some buffer types are cheaper and can be chosen strategically.
>
Is there an already ongoing progress in these two directions? Is the Cost-aware row in the above table corresponding to some existing paper?

Jiahui17, 2024-12-16:
@AyaElAkhras 

Yes! We are working on both; we had a student project on the cost-aware one

Jiahui17, 2025-01-07:
closing it now for #228

closed 2025-01-07 by Jiahui17