EPFL-LAP/dynamatic #209: [Question] Why ccache and lld are disabled by default, unless --disable-build-opt or -o is provided?

shundroid, 2024-12-10:
While I looked into why linking on my laptop is so slow, I discovered that lld is disabled by default after this commit:

https://github.com/EPFL-LAP/dynamatic/commit/9410a87912f789dac6c766102be6ec5735a04075#diff-4d2a8eefdf2a9783512a35da4dc7676a66404b6f3826a8af9aad038722da6823R122

Now `ccache` and `lld` can only be used if we specify `--disable-build-opt` or `-o` in the argument for `./build.sh`.

https://github.com/EPFL-LAP/dynamatic/blob/788a3b353bdb06199df838e9d61e997226d7fcc0/build.sh#L154-L157

Was this change intentional? For me, using `ccache` and `lld` feels like build optimization rather than the opposite.

Jiahui17, 2024-12-10:
1. Yeah from the name of the flag it makes more sense to turn them off when the flag is selected.
2. I think lld and ccache are enabled by default so this flag doesn't do anything

shundroid, 2024-12-10:
The system linker is used by default. On my laptop, it was `ld`, not `lld`. This is why the linking was slow on my laptop.
CCache is off by default.

https://llvm.org/docs/CMake.html

Thus I think we should update the shell script.

lucas-rami, 2024-12-11:
Simple goof on my side, sorry. Thanks for the catch! Will fix this soon.

commit 2da8f40e (2024-12-15): [Build] Use `ccache` and `lld` by default

This should have been the default behavior in the first place.

Fixes #209.
closed 2024-12-15 by lucas-rami