EPFL-LAP/dynamatic #214: [TypeSystem] Memory Controller Type Interface

murphe67, 2024-12-13:
Hi everyone- @lucas-rami @AyaElAkhras @shundroid 

Just thought I should check in with you guys about the side channels (to memory controller etc.) out of load and store units.
If we look at the current implementation, these channels can be tagged. Is this what we want?

For speculation, it seems to us that the memory controller does not need to know whether a load is speculative or not, and so channels to memory controllers will not have non-spec/spec signals.

Reading the [spec](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/CircuitInterface.md#start-to-end-path):
[image]
And looking at the memory controller RTL, it is specific to a 2-port BRAM.

Do we consider this an internal memory controller, one of several variants we may eventually have, or is this an external memory controller that should not receive tag information?

For Aya's out-of-order loads, is the logic inside of the load unit, or inside of the memory controller? 

In terms of the actional decision: Should the MLIR type verification insist that data and address channels into memory controllers be tagless? Are these channels leaving Dynamatic?

shundroid, 2024-12-13:
I noticed in the diagram above that there's a control path connecting the internal circuit logic to the internal memory controller.

In #197, I marked `mem_start` and `mem_end` as tagless, which I assume everyone agrees with. However, I also marked the control flow decision signal (`ctrl_end`) as tagless, and I'm unsure if this was the right choice.

For speculation, this seems fine, as only a (spec)tagless control signal should indicate the end of the control flow to the memory controller.

lucas-rami, 2024-12-15:
Thanks for bringing up the topic!

>  And looking at the memory controller RTL, it is specific to a 2-port BRAM.

Yep that's very bad, it should be an elastic (latency-insensitive) interface to the outside world, which can then be converted to a 2-port BRAM if that is the kind of memory we are in fact talking to.

> Do we consider this an internal memory controller, one of several variants we may eventually have, or is this an external memory controller that should not receive tag information?

I think the current interfaces' bad design that I mention above makes this a bit blurry, but to me the current MC/LSQ should be "internal" interfaces that have elastic interfaces to whatever "external" interfaces one wants (e.g., BRAM, AXI).

> In terms of the actional decision: Should the MLIR type verification insist that data and address channels into memory controllers be tagless? 

Yes, I would say that our current memory interfaces (`handshake::MemoryControllerOp` and `handshake::LSQOp`) should only accept data/address channels without extra signals because

1. their current implementation currently does not know how to handle them,
2. the semantics of these extra signals would anyway be undefined at this point, and
3. these semantics would be context-specific, so it would be hard/impossible to come up with a generic op interface/verifier for those.

Consequently, if somebody needs an interface that understand some sort of additional signals they will have to make a new one with the specific semantics/connectivity they care about. I think any risk for heavy code duplication can be alleviated with some smart design decisions.

> Are these channels leaving Dynamatic?

I think they could depending on the specific use case.

AyaElAkhras, 2024-12-15:
Thanks for actively working on this! :)

> For Aya's out-of-order loads, is the logic inside of the load unit, or inside of the memory controller?
>
Currently, we do not have an out-of-order load in the sense that the load operation and the memory controller are entirely in order (i.e., they return responses in the same order as that of the requests) and identical to the old designs in legacy Dynamatic. We only have tagged loads that have additional `tagIn` and `tagOut` ports only for the sake of propagating the tag to work with the rest of the circuit that have other operations supporting out-of-order execution.

> Consequently, if somebody needs an interface that understand some sort of additional signals they will have to make a new one with the specific semantics/connectivity they care about.
>
I agree!

> In terms of the actional decision: Should the MLIR type verification insist that data and address channels into memory controllers be tagless? 
>
The current design of a memory controller does not allow out-of-order execution and so extending the interface with tags while maintaining the current design is pointless; therefore, for this design, we can force the channels to be tagless. 

I think the decision depends on how we may want to support "fancier" memory systems in the future. For instance, if we want to support memories that interface with nonblocking caches, do we change the MLIR type verification? Or, do we define new operations with a new type? I have not thought it long enough and so I do not have a clear view, but just want to make sure that any decision we take now does not limit us in the future.

murphe67, 2024-12-15:
> For instance, if we want to support memories that interface with nonblocking caches, do we change the MLIR type verification? Or, do we define new operations with a new type? I have not thought it [...]

Changing the type verification in the future would be simple, so we are not closing off any options, but it is a nice sanity check to force them to be tagless while we intend them to be tagless.

> Yes, I would say that our current memory interfaces (handshake::MemoryControllerOp and handshake::LSQOp) should only accept data/address channels without extra signals because

    1. their current implementation currently does not know how to handle them,
    2. the semantics of these extra signals would anyway be undefined at this point, and
    3. these semantics would be context-specific, so it would be hard/impossible to come up with a generic op interface/verifier for those.

Perfect, sounds good!

closed 2024-12-15 by murphe67