EPFL-LAP/dynamatic #219: [Formal] Formal Properties in Dynamatic/MLIR

Jiahui17, 2024-12-16:
# [Formal] Formal Properties in MLIR

## Introduction

HDL languages commonly have extensions for verification support. For instance, SystemVerilog uses SystemVerilog Assertion (SVA) and VHDL uses property specification language (PSL). Open-source synthesis tools like Yosys also develop their language extensions to support assertions in Verilog.

In brief, verification has two main components:
1. **Assertions**: the verification goals. 
2. **Assumptions**: restricting the behaviors seen by the verifier.

Yet, in Dozens of MLIR dialects, only `cf` dialects define an `assertion` operation. This issue discusses some challenges and solutions.

## Challenges of Encoding Properties

**1. Naming of internal signals/states.** In theory, the specification should only concern the IO behavior. Yet, many verification techniques fundamentally need to annotate "invariants" over some internal states of software or hardware modules (e.g., loop invariants).

**2. Embedding properties in dialect.** We might want to embed properties directly inside the `handshake.FuncOp` or `hw.ModelOp`. As an example of embedding properties: CIRCT people develop their own dialects to encode properties
- The `verif` dialect for representing assertions, assumptions, tester modules, and model checking instances.
- The `ltl` dialect for representing temporal specification.

**3. Dialect conversion for properties.** Handshake is a decent abstraction level for analysis and deriving properties. Therefore, it might be attempting to write the assertions/assumptions in the handshake level. Yet, additional conversion logic might be needed for the properties.

**4. Type system.** The type system provides a good abstraction for dialect transformation and conversion. Yet, an assertion/assumption is commonly low level, which is incompatible with the type system (e.g., you need bundle/unbundle to expose the signals first, then right the assertion/assumption over them).

## Hypothetical Examples

### Property Embedding in Handshake Level

Suppose that we have the following subcircuit `buffer -> fork`. 

```mlir
%bufOut    = handshake.buffer [2] %mergeOut : !handshake.channel<i32>
%forkOut:2 = handshake.fork   [3] %bufOut   : !handshake.channel<i32>
```

We introduce `handshake.assert` operation. We want to verify that whenever the buffer has valid data, the fork is always ready to accept it. Yet, we cannot formulate the `i1` handshake signals. So the only workaround is to use unbundle/bundle:

```mlir
// Expose the handshake signals
%bufOut                  = handshake.buffer [2] %mergeOut              : !handshake.channel<i32>
%bufOutData, %bufOutCtrl = handshake.unbundle %bufOut                  : i32, !handshake.control<>
%bufOutValid             = handshake.unbundle %bufOutCtrl, %ready      : i32
%forkInCtrl, %ready      = handshake.bundle %bufOutValid               : !handshake.control<>, i1
%forkIn                  = handshake.bundle %bufOutData, %forkInCtrl   : !handshake.channel<i32>
// Property cone
%1                       = arith.implies %bufOutValid, %ready          : i1
// We assert that this value is never zero
%assert_result           = handshake.assert %1                         : i1
```

### Invariant Embedding in Handshake Level

For the moment, let's say the invariants are encoded as `handshake.assume`. We have an assumption that says: if the fork has duplicated a token and has sent it to any of its outputs, then the buffer must have a token.

```mlir
%bufOut    = handshake.buffer [2] %mergeOut : !handshake.channel<i32>
%forkOut:2 = handshake.fork   [3] %bufOut   : !handshake.channel<i32>
handshake.assume ????
```

Unfortunately, now we cannot do this even with the help of the type system, because MLIR does not understand the internal states of operations ("what is the state variable that says the buffer has a token? How to get this value?").

## Possible Solution

Use an ad-hoc format for properties (e.g., custom JSON or DOT files).

## Remarks

_What do you think? Any suggestions?_

gioelegott, 2024-12-17:
I think both solutions, either embedding formal properties in Handshake IR or creating an ad-hoc format, are valid. Maybe another solution we can consider is to generate this properties during the model generation process, since they are going to be used there anyways. We can further discuss this topic at the in-person meeting.

gioelegott, 2025-03-28:
Hi @Jiahui17, I would like to propose the new plan we recently discussed for the formal properties, but I wanted to know what is your current opinion on the approach you explained here. Let me know if you think we should still keep it into consideration, or if we better create a new issue with the new proposal to avoid confusion for future readers.

gioelegott, 2025-03-28:
# Formal properties

Recently we had a meeting on how to represent formal properties and we came up with an alternative approach. The formal verification flow that we envision creates these properties as a table at the Handshake level and propagate them to HW and then to the SMV model. The model checker will then verify them and propagate back the information to Handshake. 

## Downward propagation of properties

The downwards propagation of the properties will look something like this:

[image]

Where the properties are a table-like data-structure that is filled progressively going down the flow. More precisely:

### Handshake
| Channels                      | Property Name                 | Abstract Expression                      | Signal Expression | Tag          | Result |
|--------------------------------|------------------------------|------------------------------------------|------------------|--------------|--------|
| `muli_1(0)`                   | Absence of backpressure      | `muli_1(0).valid -> muli_1(0).ready`    | None             | Optimization | None   |
| `fork_0(0), fork_0(1)`        | Valid equivalence           | `fork_0(0).valid <-> fork_0(1).valid`   | None             | Optimization | None   |
| `oehb_2(0)`                   | New Correctness Check       | `???`                                   | None             | Error        | None   |
| `fork_0(0), fork_0(1), fork_0(2)` | Invariant 1               | `SUM(fork_0(i).sent) < N`               | None             | Invariant    | None   |

### HW

| Channels                      | Property Name                 | Abstract Expression                      | Signal Expression                          | Tag          | Result |
|--------------------------------|------------------------------|------------------------------------------|-------------------------------------------|--------------|--------|
| `muli_1(0)`                   | Absence of backpressure      | `muli_1(0).valid -> muli_1(0).valid`    | `muli_1.result_valid -> muli_1.result_ready` | Optimization | None   |
| `fork_0(0), fork_0(1)`        | Valid equivalence           | `fork_0(0).valid <-> fork_0(1).valid`   | `fork_0.outs_0_valid <-> fork_0.outs_1_valid` | Optimization | None   |
| `oehb_2(0)`                   | New Correctness Check       | `???`                                   | `???`                                     | Error        | None   |
| `fork_0(0), fork_0(1), fork_0(2)` | Invariant 1               | `SUM(fork_0(i).sent) < N`               | `???`                                     | Invariant    | None   |

### Post-verification

| Channels                      | Property Name                 | Abstract Expression                      | Signal Expression                          | Tag          | Result |
|--------------------------------|------------------------------|------------------------------------------|-------------------------------------------|--------------|--------|
| `muli_1(0)`                   | Absence of backpressure      | `muli_1(0).valid -> muli_1(0).valid`    | `muli_1.result_valid -> muli_1.result_ready` | Optimization | True   |
| `fork_0(0), fork_0(1)`        | Valid equivalence           | `fork_0(0).valid <-> fork_0(1).valid`   | `fork_0.outs_0_valid <-> fork_0.outs_1_valid` | Optimization | False   |
| `oehb_2(0)`                   | New Correctness Check       | `???`                                   | `???`                                     | Error        | True   |
| `fork_0(0), fork_0(1), fork_0(2)` | Invariant 1               | `SUM(fork_0(i).sent) < N`               | `???`                                     | Invariant    | None   |

The tag is used to determine what each property is going to be used for. For example a tag can be:
- Optimization: the property is used to determine the applicability of certain optimizations (e.g. rigidification)
- Error: the property is used to check the correctness of the circuit and the compilation/generation process should be stopped if it isn't satisfied
- Warning: the property is used to check the correctness of the circuit and the compilation/generation process continues with a warning if it isn't satisfied
- Invariant: the property is used to speed up the verification process (SMV invariant)

Our main focus will be on the optimization properties, so for now the others will be ignored.

## Upward propagation of properties

While transferring the properties down from Handshake is conceptually easy, it isn't clear how to bring the information gathered by the model checker back to Handshake. This is crucial for the optimization properties, that are needed at the Handshake level to apply property-based compiler passes (e.g. rigidification).

There are 2 approaches that I could come up with, but there are downsides to both so I am open to suggestions.

### Store properties in a separate file

The simplest approach is to create a file in a "custom" format and save the property table there (for example in csv). Every time the verification command is called, the flow described above will create a table as a file and it will be shared among all the tools in the verification flow: HandshakeToHW, export-rtl, nuXmv parsing program. Later when we apply a property-aware compiler pass, it will access the same shared file to learn where optimizations can be applied.

The problem of this approach is that it relies on a custom format file and MLIR is not aware of it.

[... cut ...] hake to Handshake, but internally it will do:
- Handshake to HW conversion
- call export-rtl
- call nuXmv
- parse the results

The problem with this approach is that it seems much more complex to implement and maintain, without bringing too many benefits.

---

**tl;dr**: we are proposing to introduce formal properties into dynamatic as a table to guarantee as much flexibility as possible. 2 approaches are proposed: saving the properties in an external file and saving them as attributes in MLIR.

Jiahui17, 2025-04-07:
@gioelegott 

It is great seeing new scenarios there (the row with "New Correctness Check" or "Error").

But it also makes me wonder, what is the fundamental difference between the checking that we want to support here and other sanity checks that we have already been doing? As far as I know, we have always been doing these checks:
- Type constraints: some operations must have a Boolean input pin, some operations must have matching input/output signals.
- Materialization constraints: Some passes require the circuit to have one use per value, and some don't care.
- Persistence constraints: Some units assume that the input stays constant starting from the cycle when the valid is one and until the token transfer happens.

To put them in perspective, we are proposing to check these properties:
- Circuit invariant: An invariant of the circuit state (why is it different from the persistence constraint?)
- Optimization property: Also can be an invariant of the circuit state.

Unlike assertions that are specific to a particular code snippet or a function (e.g., an assertion as a loop invariant that we believe to hold), the constraints and checking that we are doing here are much more generic, and we want to assert them throughout the compiler pass.

Jiahui17, 2025-04-07:
@shundroid Also, Shun was mentioning before that it would be great to have a set of assertions to debug the circuit.

And also the properties we had in ElasticMiter.

gioelegott, 2025-04-14:
Thank you @Jiahui17 for the comment.
While we are aiming to have a general way to insert checks in the generated code I don't think all the checks that you mentioned should be included. More precisely I don't think that Type constraints and Materialization constraints should be checked here.

The goal of the framework that we are considering is to create properties that can be checked using formal verification as from the title of the issue. Type constraints and materialization constraints don't really fit in this framework because as far as I know can't be checked with formal verification tools (model checker). The properties that we want to check are circuit level properties that appear once the code is generated so after type constraints and materialization constraints have been already enforced.

On the other hand I think persistence constraints can and should be included in this framework as they relate to circuit level signals.

To sum up I think that with this framework we should aim to allow all constraint and property checking that work at the circuit level, while other checks, that can't be translated into a formal model for model checking, should be done elsewhere.

gioelegott, 2025-05-02:
After some in-person discussion we decided to save the properties into a shared file that different compiler passes and tools can use to communicate information about properties.

This shared file is a `.json` where each property has the following structure:
```
  {
    "check": "unchecked",                // Information from the model checker (can be true, false, or unchecked)
    "id": 0,                             // Unique identifier of the property
    "info": {                            // Additional information needed to generate the property; the
      "owner": "fork0",                  // information depends on the type of property
      "owner_channel": "outs_0",
      "owner_index": 0,
      "user": "constant0",
      "user_channel": "ctrl",
      "user_index": 0
    },
    "tag": "opt",                        // Tag that identifies what the property is for (can be opt, invar, ...)
    "type": "aob"                        // Type of property (can be aob, veq, ...)
  },
```
We plan to implement the formal property infrastructure with the following steps:
- Implement an Annotate pass, that creates the formal properties and adds the ID of each property to the corresponding owner. The owner is an MLIR operation (not a Value/channel). Corresponding PR: https://github.com/EPFL-LAP/dynamatic/pull/402
- Generate the properties (in SMV) from the property database (`.json` file). Corresponding PR: https://github.com/EPFL-LAP/dynamatic/pull/406
- Integrate the model checker into a Dynamatic tool/script and parse the results back into the property database. Corresponding PR: WIP

closed 2026-02-05 by Jiahui17