EPFL-LAP/dynamatic #223: [RFC][Handshake] Use `BundleOp`/`UnbundleOp` to lower burden of writing RTL

lucas-rami, 2024-12-21:
While I was reviewing something related to the ongoing speculation work I thought of something that could help people who add extra signals to channels all over the circuit to do less RTL work to support them (related to #140). For example, here is a piece of code that requires someone to write an adder that will accept an extra signal on all its adjacent channels.

```mlir
%lhs = ... : <i32, [tag: i2]>
%rhs = ... : <i32, [tag: i2]>
%add = addi %lhs, %rhs : <i32, [tag: i2]>
```

And here is the transformation I am thinking about that could be applied automatically pre-HW lowering.

```mlir
%lhs = ... : <i32, [tag: i2]>
%rhs = ... : <i32, [tag: i2]>

%lhs_control, %lhs_data, %lhs_tag = unbundle %lhs : <i32, [tag: i2]> to _
%lhs_pure = bundle %lhs_control, %lhs_data : _ to <i32>

%rhs_control, %rhs_data, %rhs_tag = unbundle %rhs : <i32, [tag: i2]> to _
%rhs_pure = bundle %rhs_control, %rhs_data : _ to <i32>

%add_pure = addi %lhs_pure, %rhs_pure : <i32>

%add_control, %add_data = unbundle %add_pure <i32> to _  
// %rhs_tag could be used just as well as %lhs_tag below
%add = bundle %add_control, %add_data, %lhs_tag : _ to <i32, [tag: i2]>
```

No RTL work required there because the adder operates on simple channels. This would only work for operations that don't actually do anything with extra signals beyond forwarding them, in a way to be defined on a per-component basis, though I suspect a few "types of behavior" will cover many components.  

Of course this is very ugly, but nobody would really have to ever look at this. Though, and this is orthogonal, I wonder if we should not introduce a pair of ops to split/merge just the extra signals from a channel. This would make every pair of unbundle/bundle a single op.

CC-ing people who may be interested @Jiahui17 @AyaElAkhras @murphe67.

murphe67, 2024-12-21:
The main topic that seems to come up when I have this discussion with @lana555 and @Jiahui17 is what is the handshake dialect for: why do we have a single operation for a 2-input mux, with 3 input (channel) variables, instead of explicitly including a join operation and a logical mux operation, and having 9 input (signal) variables? 

For me this a direct mapping to a rectifier: is the rectifier represented in the IR as an operation which has input and output channels, or do we explicitly represent signal level information. There is no difference between a mux and a rectifier to me. 

And I feel the obvious, if maybe not full answer, is because we do buffering on a channel basis, and so we need a channel-level representation. @lana555 has added to this by saying we may have other channel-level algorithms in the future. The handshake dialect is our channel-level representation, and (in my opinion) should remain channel-level in all philosophical details.

RTL is our signal-level representation. You have the comment "no RTL work required": there **should** be RTL work, because this is signal-level work. If we have complex signal-level work that we cannot do using RTL, we should add an additional IR level, not corrupt our existing one. 

For me the signal-level work to be done is extremely simple, and so we should use python to directly generate the RTL implementation. How we represent this in handshake is a different question: you seem to have objections to us directly using operations to represent channel mutations, but I find them unconvincing, and will be using operations to implement speculation.

I can't speak for Jiahui on his specific plans for the rectifier, but I know he agrees with me on a general philosophical level, if maybe not with the same intensity.  

To go a step further, I would argue bundle-unbundle should be scrapped entirely, as should additional upstream signals. If there are those (I feel @lucas-rami and @paolo-ienne are the main champions) who want to do all processing at the handshake level, some proposal must be made that creates coherency with the core assumptions of buffering, and therefore the core assumptions of handshake.

(The side note for removal of upstream signals: The valid and ready signals provide a handshaking mechanism for a **one-way** data transfer. There is no mechanism to transfer data up-stream. Any project which wants to add additional upstream signals will have to rewrite the entirety of our current type system, and we should stop pretending otherwise with an ignored "downstream" variable.)

lucas-rami, 2024-12-21:
You seem to be wholly missing the point of this particular use case. This is an hypothetical below-buffering transformation, so here I really don't care for the limitations of our current buffering algorithms. You can call it an intermediate IR level between Handshake and HW if you wish, I have myself been explicitly avoiding that specific phrasing because every idea I brought forward along those lines was shut down based on the concept alone it seems. But indeed it is basically what those operations are, in this use case at least. We can create a new dialect if people like code overhead, or put them in HW, but where they are really is a cosmetic difference.

I don't understand why anybody would want to make people write RTL if I am offering a solution that does not require it at no performance cost. It is indeed signal-level work, one so trivial that the compiler can do for you. As I have mentioned elsewhere, the generators for bundle/unbundle are trivial to write, and it is trivial to lower those ops in the current pipeline.

On upstream signals, I have absolutely never championed them myself and have 0 use case in mind of what they could be useful for. When proposing the current type system design I relied on a panel of around 10 researchers (which you were a part of) to tell me whether that made any sense, and nobody objected, so I went ahead to enable any potential future use cases. If everybody agrees they will never make any sense, let's nuke them immediately, they complexify the code.

> you seem to have objections to us directly using operations to represent channel mutations, but I find them unconvincing, and will be using operations to implement speculation.

I have scalability concerns (more than objections) indeed. I am afraid those only become more obvious once one has more experience writing compilers.

murphe67, 2024-12-21:
> You seem to be wholly missing the point of this particular use case.

Jiahui and I are more or less the only end users of bundle-unbundle: @AyaElAkhras will use immutable types (types which do not change between operations), with the same tag forwarding system we will build for speculation. I wrote the original python script that we used for rectification on "Eliminating Excessive Dynamism of Dataflow Circuits Using Model Checking". I know both of these problems, and what use cases we have, intimately. 

>  This is an hypothetical below-buffering transformation

While currently we buffer and then perform speculation, it is easy to imagine kernels where we wish to add speculation first and then add speculation-aware buffering. I have no use for representations of speculation which do not support buffering.

For rectification, it will always be the final step in compilation. The only real factor for consideration is how easy RTL generation is. Bundle-unbundle makes difficult to read code which will be annoying to lower.

> I don't understand why anybody would want to make people write RTL if I am offering a solution that does not require it at no performance cost. 

Because it breaks buffering.

> I have myself been explicitly avoiding that specific phrasing because every idea I brought forward along those lines was shut down based on the concept alone it seems. 

Until we have concrete issues with RTL as a signal-level representation, I would not support any suggestions for a new one.

> When proposing the current type system design I relied on a panel of around 10 researchers (which you were a part of) to tell me whether that made any sense, and nobody objected,

A description of a decision-making process is not an actual defense of a decision.

>  I am afraid those only become more obvious once one has more experience writing compilers.

An ad hominem fallacy is also not an actual defense of a decision.

lucas-rami, 2024-12-22:
Again, this would be *pre-HW lowering*, meaning post-buffering, post-speculation, post-whatever. It could even be part of `HandshakeToHW` if we did not care for modularity.

> A description of a decision-making process is not an actual defense of a decision.

When you talk to people who are supposed to know more than you and care about the decision and they advise incorrectly/incompletely it kinda is. Personal responsibility is rough, I know. I think I have publicly apologized for and addressed bad design decisions and bugs of mine enough to know something about it.

> An ad hominem fallacy is also not an actual defense of a decision.

I am not sure you understand what that means. My comment is rooted in quantitative verifiable data. I see you have 0 lines of code contributed to this particular compiler (sorry, integration tests don't count), so I am just wondering where you get your design opinions from given your lack of actual contributions.

---

Here is an actual piece of feedback that would have been useful to provide. My solution/example does not work if `addi` is not combinational, which makes this idea generally less applicable. Perhaps it is what you meant by "it breaks buffering", in that case sorry for missing it. I think my phrasing is slightly shorter though. Closing.

closed 2024-12-22 by lucas-rami

murphe67, 2024-12-22:
Well, thank you in any case for the lively discussion 😁. Have a merry Christmas!