EPFL-LAP/dynamatic #228: Refactoring & Extending MILP-Based Buffer Placement

QinYuan2000, 2025-01-06:
# Refactoring & Extending MILP-Based Buffer Placement

## Introduction

Buffer placement is the core of the entire Dynamatic HLS tool. Its role is similar to scheduling in Vitis HLS. Over the years, many buffer placement algorithms have been proposed; here is a short list:

1. **Pre-characterized delay**: Using a pre-characterized delay to estimate critical paths.
2. **Technology-mapping-aware**: Using subject graphs to estimate critical paths after mapping.
3. **Stall-eliminating**: Eliminating stalls wherever possible to remove circuit redundancy.
4. **Cost-aware**: Some buffer types are cheaper; choose them strategically to reduce resource usage.
5. WIP: other approaches

Currently, Dynamatic only implements the first approach (pre-characterized delay). However, we want to integrate multiple approaches of buffer placement, all modeled as MILP problems but each with differing constraint sets and objectives.

## An Overview of Our Proposed Flow

### Buffer Placement MILP

A buffer placement algorithm has three main goals:

- Timing closure. The produced circuit should respect a certain clock frequency target.
- High throughput. A higher throughput is always desired.
- Lower area. Practical systems are usually resource-constrained; a higher area often means higher power.

Following this classification, we can visualize the relationship between 5 different approaches as the following:

[image]

For each of the three goals, every approach chooses one and exactly one set of constraints (i.e., one big box) or one objective. Also, although the graph structure is nearly identical across all 5 algorithms (i.e., CFDFCs), the constraint formulation are fundamentally different.

The constraint sets (D, T) and the objectives (O) share a set of variables that describe the buffers' characteristics:

- Buffer latency (L) on three signals respectively: Data, Valid, and Ready;
- Buffer slot number (N);
- Shared enable signal (SE, added in future);

In this way, adding a new buffer algorithm (e.g., replace D2 by another set of delay constraints) doesn't require significant changes (if any) to other constraint sets or objectives (e.g., to the throughput constraints).

In terms of implementation, we have a base class (BufferPlacementMILP.cpp) as a database for all the constraint sets and objectives (currently, we have D1, T1, O1, and we plan to extend it with D2, T2, T3, O2, and O3). Each algorithm (e.g., FPGA '20) is a new class (e.g., FPGA20Buffers.cpp) derived from the base class, which loads the constraint sets and the objective (i.e., D1, T1, and O1).

### Interpret the MILP Solution

The MILP solutions do not directly specify which buffers to use. Instead, they provide buffer attributes (e.g., L and N) for each channel. In the RTL backend, multiple buffer components may be inserted, each assigned a slot number. Therefore, we need an intermediate step that translates the MILP output into concrete buffer placements that the RTL backend can handle.

[image]

In the figure above, each MILP approach generates variables (i.e., L, N, and possibly SE) that define the buffers’ characteristics. A heuristic decision in the buffer placement pass maps these variables to slot numbers for each buffer component, which are then propagated through the handshake-to-HW pass for actual buffer instantiation in RTL.

The detailed steps for the interpretation are proposed in (#).

### Proposed New Features

- The structure of MILP variables contains cut Data, Valid, and Ready. We will modify them to represent the latency of Data, Valid, and Ready.
- Dynamatic currently includes D1, T1, and O1. Modification is needed to make them compatible with the changes in the MILP variable structure. In addition, we introduce more buffer placement algorithms (D2, T2, T3, O2, O3), considering the standardized interfacing between the constraint sets (e.g., D2 can be combined with any of T1, T2, T3).
- Dynamatic currently has derived classes for FPGA ’20 and FPL ’22. We will add derived classes for MapBuf, Cost-aware, and FPGA ’24.
- The interpretation heuristic is to be implemented.
- New HDL files for buffers are to be added to support the new MILP features.

## Detailed Tasks
This section introduces the detailed tasks, we will devise a separate tracking issue followed by a separate PR for each task.

### Integrate T2 & O2 and set up CostAware

- Add new constraints and objective to BufferPlacementMILP class (T2 and O2).
- Add a derived class CostAwareBuffers, which uses D1, T2, and O2.
- Implement the heuristic for interpreting MILP results.
- Add new HDL buffer modules to support features in T2 and O2.

### Integrate the new buffer attribute

- Add shared-enable (SE) to variables that describe buffers' characteristics. Modify T2 and O2 to accommodate the new variable.
- Add new HDL buffer modules to support SE.

### Integrate D2 and set up MapBuf

- Plan for this part is WIP

### Integrate T3 & O3 and set up Stall-Eliminating

- Plan for this part is WIP

lana555, 2025-01-13:
Thanks, looks good. @Carmine50 is there anything related to MapBuf that we forgot?

Carmine50, 2025-01-14:
I think not. But we should change the name convention, as we mentioned last time. For example, the word "cut" has different meanings depending on the context, so it might be wise to use a unique name for each context. The word "cut" is used a lot in technology mapping papers so it might be wiser to change the word "cut" in the context of buffer placement to something else.

Carmine50, 2025-01-14:
Btw, there is a PR #234 that adds MapBuf code inside dynamatic. The code structure does not change with respect to the other existing buffer placement.

paolo-ienne, 2025-01-24:
Question: should not this be a Discussion rather than an Issue? If you agree, there is a nice button up right "Convert to discussion".... :smile:

Jiahui17, 2025-01-26:
@paolo-ienne 

> Question: should not this be a Discussion rather than an Issue? If you agree, there is a nice button up right "Convert to discussion".... 😄

Should we do this consistently and move all high-level project plans to "Discussions"?

paolo-ienne, 2025-01-26:
I would think it appropriate, but let's discuss it with others (@lana555).

pcineverdies, 2025-01-27:
> Should we do this consistently and move all high-level project plans to "Discussions"?

Most of our issues are in between a discussion and an issue tbh, but probably I'd say that an issue is more of a "hey everyone, this thing is not working" rather than "I'd like to to this". My own FTD issue is actually more of a discussion in this sense.