EPFL-LAP/dynamatic #235: [Formal][SMV] SMV unit library and generation proposal

gioelegott, 2025-01-15:
# Unit generation
The unit generation and instantiation is managed by the a matching system, that matches a hw type to the corresponding way to generate it. The matching process is determined by the `rtl-config-smv.json` file as described in [the docs](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/Backend.md#component-description-format).

In the json file we differentiate 2 different components:
- **generic components**: they can be directly translated to .smv and they can be a dependency of other components.
- **generated components**: they are parameterized components that need a code-generator. The code-generator creates an ad-hoc unit for each combination of parameters that is needed in the model. In general components can't depend on generated components.

## dot2smv (smv backend in the legacy Dynamatic)
The model generation problem was already solved by the [dot2smv](https://github.com/Jiahui17/dot2smv) project, but it needs to be properly integrated in the current system. We will try to reuse it as much as possible, or at least use it as a guide to to implement the model generator. In the following sections we will discuss the problems that arise with the model generation, analyze how they are handled in the dot2smv project and propose some solutions for our implementation.

## Generic components
Generic components are generally easy to implement and to use, because they have a 1-to-1 correspondence to hw units. We will use a similar approach for generic components as the one used for VHDL: we plan to have a directory that contains all the generic components, organized in arithmetic, handshake and support components. For each category we will have an additional sub-directory with the dataless version of certain components.

### dot2smv implementation
The equivalent of generic components in the dot2smv project is in [src/elastic_components.py](https://github.com/Jiahui17/dot2smv/blob/master/src/elastic_components.py) but it has the following shortcomings:
- the signal naming convention is not respected
- code readability is poor and all the components are packed in the same file
- there is no distinction between dataless and non-dataless components
- there is no easy way to understand which component corresponds to which hw type

To reuse dot2smv we plan to go through every component and revisit it to integrate it in the new system.

## Generated components
Implementing generated components requires more care as they need some automatic generation process that takes into consideration one or more parameters. Ideally we need a program takes a specific input to produce the corresponding model. Then we will match the hw type, with a command that will call the program with the correct inputs.

### dot2smv implementation
The generated components are created in [src/component_writer.py](https://github.com/Jiahui17/dot2smv/blob/master/src/component_writer.py) and it currently supports the generation of: memory, fork, lazyfork, merge, exit, fifo_inner, different types of buffers, delay, join, operator, decider. The following units are missing: mux, control_merge (they are implemented as generic). 

There are two options for the code generator:
- reusing the component_writer
   - pros: easier parsing and string manipulation in python, component generation is already implemented and is already correct
   - cons: effort is needed to update the component writer (naming convention, operation missing, etc.)
- writing a C++ program
   - pros: integration as an additional backend generator [tools/backend](https://github.com/EPFL-LAP/dynamatic/tree/main/tools/backend), lower implementation effort ([wip](https://github.com/EPFL-LAP/dynamatic/blob/dev/gioele/export-smv/tools/backend/smv-generator/smv-generator.cpp))
   - cons: C++ can be very verbose for parsing inputs, some decision that were made in the dot2smv tool may be lost, correctness checks

### Interface
The code-generator interface defines how we can call it to create a model. Our goal is to have a flexible code-generator that can generate all the parameterized components with all combination of parameters. An example of how the interface would look like is:

```
/bin/generators/smv-generator output/muli.smv bop muli int32
``` 
With this interface we specify in order the output file, the type of operation (in this case binary operation) and some additional parameter (in this case the binary operation is integer multiplication for int32).

### Dependencies
Some components can internally instantiate other components, creating a dependency. The dependency system is described [here](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/Backend.md#dependencies). Something that is clear from the documentation is that sometimes it is not possible to manage dependencies of generated components.

There are 4 types of dependencies:
- generic component (g1) depends on generic component (g2): no problem
- generated component (G1) depends on generic component (g2): no problem
- generic component (g1) depends on generated component (G2): since generic components are static, also the parameters that would generate the generated components are. Therefore we can statically generate G2 for the specific parameters and use the dependency system freely. This is the case for all binary operations (generic components) that depend on join (generated component). All the binary operations use a join with 2 input channels, therefore we'll create a generic component `join_2_1_generic `for it.
- generated component (G1) depends on generated component (G2): if a generated component depends on another generated components, but the parameters are constant, then we can reuse the approach of the previous point. If the parameters of G2 are not constant, i.e. they depend on the parameters of G1, then we need to solve the dependency internally to the generator. This means that every time we generate G1 the generator will also generate the corresponding G2. To avoid name-clashing we need a unique name for G2, that identifies it as the component generated as a dependency of G1.

Jiahui17, 2025-01-16:
@gioelegott Thanks for the proposal! @murphe67 I would also like to hear your thoughts :D

---

One idea on the file organization. Does it make sense to organize the generators like this?

```
dynamatic
└── experimental
    └── tools
        └── smv-unit-generators
            ├── smv_arith_generator.py
            ├── smv_buffer_generator.py
            ├── smv_fork_generator.py
            ├── smv_join_generator.py
            ├── smv_mem_generator.py
            └── smv_utils.py
```

If yes, maybe we could outline how the files would look directly in this issue.

gioelegott, 2025-01-16:
I like the idea of splitting the generator into separate unit generators; the generation process of some units can be quite complex and it is usually independent from other units.

For the structure of these generators I think we should follow these points:
- similar interface (e.g. they take the name of the output file as input)
- generate units with the same stile
   - snake case
   - append the number of inputs and outputs to the name, e.g. `fork_1_4` (this is open to discussion @Jiahui17, maybe it makes more sense to append the value of the parameters to the name: `fork_<out_ports>`, `arith_op_<latency>`)
   - generate as much as possible (as @Jiahui17 suggested reducing the number of free parameters and of complex data structures like arrays will make the generated code less elegant, but will greatly improve the runtime).
   - same naming convention for signals and internal components
- handle generated-generated dependencies without name clashing

murphe67, 2025-01-16:
Hello, thanks for the very nicely explained proposal!

I have some (mostly instinctive) opinions that I have not thought through fully, so there may be some obvious reason against them but:

1. On a language level, generic and generated look very similar. Consider changing the categories to something like "basic and parametrized" to make code/documentation easier to read.
2. I would definitely recommend Python for this.
3. Consider moving all of the generation details out of C++ and into Python: make a generation script for every single HW operation, even the generic/basic units. The C++ caller does not need to know about the dependencies or generation method. Each python script knows its own dependencies, and can check the output folder to see which operations have already been generated (would require consistant naming). In this way, the only information you should need to pass to the scripts are the HW attributes, which could be flags to the python script. (Rather than passing every HW attribute to the python script, which could require adding a lot of flags that you ignore, the C++ caller should only pass along HW attributes whitelisted per script.)
4. Structure the python library so that each python script can call the required external functions of the other scripts, so that you don't need to go through bash to generate dependencies.
5. Use very explicit naming for operations e.g. fork_out4 instead of fork_4 or add_float_lat5 instead of add_5. When auto-generating code, it should be more verbose than normal code.

This would end up with a lot of very short python scripts (especially for the basic units, which would just copy a file), but makes the interface between MLIR and the generation very simple. With a clean interface, a python library would be well set-up to handle dependencies (the ability to check what has already been generated would make things much easier for you guys)

Additionally, having a python script per unit makes it very easy to change a basic unit to a parameterized or vice versa, you are never locked into any weird corners.

murphe67, 2025-01-16:
On a short second thought: you probably don't need a different script for every unit, but rather that the C++ asks to generate a HW op, and that calls a python function specific to that OP. So the C++ knows nothing about how that operations maps into model checking.

gioelegott, 2025-01-16:
Thank you for the comments:

1. At first I was a bit reluctant to change the name because I wanted to keep the same naming convention as in [here](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/Backend.md#generic), but if it is causing some confusion I better change it.
3. We are currently treating SMV as a new HDL backend, so the C++ caller already has a way to manage dependencies with the [config file](https://github.com/EPFL-LAP/dynamatic/blob/main/data/rtl-config-vhdl.json). We can argue that the dependency system isn't perfectly suited for SMV, because of the dependencies with generated components, but creating another dependency system in python is maybe a bit redundant. I haven't completely made my mind up though, so if you have any other insights on why recreating the dependency system would be a good/bad idea please let me know.
5. Great suggestion, thanks!

I think the main question to answer is if we actually want to treat SMV as an additional HDL like VHDL or SV. If so then I think it makes sense to keep the C++ aware (or at least as aware as possible) of the dependencies, like it's aware of them for VHDL.
If we want to consider SMV as something separate, then we can even have a separate "python project" that generates SMV units and is used as a black box in the matching process.

murphe67, 2025-01-16:
In my opinion, it would be better if the HDL generation process also worked as I described, and this would be a nice (small enough) case to try and test drive it- but it depends on how happy you are to dive into something more unknown vs. ending up with the same headaches as before (but they are at least predictable ones).

We are currently trying to extend the current backend in #172:  units that were basic before are now suddenly parameterized, the dependencies can be situationally completely different, and we are going to end up with 5 different places that generate RTL, all using different code structures. 

So a favor to me to build evidence I would appreciate it, but I understand the urge to try and replicate the current system also.

Jiahui17, 2025-01-17:
@gioelegott 

1. Emmet's point is that the current config file thing is quite limited even for VHDL/Verilog itself. I think this is a limitation of the hw dialect. In theory, this file should give us 100% information on how the RTL should be generated or what files we need exactly, but now one `hw.mlir` only has partial information and it relies on the config file to get the dependencies. But this doesn't work if some unit depends on some other units which need to be generated by a generator.

> append the number of inputs and outputs to the name, e.g. fork_1_4 (this is open to discussion @Jiahui17, maybe it makes more sense to append the value of the parameters to the name: [...]

2. If we do everything in one script then we could track the list of generated modules to avoid duplications.

murphe67, 2025-01-17:
For single script, would you just pass it a list of operations?

gioelegott, 2025-01-18:
@Jiahui17 are you suggesting to completely discard the matching system with the config file? Or do you mean we should have a single script that can generate any module depending on some parameters and call it through the config file?

Honestly I would like to keep the config file matching system for now, and focus on creating one (or more) generator(s) that can generate a unit with all the dependencies independently from the config file. I liked @murphe67's idea to check the output folder if some dependencies have been already generated to avoid duplicates.
I think that by keeping the dependency system independent we can easily stop relying on the current matching system once we decide for another solution.

murphe67, 2025-01-18:
For first instincts on the matching logic- it does not look like it would be difficult to duplicate exactly-as-is inside of python, the matching rules are not too complex.

Once you have an exact replica, from there it would probably be wise to look at how much of the complicated config logic you can move out of the matching and into the generation, and whether there are any matches you can merge to use the same generation function.

If you rely on the existing matching system to do generation faster at the beginning, you would still have to recreate it afterwards to get the freedom you need to start fixing some of the problems.

gioelegott, 2025-01-18:
Are there other shortcomings of the current matching system, apart from the dependencies with generated components? It would be nice to know which are these so that we don't create a new system with the same problems.

Anyways I think the matching problem and the generation problem can be considered separate (correct me if I'm wrong). If you have a generator that can be used with the current matching system (excluding the dependency, that would be handled internally), then it should be easy to develop a new matching system that uses the same generator.

Based on the discussion so far I was thinking to develop a single generator file that handles all the file generation (and testing!!!).
This generator would take as an input a file name that uniquely identifies a unit with all its parameters. I can imagine having a class for each unit type with the following fields:
- initialization method that can parse a file name to extract the parameters
- dependency list, with the names of the files it depends on (created at initialization time)
- generation method:
	- check if the file already exists
	- if it doesn't create it and generate the unit
	- generate recursively the file names in the dependency list
- test method (it isn't really the topic of the discussion, but I feel it is important anyways):
	- generate a main module that instantiates the unit
	- run nuxmv on the main module to be at least sure that the model can be built
	- (optional) add more tests

With this approach we would have a generator that can be used in the current matching system. If then we want to recreate the matching system in python, we could simply call the generation function over all the operation types that are present in a given MLIR file. 

One problem I can see with this solution is the scalability of putting all the parameters in the file name. While the use of the generator will be most of the time automatic, I wonder what will happen if new parameters need to be added.

Also let me know if I'm missing something in the relationship between the matching and generation system, that would make the two processes more tied to each others.

murphe67, 2025-01-18:
My general issue with the matching system is that it shouldn't exist: 
If we take the example from the specs:

[image]

The matching systems decides dependencies and generation method by both hw.name and hw.parameters. However, I believe it should simply decide a generation method based on hw.name, and inform it of the hw.parameters, and the generation script is responsible for the rest. 

[image]

murphe67, 2025-01-18:
@shundroid 
do you think we actually should just rewrite export-rtl like this already? it is a much simpler change than I had originally imagined, and it would make our lives a lot easier for implementing the signal manager.

Jiahui17, 2025-01-18:
@emmet

Thanks for sharing your thoughts! I like the idea of offloading more dependency generation logic into the generators themselves.

One thing that is not super clear to me now is name sanitization (a fancy way to say: make sure that one module is generated only once): do you have in mind a way to systematically make the module name unique per each parameter combination (either specified in hw.mlir or inside the extern module)? Also, the solution is probably not to dump the parameter into the filename (sometimes the configuration gets too long to be written on a filename). Our current RTL backend tracks the set of unique parameter combinations (in C++) and only generates one for each set.

murphe67, 2025-01-18:
I have no experience with this at all, though I agree dumping the parameter values is not very scalable.

With your point of having a single RTL generator passed a single data structure, we could move whatever we have now directly into the generation script.

Otherwise maybe just numbering them, with some sort of dictionary mapping numbers to specific parameter combinations?
In single script form it could be internal, but otherwise you could keep the dictionary stored in a file?

The output would be super unreadable though, if you open the RTL the number doesn't give you any information.

Jiahui17, 2025-01-18:
> With your point of having a single RTL generator passed a single data structure, we could move whatever we have now directly into the generation script.

I just realized that I wasn't taking into account units with extra signals. Now I can totaly imagine that even for very basic units, using generic value to control the instantiation is just way too clunky. Emmet, do you mean that we should generate all components (even for things like OEHB, TEHB that we have been using generic value to control the instantiation for years)?

Just want to clarify. I still think the config file does a very good job in handling the **generic compontent**s. In fact, even if we find a good way of handling the generated units, the netlist generator has to understand what generic value(s) it needs to put.

Jiahui17, 2025-01-18:
> Otherwise maybe just numbering them, with some sort of dictionary mapping
> numbers to specific parameter combinations? vIn single script form it could
> be internal, but otherwise you could keep the dictionary stored in a file?
> 
> The output would be super unreadable though, if you open the RTL the number
> doesn't give you any information.

I think this is fine. We could print the configuration parameters into a
comment for debugging.

If we decide to go full on generators, we can still keep the config file for
now (each entry specifies a generator). For instance, C++ will call the generator like

```bash
// HDL file printed to stdout?
python3 vhdl-generator.py --name "handshake.buffer" --parameters "{ DATA_WIDTH = 16 : ui32, SLOTS = 1 : ui32, BUFFER_TYPE = "OEHB" }" --module_name "handshake_oehb_1"
```

Here the "module_name" is sanitized in C++.

If we want to be extra interoperable, maybe we could do:

```bash
// HDL file printed to stdout?
python3 hdl-generator.py --hdl "vhdl" --name "handshake.buffer" --parameters "{ DATA_WIDTH = 16 : ui32, SLOTS = 1 : ui32, BUFFER_TYPE = "OEHB" }" --module_name "handshake_oehb_1"
```

murphe67, 2025-01-18:
> Here the "module_name" is sanitized in C++

My first instinct was to agree with this, but my second thought is "does this complicate a unit generating its own dependencies?" If want to do that, I feel like there needs to be some way for a generator to request the corresponding number for a combination of parameters. Maybe each generator knows how its own parameter combos map to numbers? But then export-rtl would need to request the numbers from the generators in order to do instantiation correctly.

> Emmet, do you mean that we should generate all components (even for things like OEHB, TEHB that we have been using generic value to control the instantiation for years)?

I think this is more or less our only option? Unless we want to double the number of HW operations.

> I still think the config file does a very good job in handling the generic components.

Yes, I think we have a lot of things that are working well in the current system that we would replicate exactly. Generic units would be generated the same way, but the info to do so would be in the generator code file instead of in a JSON config file.

murphe67, 2025-01-18:
The naivest way of dependency generation with numbers is to just add an underscore after the number and restart- e.g. if export-rtl calls "generate an add with these parameters and number it 47", and that needs a join, the add generator calls "generate a join with these parameters and number it 47_0".

But you would get a lot of duplication in the generated library.

Other option is just somehow keep the mapping of combos to numbers in a file, and have both export-rtl and the generators able to access and edit it.

Jiahui17, 2025-01-19:
> The naivest way of dependency generation with numbers is to just add an
> underscore after the number and restart- e.g. if export-rtl calls "generate
> an add with these parameters and number it 47", [...]

I think it is fine like this and file duplication is a secondary concern. The
generators just take a "MODULE_NAME" (e.g., "handshake_buffer_47") argument and
prefixes all the internally generated modules with "MODULE_NAME__" (e.g.,
"handshake_buffer_47__join_0").

With this, we should completely avoid name collision.

---

Separate note: some generators follow this naming policy (XLS prefixes all the modules
with the filename in the original code + "__" and I think Vitis HLS also 
does something similar) but some do not:

For instance, Flopoco allows you to specify a list of arith units that you want
to generate. But as far as what I have seen, there is no way to avoid name collision
after multiple calls to the generator: you could specify the name of the
top-level module, but the internal modules do not have any 
specific prefix based on that.

Example here: https://github.com/EPFL-LAP/dynamatic/blob/197367e1a6a11c1d202fd6bc441d0d3b6bc4c629/data/vhdl/support/flopoco_ip_cores.vhd#L262

murphe67, 2025-01-19:
This looks nice to me - I agree I don't think it matters if we have duplication in the generated library, neither logic synthesis tools or formal verification tools care if there are 10 identical join units.

It is probably (slightly) less runtime efficient to repeatedly generate units but unit generation is such a small workload I don't think it matters.
 
For the most core of dependencies, which will be duplicated most often, generation will  simply be copying the pre-written unit, which is especially cheap.

shundroid, 2025-01-20:
Sorry for the delay!

> do you think we actually should just rewrite export-rtl like this already? it is a much simpler change than I had originally imagined, and it would make our lives a lot easier for implementing the signal manager.

Yes I agreed. We’ve confirmed that the signal manager can be integrated much more easily into the full-generator-based composition.