EPFL-LAP/dynamatic #281: [CI] Parallelize Integration Tests

shundroid, 2025-02-09:
Thank you for introducing the CI!
I realized the integration tests take long time, likely because the integration tests are performed non-blockingly and one by one.

https://github.com/EPFL-LAP/dynamatic/blob/e1e94cbf807d9e58be8ed7411da840833bef62af/tools/integration/run_integration.py#L256-L285

I think this can be improved by replacing `subprocess.run` with `subprocess.Popen` and using `threading.Thread` to monitor the subprocess.

https://github.com/EPFL-LAP/dynamatic/blob/e1e94cbf807d9e58be8ed7411da840833bef62af/tools/integration/run_integration.py#L155-L163

@ebosnjak @pcineverdies What do you think?

shundroid, 2025-02-10:
Thank you so much!! I'm looking forward😄

> do you have any idea what happened here? I guess you did not have such an issue locally with if_loop_3... I wonder why we had such a non-deterministic behavior.

I don’t know much beyond what the log shows. In my environment, this integration test finishes quickly (under a minute).

From the raw log, it looks like it actually took longer. It might be that the simulator process was interrupted by the host, for example.

[image]

ebosnjak, 2025-02-10:
Sorry for not replying, I’ve been ill so I’m not really active on the project.

Regarding parallelizing, that’s exactly what I was thinking about a few days ago when I saw the pipeline struggling a bit. It’s not a difficult task and I’ll do it when I can.

As for random timeouts, I don’t know what might be the cause. Maybe the versions of modelsim are different? The runner uses 20.1.

ebosnjak, 2025-02-12:
I opened a PR (#289) with the features you requested. The integration tests are run with a `ProcessPoolExecutor` and the performance improvement is significant (from ~1h 10 min down to ~12 min). I also added a step in the Actions workflow which uploads `report.txt` from all tests along outputs from running the Dynamatic binary itself. This data is available in the form of artifacts that are visible in the Actions UI after running the tests and can be downloaded from there.

Let me know if you are satisfied and if you would like any other improvements!

closed 2025-02-18 by ebosnjak