EPFL-LAP/dynamatic #284: [Refactoring] Abolishing `runDynamaticPass` method

shundroid, 2025-02-11:
### Issue: `runDynamaticPass` Obscures Pass Behavior  

In Dynamatic, `runDynamaticPass` serves as the entry point for *most* passes, such as:  

https://github.com/EPFL-LAP/dynamatic/blob/261ba42088824e037c53851d7ff7116f1ec93ef3/lib/Transforms/PushConstants.cpp#L66-L72

https://github.com/EPFL-LAP/dynamatic/blob/261ba42088824e037c53851d7ff7116f1ec93ef3/lib/Transforms/MarkMemoryDependencies.cpp#L62-L66

However, I find this design problematic for a few reasons:  

1. **It’s not general.** In MLIR, `runOnOperation` is the standard entry point for passes, as described in the [MLIR documentation](https://mlir.llvm.org/docs/PassManagement/) and even [our own tutorial](https://github.com/EPFL-LAP/dynamatic/blob/261ba42088824e037c53851d7ff7116f1ec93ef3/docs/Tutorials/CreatingPasses/2.WritingASimplePass.md) (which, interestingly, doesn’t mention `runDynamaticPass` at all).  

2. **It hides important details.** For example, when I first examined how operation names are assigned after each pass, I couldn't immediately tell where this happens. I later realized that `runDynamaticPass` is invoked inside `runOnOperation`:  

https://github.com/EPFL-LAP/dynamatic/blob/261ba42088824e037c53851d7ff7116f1ec93ef3/include/dynamatic/Support/DynamaticPass.h#L68-L90

Here, name validity is checked *before* calling `runDynamaticPass`, and naming is performed *after*. This implicit behavior is unintuitive.  

Furthermore, I don't think here is the right place for ad-hoc procedures like this.

Overall, I'm concerned that this design makes it harder for newcomers to Dynamatic to understand the procedure of each pass.

### Proposal  

Each pass should explicitly define `runOnOperation` and handle necessary procedures there, treating validation and naming as explicit calls to library functions. This would improve clarity and make each pass’s behavior more self-contained.

shundroid, 2025-02-11:
It seems #45 introduced this

pcineverdies, 2025-02-11:
I remember one concern from Lucas (maybe I'm able to find the comment somewhere) regarding the possibility of having as many analysis pass as possible, similar to what the current `NameAnalysis` does. For instance, right now the BB of each operation is annotated as a string in the operation itself only, without a class to handle this.

I find this approach quite reliable to centralise the way attributes are handled and to keep them coherent across different passes. 

From my point of view, the more analysis passes you add, the more you also need a way to always make sure they are updated. This is probably the reason why he thought about this common entry point for passes. 
Don't you think that it might get messy without such a centralised approach? 

One can argue "it's just a matter of invoking a couple of functions", but I also feel that this might get chaotic after a while.

(I have no emotional attachment towards the `runDynamticPass`, I just see the point of having it. Also, off topic, I believe that the tutorial you link should need an update, since most of its content is 2 years old...)

shundroid, 2025-02-11:
Thank you for your opinion.  

I think performing procedures before or after `runDynamaticPass` is acceptable *only* if it is natural and reasonable to do so. However, I don’t believe these procedures meet that standard.  

For the naming process (`nameAnalysis.walk`), in some cases, it's already too late to perform naming at this stage. The pass cannot access the names of operations it generates itself—something I personally ran into. While it's possible to manually call `nameAnalysis.walk` inside the pass, doing so is redundant. I don’t agree with a design that forces developers to deal with such inconsistencies, where some operations are named while others are not—at least until they realize the issue. Ideally, every operation should be uniquely named at creation.  

I also believe the other procedures handled here are not a good fit for this location.  

Thanks for sharing the message! I'll read through it, but it might take some time. It may help me better understand the rationale behind `handshake.bb` and `handshake.name` attributes, which have always seemed like two particularly strange features of the Handshake IR to me.

pcineverdies, 2025-02-11:
>  which have always seemed like two particularly strange features of the Handshake IR to me

What do you mean by _strange_? Useless or handled in the "wrong" way?

shundroid, 2025-02-11:
I read through the discussion and really appreciate your great work on FTD—thank you for that!

First, I want to clarify that I’m not proposing adding another pass or attribute—just that we shouldn’t call `NameAnalysis` from outside `runDynamaticPass`. I also want to confirm that `NameAnalysis` isn’t treated like an analysis pass in LLVM, right?

I thought of an alternative approach: naming operations directly in the builder function, like this:

```
builder.create<handshake::ConditionalBranchOp>(..., getUniqueNameFor<handshake::ConditionalBranchOp>())
```

What do you think?

For ensuring unique names across all operations, how about using [AnalysisManager](https://mlir.llvm.org/docs/PassManagement/#analysis-management)? But I don't think `NameAnalysis` is an LLVM/MLIR-style analysis pass. It’s just a library class that actually modifies the IR (e.g., replaceOp).

That said, I don’t think either of these changes are urgent.

> What do you mean by strange? Useless or handled in the "wrong" way?

- For `handshake.bb`: I initially wondered why it’s an attribute rather than using MLIR's built-in block notation (e.g., `^bb0` in the `cf` dialect). After reading the [discussion](https://github.com/EPFL-LAP/dynamatic/discussions/253), I see the reasoning—Handshake IR is meant to represent only hardware-relevant details. But I still think the CFG does exist in this IR 😅. For example, we need to identify backedges in the SpeculationPass at the Handshake level.
- For `handshake.name`: It felt magical at first because I wasn’t sure how or when names were assigned. My proposed improvement (above) is meant to address that.

pcineverdies, 2025-02-11:
- `handshake.bb` problem: I'm happy to discuss the problem of the CFG annotations in the other issue :)

---

- `handshake.name` problem: 

> I don’t agree with a design that forces developers to deal with such inconsistencies, where some operations are named while others are not—at least until they realize the issue. Ideally, every operation should be uniquely named at creation.

Agree! This seems like a reasonable model for the naming problem. 

> But I don't think NameAnalysis is an LLVM/MLIR-style analysis pass. It’s just a library class that actually modifies the IR (e.g., replaceOp).

You definitely have more insights on MLIR than I do, so I don't really feel able to address any problem related to the standard ehehe. I'm open (and happy to help) for whichever design choice that is supposed to bring benefits in the long run :)

---

- `runDynamaticPass` problem:

Again, if we can find a general model to make sure that nothing should be centralized for all the passes, then I think that getting rid of it is more than viable! I guess that the main difference would be: right now the name analysis is included in any pass; in your proposal, the name analysis (or anything else) will be included only if needed by the pass itself. Am I wrong?

murphe67, 2025-02-12:
For me this seems more like a documentation issue than anything? The improvements seems nice, but the real issue (to me) is the confusion caused. 

@shundroid could you make a document for this and add it to the docs folder? We seem to have lots of decisions which were well discussed and then never recorded, and it would be nice to start building the docs folder as we open these issues.

shundroid, 2025-02-12:
That's a good idea! So I'll create a document, including (1) how it works currently (2) the reason why it was implemented in this way (3) the current problems it has (4) possible improvements in the future

murphe67, 2025-02-12:
that would be great, thanks Shun!

closed 2026-10-19 by Jiahui17