EPFL-LAP/dynamatic #285: [Handshake] CFG representation in the IR

murphe67, 2025-02-11:
This is mainly for @pcineverdies and @AyaElAkhras from what @shundroid has brought up, but I thought I would make it a specific issue-

It seems obvious to me that the handshake.bb attribute was a bad decision, and we should be using block notation in the handshake IR: it is an optimization IR, and should contain all information required for transformation passes in the most usable form.

Are you guys opposed to changing this (@shundroid seems to think it would make your analysis much much simpler)? The original motivation from Lucas seems to be "handshake should be as similar to RTL as possible", which I couldn't disagree with more. 

To quote @pcineverdies from the original discussion: 
> My idea comes out of this thought: "It's so simple to manipulate the IR in cfToHandshake before calling flattenAndTerminate. Since no major information gets lost in the process, why can't it be that simple afterwards?"

Lucas's reply was-

> From Handshake forward, the original CFG no longer exists---and I mean that from a conceptual point of view, not just as in "it's no longer visible in the IR"---even if we want to believe really hard that it still does. 

which is straight forwardly wrong: we have many many many optimizations and transformations which require direct CFG analysis.

pcineverdies, 2025-02-11:
Thanks for bringing this up!

Unfortunately I don't have any progress with respect to the content of [this message](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991990). This is exactly the setup I am using at the moment, reliable in all my use-cases. 
In the handshake IR, I temporary re-introduce a CFG structure in order to use the CFG functionalities without relying on the bb annotations only, and then I get rid of them afterwards. The CFG structure is annotated to the `handshake.func` operation.

I've always thought that there is still a strong connection between the handshake IR and the control dependencies between BBs. This is, as you said, highlighted by the fact that we need many times the CFG information before lowering down to `hw` (in this case, I guess there's never been need at all of BBs). 

I don't have any philosophical view over the problem and the way the dialect should work - I just feel the limitation of the current setting. If you believe this can be worth doing, I would say I'm okay. However, just as a side note, what do you (and anyone else) think about this other comment by Lucas?

> However, tomorrow someone will invent a new even more aggressive optimization that transforms the circuit to such an extent that we won't be able to semantically link some operations to any BB that used to exist, and we should be able to handle that i.e., we should not rely on the original CFG.

I think that his concerns were about what might happen in the future, rather than what Dynamatic has right now...

murphe67, 2025-02-11:
@Jiahui17 I was noticing your elastic-miter stuff moves operations from one BB to another- how does the current 'CFG-less' system work for you? 

Would restoring the CFG to the IR negatively impact your stuff?

Jiahui17, 2025-02-12:
@murphe67 @cappellini 

The BB number in the input mlir files doesn't have any semantical meaning to us. We attach a BB number to each node just to let the dot printer to print visually separated subgraphs for different parts

Jiahui17, 2025-02-12:
I think if we use mlir Blocks we need to somehow follow the principle of SSA stuff: A block has some block entry points (block args) and exit points. Do you want to enforce that mux/merge/cmerge must be the entry points and branches must be the exit points?

pcineverdies, 2025-02-12:
> I think if we use mlir Blocks we need to somehow follow the principle of SSA stuff: A block has some block entry points (block args) and exit points

Thanks Jiahui for the concern: now that I think about it, I remember another MLIR issue that should be handled with this mechanism. I invite you to read [this message of mine](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991978) and [the answer from Lucas](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991979). @shundroid do you know if we can keep the current handshake semantics of "we can use a value before it is declared" by re-introducing basic blocks?

shundroid, 2025-02-12:
I see the main benefit of refactoring Handshake IR to represent CFG as making Fast Token Delivery easier to implement. So, I think we should decide the design based on how much it actually helps with that while ensuring it doesn’t interfere with existing optimizations or require excessive implementation effort.

I see two possible approaches:  

### 1. Using the SSACFG Region  

One option is to use an [SSACFG region](https://mlir.llvm.org/docs/LangRef/#control-flow-and-ssacfg-regions) to represent the CFG structure, similar to the `cf` dialect (as @pcineverdies suggests in the last message).  

However, this comes with a restriction: Operations cannot refer to values defined later in the same basic block (BB), or in another BB which they are not dominated by.

Because of this, we wouldn't be able to explicitly define `MuxOp` or `CMergeOp`, since their operands might come from BBs that do not dominate them.  

> @shundroid, do you know if we could maintain the current Handshake semantics—where values can be used before they are defined—by reintroducing basic blocks?  

My understanding is that this works fine in most cases except for Fast Token Delivery. The restrictions are naturally satisfied unless an optimization pass introduces backedges *inside* a BB.  

But I’m unsure how feasible this approach is specifically for Fast Token Delivery. I assume you'd need to add a GSA `mu` node ([comment](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991978)), which wouldn't be expressible under the restriction, or potentially a large number of `MuxOp`s, which aren’t typically explicitly defined in an SSACFG region.  

Please let me know if I'm misunderstanding something.

### 2. Representing the CFG Differently  

An alternative is the approach @pcineverdies proposed [here](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991986), where CFG information is stored as an attribute of the top-level `FuncOp`:  

```mlir
handshake.func @fir(...) -> (...) {
  ...
}{[{"src": 0, "dstT":1, "dstF": 1, "cond": ""},{"src": 1, "dstT":1, "dstF": 2, "cond": "cmpi0"},]}
```  

A key difference from the previous discussion is that we no longer hesitate to introduce CFG structure inside `FuncOp`.  

Then, another possible way is to nest [Graph Regions](https://mlir.llvm.org/docs/LangRef/#graph-regions), defining one graph region per basic block. The MLIR documentation suggests nesting as an option:  

[image]  

I’m not sure whether it is feasible to explicitly represent edges between BBs in this way, or how much it improves upon your current approach (to store CFG information as the attribute).  

---

If using an SSACFG region for Fast Token Delivery is feasible, I’d be happy to go with that.

So, could you provide insight into how well each approach supports Fast Token Delivery?

murphe67, 2025-02-12:
Yes, my original point "we should be using block notation" is wrong (sorry about that)- how exactly we represent the CFG is an open question.

I will rephrase my original point: we require the CFG to be strongly and cleverly encoded in the IR for both FTD and speculation. Any optimization that would break the CFG is incompatible with these 2 passes, and that includes our current "flattening" approach.

Hope I didn't derail the conversation too much with that! :grin: 

Possibly it is too late to change how we represent the CFG, and what @pcineverdies has implemented works well- if so I would like to see it respected more, and if we could use the same structure for the speculation algorithms.

pcineverdies, 2025-02-12:
I'll let you here some of the IRs from Fast Token Delivery: [gcd](https://github.com/pcineverdies/dynamatic/blob/FTD_RESULTS_3/results/gcd_FTD/comp/handshake_sq.mlir), [fir](https://github.com/pcineverdies/dynamatic/blob/FTD_RESULTS_3/results/fir_FTD/comp/handshake_sq.mlir), [kernel_2mm](https://github.com/pcineverdies/dynamatic/blob/FTD_RESULTS_3/results/kernel_2mm_FTD/comp/handshake_sq.mlir).

Is it really a problem in fast token delivery only? Whenever you have a multiplexer at the beginning of a BB (which used to be a block argument in the `cf` dialect) you are using as inputs:
- a value coming from a dominator from the node;
- a value coming from a non-dominator from the node.

Just imagine a snippet like this:

```
i = 0;
while( i < N) i++;
return i;
```

```
^BB0:
  %1 = 0;
  cf.br BB1(%1)

^BB1(%2):
  %3 = cmpi %2, N
  cf.cond_br %3, ^BB3(%2), ^BB4(%3)

^BB3(%4):
  %5 = %4 + 1
  cf.br ^BB1(%5)

^BB4(%6):
  return %6
```

(sorry for the MLIR pseudo-code) How do we handle `%2`? (Maybe it's trivial and I'm just dumb)

I guess that we don't really want to keep any block argument and instantiate the muxes/cmerges/merges as they are supposed to be in the final circuit. I expect MLIR not to be that much happy in a SSACFG region in the way you have mentioned. 

---

My way of storing and reintroducing the CFG structure works out alright for my needs. Maybe it's not ready to be merged into main and you might want to make it more robust, but we can discuss about it. I link you the files I am talking about: [.cpp](https://github.com/EPFL-LAP/dynamatic/blob/fpga23-queue-allocation/experimental/lib/Support/CFGAnnotation.cpp), [.h](https://github.com/EPFL-LAP/dynamatic/blob/fpga23-queue-allocation/experimental/include/experimental/Support/CFGAnnotation.h). 

Here is how I am currently using it in a transform pass after the conversion to `handshake` was done. 

```c++
ConversionPatternRewriter rewriter(ctx);

// Restore the cf structure to work on a structured IR
if (failed(cfg::restoreCfStructure(funcOp, rewriter)))
  return failure();

// ...

experimental::cfg::markBasicBlocks(funcOp, rewriter);

// Remove the blocks and terminators
if (failed(cfg::flattenFunction(funcOp)))
  return failure();

return success();
}
```

Which are the requirements for speculation with respect to the CFG?

murphe67, 2025-02-12:
> do you know if we can keep the current handshake semantics of "we can use a value before it is declared" by re-introducing basic blocks?

This seems (to me)to be the best long-term solution, but also probably a lot of work to re-implement SSACFG regions without the domination constraint: we definitely have some equivalent constraint on which variables can be passed to a mux, but it is slightly different than with sequential execution.

murphe67, 2025-02-12:
Reading the graph regions they seem kind-of built for our use case, but not super well supported- trying them out would definitely be an experiment.

> Which are the requirements for speculation with respect to the CFG?
We need to identify when control flow enters a bb, and we limit where speculation can reach differently based on different CFG shapes.

pcineverdies, 2025-02-12:
> I believe block arguments (or phi nodes) are necessary, but I don't think in most cases (including speculation) they need to be explicitly represented as muxes or control merges.

Without thinking about FTD, would this imply not instantiating block-header muxes and block-footer conditional branches until we move down to `hw`?

shundroid, 2025-02-12:
Yes, that’s what I meant. I think block-footer branches are necessary, just like in the cf dialect.
I might be missing something.

I was hoping FTD would work in this setting with SSACFG, but it doesn’t seem to (to me).

shundroid, 2025-02-12:
I realized just now that [Emmet's comment](https://github.com/EPFL-LAP/dynamatic/issues/285#issuecomment-2653571258) was progressively updated.

> So I don't look at MLIR often enough to write very accurate psuedo-code, but I am imagining something like this:
Image
Which is a kind of new structure unique to us that is not really a CFG, unclear if it would give us what we are looking for?

This is quite similar to what I was thinking of when I mentioned nested graph regions [here](https://github.com/EPFL-LAP/dynamatic/issues/285#issuecomment-2653360639).

(I thought you can embed the mux inside the basic block and don't have to pass the global vars as arguments maybe? Depending on the specification of scopes, which I'm not sure about)

So the result of inner graph region here is a handshake channel, which is passed from one BB to another.
I assume the inner graph region typically ends with:

```
%trueResult, %falseResult = cond_br %cond, %data
return %trueResult, %falseResult
```

murphe67, 2025-02-12:
Yes sorry I am doing a lot of processing in real time:

We can do an actual CFG I think:

[image]

shundroid, 2025-02-12:
I quite like this.

I'm not sure about the scopes, but we can ensure that each basic block operation (`bb1` `bb2`) or MuxOp or CMerge can only receive results from basic block operations by leveraging type verification maybe- specifying whether the channel/control is `global` or not, which means it is passed from one BB to another. For example, `!handshake.channel<i32, global>`.

This can ensure that the global variables represent CFG edges correctly by result-operand relationship.

murphe67, 2025-02-12:
Yes, MLIR would give us a lot of useful validation tools for it, for sure.

murphe67, 2025-02-12:
I think a good starting point would be a pair of passes that convert back and forth between this and what we have now, based on the bb attribute: most passes continue to use the current representation until we have more info?

shundroid, 2025-02-12:
I think this method also solves the issue of maintaining `handshake.name` or other attributes for flagging the exit of the basic block (see [comment 1](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991990) by @pcineverdies 
 and [comment 2](https://github.com/EPFL-LAP/dynamatic/discussions/253#discussioncomment-11991993) by @AyaElAkhras )

There's no need for additional attributes anymore—it's now handled simply by passing a value to the `return` op.

pcineverdies, 2025-02-12:
> I think a good starting point would be a pair of passes that convert back and forth between this and what we have now, based on the bb attribute: most passes continue to use the current representation until we have more info?

This is a very safe and conservative decision (refactor all the passes might take a while) and it gives us the possibility of experimenting with MLIR :)

I don't have a strong opinion over this graph region. Intuitively, without investigating its capabilities and limitations, it looks like a complicated solution for a minimal problem (I'm not against refactoring, but possibly limiting any non-necessary effort). I'd like to discuss with Aya the feasibility of this change in the context of FTD, but I'm pretty sure that we can find a way to make it work. 

I agree that the usage of `handshake.name` was a pitfall in my way of restoring the CFG. I should say that the main issues were resolved with #212 (this however highlights the limitations of the current naming analysis, but that's another story). 

> There's no need for additional attributes anymore—it's now handled simply by passing a value to the return op inside the basic block op.

However, the return operation should have some semantics for the values it returns. In this particular case, we need to know the value from the comparator which determines the decision of the branch. If we have a *list of return values*, we should be able to identify exactly which of these values is the one we are interested about. 

Also - but this becomes an implementation detail - it should be somehow simple to expose new values outside of the graph region. In Emmet's example, for instance, I might want (for any reason) to use `output` from `bb2` inside `bb3` without creating a new BB with the correct interface.

murphe67, 2025-02-12:
> Also - but this becomes an implementation detail - it should be somehow simple to expose new values outside of the graph region. In Emmet's example, for instance, I might want (for any reason) to use output from bb2 inside bb3 without creating a new BB with the correct interface.

Yes, I was wondering did this work for FTD: instead of the control tokens being BB parameters, you guys will have control signals passed from arbitrary points, right?

>  (I'm not against refactoring, but possibly limiting any non-necessary effort).
I agree- the issue I am worried about "is are we going to have 4 people reimplementing the CFG in 4 different ways over the next 2 years"? Our repeated use of the CFG means that at some point we will need a long-term solution, even if we do not actually implement it any time soon.

> However, the return operation should have some semantics for the values it returns. In this particular case, we need to know the value from the comparator which determines the decision of the [...]

This I don't understand well enough to be helpful- it looks (to me) ike you need both a usable CFG representation and an attribute, but maybe @shundroid can explain further?

shundroid, 2025-02-12:
> you need both a usable CFG representation and an attribute

Yes, I guess their requiremen is a CFG representation and an attribute to track an output from a comparator unit.

In this case, we could extend the `ResultOp` to have two types of operands: one for the special output from the comparator and another for general outputs (possibly a variadic operand). This way, you wouldn't need an additional attribute.

pcineverdies, 2025-02-12:
I had some time to think about the problem again. Here are my main concerns:

1. Effort. This looks like an enormous refactoring. All the passes at the handshake level need to be modified. All the buffering algorithms need to be modified. All the work currently under development needs to shift towards this new format. Unfortunately, I don't see any simple way to keep the current codebase coherent with this model. Would it really be worth it? (p.s. personally I'll slowly move on from the project in a couple of months, so it's not really a concern of mine)

2. The original problem. This issue rises from the problem of the CFG. We are thus trying to re-introduce the control blocks by adopting a new format. I am the first who wants to use `->getBlock()` anywhere in the handshake level, but this looks an extreme solution. I'd like you to consider my solution first as a common way to tackle the problem. (I have no attachment to it, but its main advantage is to be transparent to each pass that does not require it, and I am 99% sure that everything related to `CFG.h` can be restructured using those methods. HOWEVER, we can also say that what's done so far keeps `CFG.h`, while the following pieces are going to use the new CFG structure).

3. Modify things at the handshake level. There are multiple modifications that I had to do while FTD which seem really chaotic in this context. Referring to Emmet's pictures. Let's say I need to connect the output of `op2` into `bb3`. Right now I can just get its value and use it (which is what we would do with any HDL, just to remind the connection between the handshake level and a circuit). With this new approach,  I have to: add a block argument to `bb3`, add a return value to `bb2`, modify the return operation in `bb2` and finally use the value by picking the last argument from `bb3` (not necessary in this order, but you get my point). 
Now, what if I needed to use `ctrlOut1` in `bb3`? Do I have to check each time whether the value was already forwarded from `bb2` to `bb3`? Do I need to know which of the 2 arguments of `bb3` is connected to my value? Or which of the 3 results form `bb2` is it?
I hope you see my point here: we just get an enormous amount of problems when manipulating the IR - and we should aim for an environment maintainable and easy to develop on (mainly for the newcomers to start working fast on their projects).

pcineverdies, 2025-02-13:
I was also thinking that it might get difficult to navigate the IR this way, especially after materialization. 

What I usually do in my daily debugging life is to look for a component thanks to his name (`/mux20<Enter>`), go to its arguments, search them (`/%20<Enter>`) and see which are their producers. I think that this is quite of a standard approach for anyone who worked at the handshake level. 
With the graph region approach, this *backtracking* would get very tedious, as you always have to pass through the result of the bb, take note of the position of that argument (6th out of 20, for instance), go to the 6th result from the return statement of the block and start the backtrace again. 

Again, it's not like I don't want to change the way I work. I think I am always open to improvements. If tomorrow you tell me a better way to debug handshake IRs, I am super glad to introduce it to my workflow. This does not seem like an improvement though. 

I mentioned materialization because at this stage we need to move here and there many values (due to the forks, mainly), so you might eventually end up with something like this:

```mlir
%10, %11, %12, %13, %14, %15, %16, %17, %18, %19 = bb1(%20, %21, %22, %23, %24, %25, %26, %27, %28, %29, %30):

    return %40, %41, %42, %43, %44, %45, %46, %47, %48, %49
    ....
```

which is probably uselessly complicated to handle, and starts having little semantics with respect to the real circuit (I'd like to see the connections between components as they are going to be in hardware). 

---

Btw, I am also sure that this whole discussion is not intended to make some huge modification tomorrow or any time soon, as Emmett remarked (who has the time and energy to do it?). This CFG thing is clearly as issue that should be solved eventually, but it might be worth to investigate the problem more. 
For instance: right now FTD is working as I wanted it to work, but it is not compliant with FPGA'20. I don't have time to tackle the issue, but it's interesting to understand how we can integrate the two of them and what we need from an implementation point of view (the spurious connections in FTD are not coherent with the way the CFDFC are used).

shundroid, 2025-02-13:
> Effort. This looks like an enormous refactoring.

Agree.

> Modify things at the handshake level

> Let's say I need to connect the output of op2 into bb3. Right now I can just get its value and use it (which is what we would do with any HDL, just to remind the connection between the handshake [...]

I think the current state also has a downside—it’s risky that we can reference values across different BBs. In fact, the bug I found in the SpeculationPass last October happened because of this: the speculator was triggered by a control token from another BB.
If we can enforce restrictions, I think it would be beneficial in many cases. FTD might require more effort, but it still seems like a reasonable challenge. (Yes, this does go against the earlier assumption in the discussion that changing the CFG representation should benefit FTD.)

> so you might eventually end up with something like this:

```
%10, %11, %12, %13, %14, %15, %16, %17, %18, %19 = bb1(%20, %21, %22, %23, %24, %25, %26, %27, %28, %29, %30):

    return %40, %41, %42, %43, %44, %45, %46, %47, %48, %49
    ....
```

To me, the issue seems to be that the value isn’t named correctly. Unlike in Handshake IR, values in HW IR are named in a more human-friendly way. I think it should be possible to name these values.

https://mlir.llvm.org/docs/DefiningDialects/Assembly/#suggesting-ssablock-names

Also, I think we can omit arguments for `bb1`, since the inner region should be able to reference values defined outside of it.

So it would look like:

```
%trueResult_cond_br0, %falseResult_cond_br0, ... = bb1 {
    %trueResult, %falseResult = cond_br ... {handshake.name = "cond_br0"}
    return %trueResult, %falseResult
}
```

shundroid, 2025-02-13:
This is more of my opinion, so I'll leave a separate comment on it:

> The original problem

I'm working on speculation, and I think I encountered a process similar to FTD. Initially, I implemented a rough prototype by adding a `spec_tag` signal to each RTL port and confirmed that speculation works within the current Dynamatic framework. However, we didn’t merge this into main because it felt more like a temporary fix. So, we decided to take a more radical approach. We’ve already updated the Handshake IR type system to fully support extra signals, and now we’re working on replacing the existing backend to handle them properly.

Even though your current implementation seems more reasonable than my first speculation approach, and the solution we're discussing now might require even more drastic changes, I want to highlight that major updates like this often require two stages of implementation. I couldn't expect everyone to live with an ugly RTL design using `spec_tag` forever—maybe in the similar way we can’t assume everyone will always maintain the original operation’s name manually. For example, I remembered just now that SpeculationPass updates the names of a considerable number of operations—maybe I should fix that?

Unfortunately, the current Dynamatic framework isn’t designed in a way that makes this second-stage implementation easy.

murphe67, 2025-02-13:
`%trueResult_cond_br0, %falseResult_cond_br0, ... = bb1 {
    %trueResult, %falseResult = cond_br ... {handshake.name = "cond_br0"}
    return %trueResult, %falseResult
}`

Oh, also nice- so it is simply a question of exposing values.

> so you might eventually end up with something like this:

```
%10, %11, %12, %13, %14, %15, %16, %17, %18, %19 = bb1(%20, %21, %22, %23, %24, %25, %26, %27, %28, %29, %30):

    return %40, %41, %42, %43, %44, %45, %46, %47, %48, %49
    ....
```

I also don't really think this will happen (but I am guessing)- FTD is quite unique in exposing the control signals, and FPGA kernels don't have a huge amount of cross-BB variables. I don't think this problem requires incredibly good scaling behaviour.

> it’s risky that we can reference values across different BBs
Seconded- how to do this safely is **the** central question of FTD, right? The control signal generation seem relatively simple compared to passing the data tokens. Making cross-BB variables a little bit difficult could be a good idea.

Some other scatter-shot opinions: however we represent the CFG, it would be nice to generate it directly from CF, force all CFG-requiring passes to run near the top of the stack, and then discard the CFG halfway through compilation- stuff like buffering and term re-writing want an IR that corresponds pretty closely to RTL.

If we can achieve this, there is no reason to materialize the circuit until after this "flattening"- we are still in some kind of a "front-end" mode. 

This would make maybe developing new CFG-using passes more difficult, since they would need to get some kind of buffering to run, but feels like a much more logical place to me.

For @pcineverdies's general points about sending the control signals to different BBs- it is obviously more individual steps, but since they don't vary, they seem to be easy to wrap in a function? I also don't think it is a problem if a variable is returned multiple times from one BB, flattening and canonicalisation could handle this before buffering.

As a final point: we won't know if this will work until we try it, but I think any method of representing the CFG will be the same amount of work- if we cannot stand fully behind the post-hoc-cfg-reconstruction, we weaken two of our most powerful optimization passes.

rpirayadi, 2025-02-13:
> FTD is quite unique in exposing the control signals, and FPGA kernels don't have a huge amount of cross-BB variables. I don't think this problem requires incredibly good scaling behaviour.

I haven't followed the discussion carefully. But, I just wanted to mention that what I'm currently working on which attempts to sequentialize the memory accesses (ideally, only if necessary) relies on the completion signal of the memory access. In the case that the dependent memory access is in another BB, this signal must get transferred to the other BB. So, this is not just unique to FTD.