EPFL-LAP/dynamatic #293: [hls-verifier] Add cycle limit option for simulation

shundroid, 2025-02-15:
When debugging RTL, a bug can sometimes cause the simulation to run infinitely (e.g., circuit deadlock). To prevent this, I’ve been manually modifying the testbench VHDL file to enforce a maximum clock cycle limit. It would be helpful to have this as a built-in option in the tool.

I found that ModelSim allows setting a cycle limit without modifying the VHDL file by specifying the simulation time. Currently, the simulation script specifies `run -all`:

https://github.com/EPFL-LAP/dynamatic/blob/1b61c9a786f653cbfee48be8a38dff9332b45726/tools/hls-verifier/src/HlsVhdlVerification.cpp#L80

Instead, we could set a limit like `run 100ns`.
I confirmed this on my laptop and this approach also worked with the visualizer when the limit was reached.

I think this is a more practical way to implement cycle limits.

Jiahui17, 2025-02-17:
Good idea! Maybe we could estimate an upper bound latency from buffer placement and use that instead:

something like := sum for each inner-most loop: II * iteration * 2

shundroid, 2025-02-17:
That's interesting! I don't know (or consider) much about how to calculate the II of the loop or not. Is it easy? Maybe it's no longer a "good first issue" then 😅

shundroid, 2025-02-18:
I think it's easier. But is the max simulation time determined by the integration test itself? Maybe a single integration test is run with different configurations, like speculation enablement, buffer placement method, FTD, etc.

My initial assumption was that the max iteration would be specified as a script argument, like:
`> simulate --max=1000ns`

Jiahui17, 2025-02-18:
Normally the simulation would be terminated by the testbench itself: this indicates that the circuit normally exited the execution. Maybe we could set a `--max` flag and throw an error if this limit is reached while the circuit is still running (so something must have gone wrong or the simulation time is not long enough).

murphe67, 2025-02-18:
We run the integration tests with a python script right? If someone wants to specify how HLS happens, surely they can also specify a timeout value?

pcineverdies, 2025-02-19:
In my opinion, the best solution is to specify by hand an argument to the `simulate` command in the `.dyn` file, which is then moved down to the line Shun referenced. This way we can play around with the limit without risk of having something too strict. Exactly what Jihaui said. Then this can become the same way the integration script uses to notify a timeout (currently it just measures the runtime of the process).