EPFL-LAP/dynamatic #301: [Handshake] Remove HandshakeReshapeChannels pass and ReshapableChannelsInterface

shundroid, 2025-02-25:
The `HandshakeReshapeChannels` pass introduces `ReshapeOp` to merge multiple extra signals into one. For example, if a buffer processes data with two extra signals:

```
%dataOut = buffer %dataIn : !handshake.channel<i32, [spec: i1, tag: i8]>
```

It introduces `ReshapeOp` like this:

```
%dataInInner = reshape [MergeExtra] %dataIn : !handshake.channel<i32, [spec: i1, tag: i8]> to !handshake.channel<i32, [mergedDown: i9]>
%dataOutInner = buffer %dataInInner : !handshake.channel<i32, [mergedDown: i9]>
%dataOut = reshape [SplitExtra] %dataOutInner : !handshake.channel<i32, [mergedDown: i9]> to !handshake.channel<i32, [spec: i1, tag: i8]>
```

It even supports merging extra signals with the data itself:

```
%dataInInner = reshape [MergeData] %dataIn : !handshake.channel<i32, [spec: i1, tag: i8]> to !handshake.channel<i41>
%dataOutInner = buffer %dataInInner : !handshake.channel<i41>
%dataOut = reshape [SplitData] %dataOutInner : !handshake.channel<i41> to !handshake.channel<i32, [spec: i1, tag: i8]>
```

The `ReshapableChannelsInterface` is used for operations where *all* extra signals of *some* operands/results can be merged.

This pass was originally intended to simplify RTL generation. However, in practice, we are handling each extra signal separately in RTL without difficulty. Given that, I propose removing both this pass and the interface.

## Issues Beyond Just Being Unnecessary

There are additional issues beyond its redundancy:

- The pass is still active in the current compilation flow (`--handshake-reshape-channels`), meaning extra signals might be merged unexpectedly.  

https://github.com/EPFL-LAP/dynamatic/blob/1be4a411f37ddecde7c72f3fe60edf99276f13c9/tools/dynamatic/scripts/compile.sh#L173-L179

- Some operations shouldn’t merge their extra signals (e.g., most arithmetic operations like `addi`). The way extra signals from different operands should be merged at `join` varies per signal type, but this interface enforces a one-size-fits-all approach.

Additionally, the interface itself is poorly designed. As noted earlier, "`ReshapableChannelsInterface` is used for operations where *all* extra signals of *some* operands/results can be merged," which is an awkward definition. It doesn’t support selectively merging *some* extra signals (except for specific cases like `spec` in speculative units). Moreover, it imposes an unnecessary constraint that all operand/result types must match for merging, preventing operations like `ConstantOp` (ControlType input, ChannelType output) or `CmpIOp` (arbitrary bit-width input, always `i1` output) from using it. These limitations stem from a lack of consideration for real-world usage—something we should avoid in future designs.

Finally, from a practical standpoint, applying this kind of abstract transformation at the handshake level makes debugging harder. In waveforms, extra signals appear with abstract names (like "mergedDown"), obscuring their meaning. Keeping the generated RTL concrete offers significant debugging benefits, at least for now.

murphe67, 2025-02-25:
Yes, to me this pass is pretty vestigial at this point- the previously implemented features for extra signals are unfortunately a little bit distant from our existing methodology and use cases.

AyaElAkhras, 2025-02-25:
For fast token delivery and straight to the queue, I  see that we have it in the compilation script, but I don’t think it matters to us in any special way. I believe that whatever applies to the legacy Dynamatic circuits (i.e., the cfToHandshake pass in main) will also apply to us.

For the tagging and out-of-order work, we are still about to start working on it, so removing it at this point should cause no harm to us :)

@pcineverdies Tagging Giacomo just to confirm that I'm not missing something :)

pcineverdies, 2025-02-25:
I agree with Aya, clearly since FTD does not handle extra signals it should not be affected by such decision. This will only affect the implementation of the out-of-order infrastructure (so the earlier the better).

murphe67, 2025-02-25:
Ok, nice- thank you for the quick responses everyone!

shundroid, 2025-02-26:
I just noticed another downside of performing the reshape at the handshake level—we lose the ability to enforce `AllExtraSignalsMatch` for some [operations within a basic block](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/ExtraSignalsTypeVerification.md#operations-within-a-basic-block).

For example, `SelectOp` reshapes its two data inputs and output, which can cause a mismatch in extra signals between the index and data inputs after reshaping.

closed 2025-03-02 by shundroid