EPFL-LAP/dynamatic #304: [CI] Enforce Code Formatting

shundroid, 2025-02-25:
We're now using `clang-format` for C++ and `autopep8` for some Python files (`experimental/tools/unit-generators`), but formatting isn't currently enforced.

https://github.com/EPFL-LAP/dynamatic/blob/db21ab0cd6a63048213bdae1ccff5a3fba16ffb7/docs/GettingStarted.md?plain=1#L84

CI can check formatting and fail if it's incorrect. For example, `clang-format (file) --dry-run -Werror` fails on unformatted files. I’m not familiar with [autopep8](https://github.com/hhatto/autopep8), but it should have a similar option.

We might also consider `clang-tidy` for additional static analysis.

https://github.com/EPFL-LAP/dynamatic/blob/db21ab0cd6a63048213bdae1ccff5a3fba16ffb7/docs/GettingStarted.md?plain=1#L85

I expect existing code to fail the check, so we may need to fix it first.

An alternative is enforcing formatting via a pre-commit hook, but that would block temporary commits (meant to be squashed later). I think enforcing it at the pull request level is enough.

What do you think?
CC: @pcineverdies @ebosnjak (since you've recently worked on CI)

pcineverdies, 2025-02-25:
I definitely agree, this is among the operations the CI/CD should be in charge of.
It was also among the explicit requirements in #19, but somehow it passed away...
(So thanks for reminding us of the issue!)

I wonder what's the portion of the codebase which is not compliant with the standards we have... 
Unfortunately I cannot try it out now, but I don't think it's going to be a huge effort (probably some python files are off). 

I believe that the PR-level is more than enough, nobody really cares about what is going on before we go for a PR (I guess??)

schilkp, 2025-02-25:
I personally find pre-commit hooks super annoying and not trust-worthy (you can't rely on them actually being enforced as the allure of `--no-verify` is immense, you have to manually install them after cloning, to do them properly you have to rely on some pre-commit framework, ...) so CI is (in my opinion) the better solution :)

If you end up doing a large re-formatting commit, consider adding a `.git-blame-ignore-revs` to keep the git history nice. This is now also supported by github: https://docs.github.com/en/repositories/working-with-files/using-files/viewing-and-understanding-files#ignore-commits-in-the-blame-view

Clang-tidy can find a large number of bugs, but in my experience requires quite a large amount of maintenance and upkeep to deal with/ignore false positives etc. I think having it integrated so it is easy to run is definitely a good idea. It might also be nice to run it in CI, but I would not fail the CI pipeline if there are warnings.

pcineverdies, 2025-02-25:
> you end up doing a large re-formatting commit, consider adding a `.git-blame-ignore-revs` to keep the git history nice.

Thanks for mentioning this, I was slightly concerned about the possibility of messing up with the whole history (I thought it was inevitable tbh)

> It might also be nice to run it in CI, but I would not fail the CI pipeline if there are warnings.

Seems like a very good idea; it can log the info without being a mandatory step for the approval. Thanks for bringing this up :)

shundroid, 2025-02-25:
Thank you for your information! I think the implementation approach is clear now. (but don't forget to enforce Python formatting too!)

schilkp, 2025-02-26:
A less invasive approach might be something like this: https://github.com/llvm/llvm-project/blob/main/clang/tools/clang-format/clang-format-diff.py

It only clang-format checks the lines that were touched by a commit. I know many other large projects (for example the FreeRTOS kernel) do something similar.

That way "new code" is required to be formatted, but we don't need to do a bulk format of the whole codebase to get CI to be green.

ebosnjak, 2025-06-19:
Closing this as formatting checks were introduced with #318.

closed 2025-06-19 by ebosnjak