EPFL-LAP/dynamatic #309: Experimental contains malformed tests

schilkp, 2025-02-26:
I just noticed that the experimental test suite contains some test that seemingly fail not because of a code problem but simply because they miss a `// RUN:` line:

```bash
> ninja -C build check-dynamatic-experimenta
ninja: Entering directory `build'
[0/1] Running the Dynamatic regression tests for experimental work
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_rhs.mlir (1 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_rhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_lhs.mlir (2 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_lhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_rhs.mlir (3 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_rhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_lhs.mlir (4 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_lhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_lhs.mlir (5 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_lhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_rhs.mlir (6 of 8)
******************** TEST 'DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_rhs.mlir' FAILED ********************
Test has no 'RUN:' line
********************
********************
Unresolved Tests (6):
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_lhs.mlir
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/a_rhs.mlir
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_lhs.mlir
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/c_rhs.mlir
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_lhs.mlir
  DynamaticExperimental :: tools/elastic-miter-generator/rewrites/e_rhs.mlir

Testing Time: 0.03s
  Passed    : 2
  Unresolved: 6
FAILED: experimental/test/CMakeFiles/check-dynamatic-experimental /home/schilkp/st_hls/dynamatic/build/experimental/test/CMakeFiles/check-dynamatic-experimental
cd /home/schilkp/st_hls/dynamatic/build/experimental/test && /usr/bin/python3.13 /home/schilkp/st_hls/dynamatic/polygeist/llvm-project/build/./bin/llvm-lit -sv /home/schilkp/st_hls/dynamatic/build/experimental/test
ninja: build stopped: subcommand failed.
```

Are these known-broken or is this just an oversight? 

If they are not yet ready/meaningfull it might be worth excluding them for now to reduce clutter. If they are known-broken, it might be worth getting them to actually run to have a clearer reference of what the problem is.

They seem to have been added in https://github.com/EPFL-LAP/dynamatic/pull/259 :)

closed 2025-03-05 by cappellini

schilkp, 2025-03-05:
Thanks @cappellini :)

cappellini, 2025-03-05:
No problem, sorry for introducing the problem in the first place :)