EPFL-LAP/dynamatic #312: Dynamatic LIT-based test suite fails

schilkp, 2025-02-26:
Currently, on my machine, the LIT-based dynamatic test suite (`ninja -C build check-dynamatic`) fails:

```text
> ninja -C build check-dynamatic
... snip ...
Exit Code: 1

Command Output (stderr):
--
RUN: at line 2: /home/schilkp/st_hls/dynamatic/build/bin/dynamatic-opt --handshake-reshape-channels --remove-operation-names /home/schilkp/st_hls/dynamatic/test/Transforms/handshake-reshape-channels.mlir --split-input-file | /home/schilkp/st_hls/dynamatic/polygeist/llvm-project/build/bin/FileCheck /home/schilkp/st_hls/dynamatic/test/Transforms/handshake-reshape-channels.mlir
+ /home/schilkp/st_hls/dynamatic/build/bin/dynamatic-opt --handshake-reshape-channels --remove-operation-names /home/schilkp/st_hls/dynamatic/test/Transforms/handshake-reshape-channels.mlir --split-input-file
+ /home/schilkp/st_hls/dynamatic/polygeist/llvm-project/build/bin/FileCheck /home/schilkp/st_hls/dynamatic/test/Transforms/handshake-reshape-channels.mlir
within split at /home/schilkp/st_hls/dynamatic/test/Transforms/handshake-reshape-channels.mlir:34 offset :12:10: error: 'handshake.select' op failed to verify that all of {condition, trueValue, falseValue, result} have same extra signals
  %res = select %sel [%lhs, %rhs] : <i1>, <i32, [down: i2]>
         ^
... snip ...
Failed Tests (1):
  Dynamatic :: Transforms/handshake-reshape-channels.mlir

Testing Time: 0.07s
  Passed: 25
  Failed:  1
```

I bisected the issue to this commit https://github.com/EPFL-LAP/dynamatic/commit/468fcc5da886d2db40cb4b672e02fb927d621e8f but have not had time to dig into what exactly the issue is :) @shundroid  Do you mind having a look?

schilkp, 2025-02-26:
No worries no rush - just a heads up :)

I am opening an issue about CI because I also assumed that these checks would run :)

shundroid, 2025-03-02:
Sorry for the delay. I'll fix this by removing the file in #305.

shundroid, 2025-03-02:
Merged!

closed 2025-03-02 by shundroid