EPFL-LAP/dynamatic #313: [CI] Run LIT-based test-suites in CI

schilkp, 2025-02-26:
It seems that the LIT-based test suites (`ninja -C build check-dynamatic` and `ninja -C build check-dynamatic-experimental`) are not being run as part of the CI workflow, causing breaks to go un-noticed (#312 , #309)

It might be sensible to enable this. Something like:

```diff
diff --git a/.github/workflows/c-cpp.yml b/.github/workflows/c-cpp.yml
index dd213327..ec64453f 100644
--- a/.github/workflows/c-cpp.yml
+++ b/.github/workflows/c-cpp.yml
@@ -18,6 +18,14 @@ jobs:
 
       - name: build
         run: ./build.sh --release
+      
+      - name: check-dynamatic
+        run: | 
+          ninja -C build check-dynamatic
+
+      - name: check-dynamatic-experimental
+        run: | 
+          ninja -C build check-dynamatic-experimental
 
       - name: integration-test
         run: | 

```

Opening an issue first because there are a few points that should be defined:

- The simple approach above would not run, for example, integration tests if `check-dynamatic` fails. That is not a great developer experience: It would be nice to get a full CI report every time.

- I am not familiar with the self-hosted setup the CI runs on, but it seems there is caching setup? If so, it might be worth splitting the two checks into 3 separate CI flows so that all three get reported no matter how many break.

- It would be good to define a `policy` on known-broken tests: Do we keep them in-place to not loose track (but have CI be flashing red until fixed)? Do they get disabled, at the risk of being forgotten?

- Should this policy be the same for dynamatic and dynamatic-experimental?

schilkp, 2025-02-26:
This will also be relevant once https://github.com/EPFL-LAP/dynamatic/pull/307 lands - should that test suite be run in CI?

schilkp, 2025-02-26:
Shamelessly pinging @pcineverdies @ebosnjak because you have worked on the CI and might have inputs :)

pcineverdies, 2025-02-26:
It's been a while since I have talked to @ebosnjak ; this is also in-line with #304: both are definitely a must in the CI/CD!
I think it's worth on his side to maybe experiment among multiple options, and provide some alternatives before the implementation. 

Most of your concerns are technical details about the pipeline I am (unfortunately) unaware about :-( 

> It would be good to define a policy on known-broken tests: Do we keep them in-place to not loose track (but have CI be flashing red until fixed)? Do they get disabled, at the risk of being forgotten?

This is something I tried to handle some time ago in #248. Right now, only `invalid.mlir` is skipped, since it fails when Dynamatic is built in release mode. However, if we see that many of them should be skipped, we can modify that Python script to behave the same way the the integration test works - with a text file to list them as long as they do not work.

schilkp, 2025-02-26:
@Jiahui17 @murphe67 does one of you know the details about the self-hosted CI runner? Is someone maintaining that?

ebosnjak, 2025-02-26:
Hello, I am in charge of the runner. Let me address what you said.

> "The simple approach above would not run, for example, integration tests if check-dynamatic fails."

Internet tells me that [there are ways to handle this](https://stackoverflow.com/questions/58858429/how-to-run-a-github-actions-step-even-if-the-previous-step-fails-while-still-f).

> If so, it might be worth splitting the two checks into 3 separate CI flows so that all three get reported no matter how many break.

Does the point above solve this problem?

shundroid, 2025-02-26:
I like @schilkp's second approach:

> I am not familiar with the self-hosted setup the CI runs on, but it seems there is caching setup? If so, it might be worth splitting the two checks into 3 separate CI flows so that all three get reported no matter how many break.

LLVM and Polygeist aren't rebuilt each time, only Dynamatic, which is fast (< 3min).
So how about splitting the checks into separate jobs?

Also, the current job is named `integration`, which seems meant for integration tests.
[image]

ebosnjak, 2025-02-26:
> So how about splitting the checks into separate jobs?

Maybe I'm missing something, but I don't see the benefit of having multiple jobs instead of steps, since they need to run in the same environment (hence, on the same runner; the only one we have), [see here](https://docs.github.com/en/actions/about-github-actions/understanding-github-actions#jobs). Regardless of jobs vs steps, I don't really see the purpose of running ninja checks in parallel, since they are really short and fast, especially compared to integration tests.

> Also, the current job is named integration, which seems meant for integration tests.

The integration tests were kind of the most important part, so I guess that's why I put that name. I don't really have a better idea, so please let me know if you have some suggestions.

> but it seems there is caching setup? 

It's not really a caching setup; it's just that the runner works basically as we all do on our local machines, in the sense that it doesn't clean the build outputs, it just checks out that branch, builds and runs. So once polygeist gets built once, it will not be rebuilt, unless something weird happens like it did a few weeks ago when it randomly decided to rebuild it.

schilkp, 2025-02-27:
> LLVM and Polygeist aren't rebuilt each time, only Dynamatic, which is fast (< 3min). So how about splitting the checks into separate jobs? Edit: on second thought maybe 3min is still long...

In my books 3min is plenty fast for CI. 

> It's not really a caching setup; it's just that the runner works basically as we all do on our local machines, in the sense that it doesn't clean the build outputs, it just checks out that branch, [...]

Got it - I assumed as much. Sorry if I was being imprecise. I was less talking about a dedicated "caching setup", more making sure that the build progress is maintained across runs - whether that is through a cache, storing/fetching build artifacts during CI, or maintaining them in place like is done.

> Maybe I'm missing something, but I don't see the benefit of having multiple jobs instead of steps, since they need to run in the same environment (hence, on the same runner; the only one we have), [...]

Actually for me this is absolutely not about speed, but about usability:  If they are separate jobs, they will show up as such in the PR. Example from XLS:

[image]

This would make it super easy to see, at a glance, what is failing: "Oh the integration tests and normal LIT test passed but there is an error in the experimental LIT test and a formatting error". Anyway if this is a physical runner the jobs should be configured to run sequentially.

Separating formatting into its own job is, in my eyes, very sensible because the overhead for one additional clone before checking formatting is on the overhead of a few seconds:

[image]

Since you have some form of maintaining build state across jobs, also splitting the tests into multiple jobs is pretty "cheap": At worst, you have to re-build dynamatic which takes about 3min.

If that is still a bottleneck - is ccache setup on that machine? I would assume that it would greatly speed-up the rebuild of dynamatic.

schilkp, 2025-02-27:
A question about https://github.com/EPFL-LAP/dynamatic/pull/315 (keeping it here to not split the discussion):

With this setup, if one of `check-dynamatic` or `check-dynamatic-experimental` fails, it continues on to the integration tests but does the CI run get marked as a fail or pass?

schilkp, 2025-02-27:
This is of course also relevant for https://github.com/EPFL-LAP/dynamatic/pull/318 - just saw you already opened that :)

ebosnjak, 2025-02-27:
> With this setup, if one of check-dynamatic or check-dynamatic-experimental fails, it continues on to the integration tests but does the CI run get marked as a fail or pass?

It depends on which "trick" is used to continue running after an error happens. Compare [the first run](https://github.com/EPFL-LAP/dynamatic/actions/runs/13548856862/job/37867252004) vs [the second run](https://github.com/EPFL-LAP/dynamatic/actions/runs/13549214547/job/37868469880). So, it is a matter of defining what we want to happen.

shundroid, 2025-02-27:
I also prefer multiple jobs, mainly for the usability benefits @schilkp mentioned. But I also see a strong advantage in distributing jobs across different runners.

I'm not sure why you're using a self-hosted runner currently—do you need a commercial simulator for integration tests? If so, unit tests and formatting checks don’t require it. You can separate them into different jobs and run them on GitHub runners instead.

If the build process is heavy and that's why you're using a self-hosted runner, formatting checks don’t even need a build. You could offload them to a GitHub runner to reduce the load.

ebosnjak, 2025-02-27:
> I'm not sure why you're using a self-hosted runner currently—do you need a commercial simulator for integration tests?

Yes, Modelsim is required for running integration tests, and also Gurobi for some `compile` options in Dynamatic. 

> If so, unit tests and formatting checks don’t require it. You can separate them into different jobs and run them on GitHub runners instead. If the build process is heavy and that's why you're using a self-hosted runner, formatting checks don’t even need a build.

As far as unit tests go, the problem is the build process, as you said. Github's runners are ephemeral, which would mean rebuilding everything on every run, and we've all seen how long the runs take when polygeist needs to be rebuilt. Not only that, but the build process would be run in the Github runner and in the self-hosted runner (to run the integration tests), which would be quite wasteful. But I can separate them into different jobs to have the nice overview (like the one in @schilkp's image), I just need to make sure that the build outputs are not lost between jobs.

And regarding the formatting check, it can be done in a Github runner easily, but the performance effects of this are questionable for me, at best. The formatting check is something that is really fast; it took 4 seconds on my last run. However, the Github runner would also need to checkout the branch and clean up (the first and final steps in every run). Looking at my last run, these "overhead" steps took 7 seconds in total. That would mean that we would have sped up the self-hosted runner's task by 4 seconds, but we would have another runner (although in parallel) running for 11 seconds, and that just seems wasteful, again. Not to mention that a few seconds are completely irrelevant in terms of an Actions workflow that takes time on the order of magnitude of tens of minutes. 

So, to conclude, I agree with @schilkp that I should split it into multiple jobs to make it prettier. On the other hand, I really don't see the purpose of moving some parts to Github's runners, for the reasons I gave above. I might be missing something, so please correct me if I'm wrong.

schilkp, 2025-02-27:
> > I'm not sure why you're using a self-hosted runner currently—do you need a commercial simulator for integration tests?
> 
> Yes, Modelsim is required for running integration tests, and also Gurobi for some `compile` options in Dynamatic.

I think sticking on a self-hosted runner because of modelsim and gurobi is sensible :)

> As far as unit tests go, the problem is the build process, as you said. Github's runners are ephemeral, which would mean rebuilding everything on every run, and we've all seen how long the runs take [...]

Note that ephemeral runners and maintaining build state are not mutually exclusive. There are ways to store the build state at the end of a CI run, and restore it again at the beginning.

For example, Google's XLS is a much bigger project (500'000+ lines of C++) with some super heavy dependencies (including MLIR + LLVM), and they still run on ephemeral github runners with manual build cache management:

https://github.com/google/xls/blob/64d29a01e5acd66b2d3008aae8de58c14ee88f4c/.github/workflows/continuous-integration.yml#L31

To be fair this is for a different build system, but it also exists for cmake-based builds. 

> Not only that, but the build process would be run in the Github runner and in the self-hosted runner (to run the integration tests), which would be quite wasteful. But I can separate them into [...]

As noted above, I agree that the keeping everything on self-hosted but splitting it into jobs is the easiest way forward.

---

Not related to this PR, but in general related to CI: I am a little uneasy about the way CI state is maintained between runs. It can easily lead to consecutive CI jobs interfering with each other, by carrying forward unintended state:

For example, right now, the release flag the CI job passes to the build script (`./build.sh --release`) is actually ignored in the CI workflow in most cases. The build script only re-configures cmake if either the `./build/CMakeCache.txt` file is missing, or the `--force` flag is specified. Because of the way the runner maintains state, the `CMakeCache.txt` is always there and cmake configure is never called.

In short, the CI jobs only run in release mode because the previous CI job ran in release mode, which in turn only ran in release mode because the previous CI job ran in release mode, .... 

If I were to add a PR that forced the CI to a debug build and ran the CI, all future CI runs would be in debug mode.

This is one specific example that I have noticed / traced-out, but interdependence between CI runs can lead to super subtle and hard to track-down issue. 

Another place this could rear it's ugly head is if we want to actually test the XLS integration (which requires a different build config) in CI. With the current setup we would need to either force a reconfigure which leads to a recompile (bad), or maintain separate build folders for separate build configs. Otherwise, if I ever run a single CI job with XLS enabled, it will stay enabled for all other CI jobs...

Even if we run on the self-hosted runner (which, again, I support), I think it might be worth considering an approach that completely removes inter-CI-run state and relies on cmake's/ninja's dependency analysis to decide if the build files from the previous run are still valid or need to be re-built.  

Note that part of the issue here is the CI management, but also part of the issue is the limitations introduced by the `./build.sh` script: With a "normal" cmake workflow, configuring and building are different commands. With the uniform `./build.sh` script there is no clear seperation.

schilkp, 2025-02-27:
Maybe that second part should be it's own issue - sorry for the brain dump. Pinging @Jiahui17 because we talked about this today :)

shundroid, 2025-02-27:
> Github's runners are ephemeral, which would mean rebuilding everything on every run

Can't we use Github's cache action?

https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/caching-dependencies-to-speed-up-workflows

Also, I’m generally not in favor of relying on a single self-hosted runner—it means Dynamatic is only tested in a potentially highly customized environment.

---

I think I put too much emphasis on using GitHub runners. But my point was more generally about splitting jobs and distributing them across different runners instead of maintaining a monolithic job only executed on one runner. We might have multiple self-hosted runners in the future (e.g., one at EPFL and one at ETH). Splitting integration tests by category could also help.

schilkp, 2025-02-27:
> Also, I’m generally not in favor of relying on a single self-hosted runner—it means Dynamatic is only tested in a potentially highly customized environment.

I don't think the self-hosted runner is the issue per-se. There is little way around it due to gurobi and modelsim. But it would be good to standardize the reset of the environment. @Jiahui17 suggested a docker container.

I will try and sum all this up into a separate "CI/Infrastructure Issue" tomorrow - Let's keep the scope here limited to just adding multiple jobs for the different tests.

ebosnjak, 2025-02-28:
@schilkp I absolutely agree with you on the issue of state in the CI. In fact, when your xls-integration broke my clang-format today, I completely missed what had happened in the repo and I thought that I was having a major issue with random files from previous runs being kept for some reason. Luckily that wasn't the case, but it got me thinking generally in the direction of your message above. So yes, you are completely right that the current way that things are handled on the runner are kind of unfortunate :c 

We can discuss in more detail once you open the "infrastructure issue" you mentioned, but I would like to get the formatting and ninja/jobs issues out of the way first, since this matter is quite non-trivial compared to running clang-format.

schilkp, 2025-02-28:
> We can discuss in more detail once you open the "infrastructure issue" you mentioned, but I would like to get the formatting and ninja/jobs issues out of the way first, since this matter is quite non-trivial compared to running clang-format.

Absolutely. Sorry to derail this issue.

ebosnjak, 2025-02-28:
No worries, I'm glad that we're sorting this out!

closed 2025-03-27 by ebosnjak