EPFL-LAP/dynamatic #324: [Simulator] Runtime Assertion

shundroid, 2025-03-02:
When a simulation fails and I check the waveform, I find myself following almost the same debugging steps every time:

- **Handshake violation inspection**: Some complex units (like the speculator) have non-trivial handshake logic. For example, the speculator used to modify its output value while holding `valid = 1`, losing a speculative token and causing a token mismatch.
- **Unknown (`x`) bit inspection**: Sometimes, units emit unknown bits (`x` in ModelSim), which is usually a bug.
- **Deadlock cause inspection**: Some units never set their input `valid` signal to `1`, and I need to identify which unit is responsible. This is actually the most frequent debugging case for me.
- **Domain-specific rule checks**:
  - In speculation, buffers inside the speculator and save-commit units should be empty after all mis-spec tokens are killed. If not, it leads to a token mismatch, which has been a frequent bug. Debugging this has been particularly hard, because I had to track the movement of Head/Current/Tail pointers.

I have a lot of this kind of checks that I’d like to be asserted at runtime, which I believe would make debugging much easier. Most of them could likely be implemented as a simple Python script that reads the waveform file after a failed simulation.

What do you think? (CC: @murphe67 @Jiahui17)

shundroid, 2025-03-02:
I just encountered a bug caused by an overflow in the cyclic buffer inside a save-commit unit. To efficiently find this, I will try to do the following:

1. Convert the .wlf waveform to .vcd (using `wlf2vcd`, though this might be possible directly within the ModelSim flow).
2. Read the .vcd waveform using the `pyvcd` Python library (Here we use a third-party library).
3. Analyze the head/tail pointers and assert that no overflow occurred.

Jiahui17, 2025-03-03:
@shundroid this sounds like the original project proposal that we talked about 8 months ago...:)

shundroid, 2025-03-03:
Wow, I just remembered it 😂 But it seems this really helps
Maybe I'll develop a domain-specific version for me, to detect speculation bugs