EPFL-LAP/dynamatic #328: The light green color used for buffers in the dot file may not render correctly.

qianxu1998, 2025-03-05:
Currently, in the final `dot` generated after compiling the benchmark, all `buffers` are assigned the color `lightgreen`, which can't be rendered correctly in the generated `png` file, like the following:

[image]

Changing the color to `green` will fix this issue.

shundroid, 2025-03-09:
I remember I had the same issue but I solved it and I forgot what I did😅
I think it's an issue of graphviz. Maybe you can try things written here: https://gitlab.com/graphviz/graphviz/-/issues/1446

shundroid, 2025-03-10:
After I carefully checked the thread, I realized:

> if the graphviz version is <=2.40.1 the lightgreen color is not available ([graphviz/graphviz#1446 (closed)](https://gitlab.com/graphviz/graphviz/-/issues/1446))
https://gitlab.com/sardana-org/sardana/-/merge_requests/1961#note_1721557654

Yeah I think avoiding lightgreen is a good way for this.

PR #350 (2025-03-17): [export-dot] Compatible with graphviz <= 2.40.1
The color of buffers: lightgreen -> green

fixes #328
commit 91286b06 (2025-03-17): [export-dot] Compatible with graphviz <= 2.40.1 (#350)

The color of buffers: lightgreen -> green

fixes #328
closed 2025-03-17 by qianxu1998