EPFL-LAP/dynamatic #329: Generate the final PNG file displaying operator names explicitly, rather than an abstracted node view.

qianxu1998, 2025-03-05:
Currently, the final `png` graph is like:

[image]

Each node in the graph has an abstracted node view instead of the actual node name:

It would also be beneficial to generate a graph showing actual node names, helping users clearly understand the connections between nodes, as illustrated below:

[image]

murphe67, 2025-03-07:
@shundroid you mentioned something about this already being supported?

shundroid, 2025-03-07:
Yes, thank you for reminding me😄

You can specify `--label-type=uname` to export-dot and the graph looks exactly like your suggestion.

Hope this helps!

closed 2025-03-07 by qianxu1998