EPFL-LAP/dynamatic #336: [TypeSystem] Create HandshakeType as a type, instead of ChannelType and ControlType

shundroid, 2025-03-10:
Currently, there are two types ChannelType and ControlType, defined in tablegen.

https://github.com/EPFL-LAP/dynamatic/blob/fb5791399e476868d5e9a5843a92e0890488763e/include/dynamatic/Dialect/Handshake/HandshakeTypes.td#L37-L39

https://github.com/EPFL-LAP/dynamatic/blob/fb5791399e476868d5e9a5843a92e0890488763e/include/dynamatic/Dialect/Handshake/HandshakeTypes.td#L65-L67

These are defined as types, and can be used as C++ class.

We have `HandshakeType` as a **type constraint**, defined as ChannelType+ControlType, which can be only used from tablegen:

murphe67, 2025-03-11:
I am a bit scared by the idea of nullptr's floating around- what are the possibilities of representing the control type in this field?

(forgive my incorrect type formatting but)
!handshakeType<i5>, !handshakeType<f64>, !handshakeType<x0>

murphe67, 2025-03-11:
Additional question: 
how would this effect attempting to find the control token for a BB, i.e. forking the trigger token for the speculator?

shundroid, 2025-03-11:
Since we don’t directly access the raw pointer, specifying `nullptr` here should be fine to me. But to be safe, we could define a dedicated type for `ControlType`’s data, such as a custom void-like type or `mlir::IntegerType` with bitwidth 0 (which is valid—I checked).

For the type’s assembly format, we can still use a shorter form for control-like `HandshakeType`, keeping the parsing logic concise.

> how would this effect attempting to find the control token for a BB, i.e. forking the trigger token for the speculator?

Good point. In that case, we’re still traversing branches and checking whether a type is control-like, rather than just looking for `ControlType`-typed operands:

https://github.com/EPFL-LAP/dynamatic/blob/8cf6142c31bf65b0f8c88a4f92c35520bc6fb5e5/experimental/lib/Transforms/Speculation/HandshakeSpeculation.cpp#L440-L456

We can simply replace `isa<ControlType>` with `.cast<HandshakeType>().isControl()`.

shundroid, 2025-03-11:
I heard @Jiahui17 is also in favor of IntegerType with bitwidth 0.

@pcineverdies excuse me, do you have any opinion on this?

shundroid, 2025-03-11:
@murphe67 I heard from you Lana asked us about the renaming of Handshake/Channel/Control. I think we can do it here when we unify types.

Jiahui17, 2025-03-11:
>    how would this effect attempting to find the control token for a BB, i.e. forking the trigger token for the speculator?

I think the strongest point is that ControlType is not used any differently from ChannelType in Dynamatic and this extra annotated info (the channel is a ControlType instead of a ChannelType with i0) is not that useful: (I guess) you don't even need to check if the channel is a zero-width channel to correctly extract the whole control network (you can extract it by just looking at the topology/unit types).

AyaElAkhras, 2025-03-12:
From my perspective, I think it makes perfect sense to unify them and rely on data bitwidth. Thanks for the clear description of the problem and solution! :)

pcineverdies, 2025-03-12:
Thanks Shun! I do not see any problem for the experience I got by working at the handshake level. I briefly skimmed through the codebase and It does not seem like such distinction is effectively used in many places...

shundroid, 2025-03-12:
Thanks for your input, everyone! I'm open to making changes when I have time, but I suspect this might conflict with ongoing updates. Maybe I should give everyone a heads-up first.

lana555, 2025-03-13:
Thanks for the detailed description, sounds good! :)) 

> @murphe67 I heard from you Lana asked us about the renaming of Handshake/Channel/Control. I think we can do it here when we unify types.

I don't know if it is just me, but I did not find the names "channel" and "control" particularly intuitive... It's a detail, though :)

murphe67, 2025-03-17:
The extra signals type system effectively did not exist when we began in December, the control vs. channel types is the last remaining contribution that we have not removed or rewritten.

I am unclear what especially you are referring to as "the extra signal signal handling referred in spec", but this document is basically 100% obsolete. 😅

shundroid, 2025-03-17:
@paolo-ienne Regarding the documentation on type systems you mentioned, I honestly don’t think it explains well "how to carry extra bits." Aside from bundle/unbundle, it mostly focuses on IR serialization as you said, which is just about how types are printed and nearly nothing to do with the type system. It also lacks key aspects like type verification.

Instead, we wrote [a separate document](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/ExtraSignalsTypeVerification.md) focusing on type verification. It also covers the basics of our type system, including ChannelType, ControlType, and extra signals ([see here](https://github.com/EPFL-LAP/dynamatic/blob/main/docs/Specs/ExtraSignalsTypeVerification.md#types)).

If you haven’t seen it yet, we’d really appreciate it if you could take a look and share any suggestions. It may not be fully standalone, as it still refers to the original documentation for basic context (in the first paragraph). We used figures instead of MLIR serialization, but if you’d prefer seeing the actual IR, please let us know.

One issue with the old documentation is that it’s quite abstract (which also ties into [my comment in #348](https://github.com/EPFL-LAP/dynamatic/discussions/348#discussioncomment-12530148)). In contrast, our new documentation is based on the actual implementation and provides concrete reasoning for each design decision. We’d love to hear your thoughts on that as well.

---

I’ve read your message 10 times, but I still don’t see how it relates to this topic—sorry if I’m missing something. Could you clarify what you mean by "the current proposal" in this issue?

murphe67, 2025-03-18:
(I also am not sure this conversation has anything to do with @shundroid 's proposed change.)

> Aside from bundle/unbundle, it mostly focuses on IR serialization as you said, which is just about how types are printed and nearly nothing to do with the type system.

I would echo this- while the serialization is not un-important, the "code" impact is limited to basically only the printers and parsers. (Though we did completely rewrite the parser)

What's much more important is how the type system is represented inside of a pass, which the original specification document doesn't mention.

(A good chunk of what was proposed in that document didn't work for speculation, and so what technical details are mentioned are mostly no longer true 😅)