EPFL-LAP/dynamatic #338: [CI] Avoid inter-run dependency

schilkp, 2025-03-11:
This was discussed in https://github.com/EPFL-LAP/dynamatic/issues/313 but I am pulling it out to a different issue.

I am a little uneasy about the way CI state is maintained between runs. It can easily lead to consecutive CI jobs interfering with each other, by carrying forward unintended state:

For example, right now, the release flag the CI job passes to the build script (`./build.sh --release`) is actually ignored in the CI workflow in most cases. The build script only re-configures cmake if either the `./build/CMakeCache.txt` file is missing, or the `--force` flag is specified. Because of the way the runner maintains state, the `CMakeCache.txt` is always there and cmake configure is never called.

In short, the CI jobs only run in release mode because the previous CI job ran in release mode, which in turn only ran in release mode because the previous CI job ran in release mode, ....

If I were to add a PR that forced the CI to a debug build and ran the CI, all future CI runs would be in debug mode.

This is one specific example that I have noticed / traced-out, but interdependence between CI runs can lead to super subtle and hard to track-down issue.

Another place this could rear it's ugly head is if we want to actually test the XLS integration (which requires a different build config) in CI. With the current setup we would need to either force a reconfigure which leads to a recompile (bad), or maintain separate build folders for separate build configs. Otherwise, if I ever run a single CI job with XLS enabled, it will stay enabled for all other CI jobs...

Even if we run on the self-hosted runner (which, again, I support), I think it might be worth considering an approach that completely removes inter-CI-run state and relies on cmake's/ninja's dependency analysis to decide if the build files from the previous run are still valid or need to be re-built.

I don't have much experience here, but there are many large C++ projects out there. It might be worth checking out how they do inter-run build caching.

Note that part of the issue here is the CI management, but also part of the issue is the limitations introduced by the `./build.sh` script: With a "normal" cmake workflow, configuring and building are different commands. With the uniform `./build.sh` script there is no clear separation.

schilkp, 2025-03-15:
With @shundroid contribution https://github.com/EPFL-LAP/dynamatic/pulls this should be significantly easier to do now:

On my machine, deleting the build folder and re-building (which forces a CMake reconfigure by the build script) can be done in around 8 seconds:

[image]

(Assuming that ccache is installed,  well-soaked, and that the ccache is maintained between runs)

ebosnjak, 2025-08-22:
This is now resolved, by means of completely emptying the workspace folder before running the CI. Actions/checkout runs `git clean -ffdx`, which should clean all untracked contents (e.g. leftover build artifacts), so I'm not sure why these leftover state issues were occurring in the first place. As you said above, ccache handles caching of compilation artifacts locally, so the build is relatively fast.

closed 2025-08-22 by ebosnjak