EPFL-LAP/dynamatic #342: [Refactoring] All Buffers Should Be Placed by the Buffer Placement Algorithm

kb29x37, 2025-03-13:
Some components have transparent buffers in their implementation (Multiplexers, Merges, CMerges, see example afterward). To simplify the flow and isolate the buffer placement pass better, should we not let the it place these buffers instead if they are necessary? 

[image]

Jiahui17, 2025-03-14:
Two reminders:
- The TEHB is required only for Merge/CMerge and not Mux (we should remove it at some point).
- the Load ports need a buffer slot. (there is an example in #136

The TEHB are for two different purposes:
- In Merge/CMerge: avoid token reordering
- In the load port / sharing wrapper: avoid resource sharing deadlock

Now I feel it is more natural to go this way:
The buffer slot that manages sharing is always invisible to the circuit side (i.e., when designing the circuit's side, you don't need to worry about how many slots to put to avoid sharing deadlock). And it is up to the "wrapper" that manages the sharing to add those slots. For instance, the memory controller, LSQ, or sharing wrapper.

kb29x37, 2025-03-14:
Hi Jiahui,

Thanks, I did not know all of this ;)

On my end, the only one I really care about is the one in the Mux... Maybe the issue could be renamed to "[Refactoring] Removing the TEHBs Inside Muxs" ?

Thanks in any case !

Jiahui17, 2025-03-14:
> Thanks, I did not know all of this ;)

> On my end, the only one I really care about is the one in the Mux... Maybe the issue could be renamed to "[Refactoring] Removing the TEHBs Inside Muxs" ?

Maybe just do a pull request directly, I think it is still useful to have exactly this discussion here.

QinYuan2000, 2025-03-21:
> * In Merge/CMerge: avoid token reordering

@Jiahui17 Hello Jiahui. 

I’m not clear about this problem. Does it mean that for a 2-inport merge unit, the token from inport number 2 arrives first but is blocked, then in the next cycle the token from inport 1 arrives and the merge unit receives the later token according to its for loop? If that’s the case, the MILPs are not aware of it. They only balance token occupancies to optimize throughput and use as few buffer slots as possible. It seems that we should tell the MILPs to put one more slot than they actually need to optimize throughput.

QinYuan2000, 2025-03-21:
Furthermore, according to the current buffer placement order (which saves a lot of area and is preferably not changed), the buffer distribution on the channel is as follows:

Producer → OEHB (if present) → TEHB (if present) → Consumer

For the merge unit with an internal TEHB:

Merge → Internal TEHB → OEHB (if present) → TEHB (if present) → Consumer

If we remove the internal TEHB and rely on the buffer placement algorithm to add it:

Merge → OEHB (if present) → TEHB (must exist) → Consumer

Now, the buffer after the merge is OEHB rather than TEHB. To analyze the impact of this change, we need to fully understand how to eliminate the reordering problem with the fewest slots.

QinYuan2000, 2025-03-22:
> We don't have 2-input merges in any MILP

So, why is there a token reordering problem? It shouldn’t happen if handshake signals are present.

Jiahui17, 2025-03-22:
[image]

Here is an example: 

- the token enters the loop through the left input of the merge
- there is no buffer before the merge and the first eager fork

Suppose the first eager fork is backpressured by one of its outputs, but not backpressured by the output that circulates the token back to the right input of the merge. Then, there is a risk that the fork duplicates the token and passes it to the right input of the merge while there is still an incoming token to the left input of the merge. And merge might reorder these two tokens.

But if we always make sure that there is a buffer in between the merge and the first eagerfork below it, there is no such problem:

[image]

The point that I wanted to make here and also in #356 is that I don't see why removing this buffer causes a performance drop.

QinYuan2000, 2025-03-22:
@Jiahui17. Thank you for clarifying the issue! Now I can investigate the cause of the throughput degradation in the right direction.

QinYuan2000, 2025-03-24:
Does it make sense to say that ensuring at least one buffer—either an OEHB or a TEHB—between Merge and EFork would be sufficient to handle this condition?

Jiahui17, 2025-03-24:
@QinYuan2000 or any buffer with >= 1 slots.

shundroid, 2025-03-29:
Thanks for working on this! We just ran into the exact same issue (in thought) that @Jiahui17 described in [this comment](https://github.com/EPFL-LAP/dynamatic/issues/342#issuecomment-2745567695). I'm fine with removing internal buffers, but I think the external buffers needed for the unit to function properly should be explicitly stated in the form of op declaration. This isn't urgent, so I don't want it included in PR #356. I’ll open an issue instead.

QinYuan2000, 2025-05-16:
Here is the description of #356, which solves this issue:

## Motivation

Previously, Merge/ControlMerge/Mux units had an internal TEHB slot on their HDL output ports. This TEHB influences circuit performance, but the buffer placement MILPs are unaware of it. To ensure that the MILP models accurately describe circuit performance, they use a variable `minTrans`—indicating the minimal transparent buffer slot number—to enforce at least one TEHB on the output channels of Merge/ControlMerge/Mux units, thereby mimicking the internal TEHB behavior.

Previously, the value of `minTrans` could be assigned in two ways. The first way is to find all Merge/ControlMerge/Mux units and assign it on their output channels. For example:

```c++
// Merges with more than one input should have at least a transparent slot
// at their output
for (handshake::MergeOp mergeOp : funcOp.getOps<handshake::MergeOp>()) {
  if (mergeOp->getNumOperands() > 1) {
    Channel channel(mergeOp.getResult(), true);
    channel.props->minTrans = std::max(channel.props->minTrans, 1U);
  }
}
```

The second way is to specify the minimum buffer slot number for Merge/ControlMerge/Mux in the JSON file and then add this value to `minTrans` as follows:

```c++
void Channel::addInternalBuffers(const TimingDatabase &timingDB) {
  // Add slots present at the source unit's output ports
  if (const TimingModel *model = timingDB.getModel(producer)) {
    props->minTrans += model->outputModel.transparentSlots;
    props->minOpaque += model->outputModel.opaqueSlots;
  }
```

This approach forces the MILP model to place extra TEHBs to correctly reflect the behavior of the internal TEHB. After solving the MILP, the buffer placement result should deduct the extra TEHB (the same number specified in the JSON file) because the internal TEHB already exists, and it should not be placed repeatedly on the channel.

## Problem

Using two methods to assign `minTrans` complicates the process. It is possible to perform the same adjustment twice, resulting in an incorrect value. For instance, if the desired `minTrans` is 1, executing the first method sets it to 1, but then executing the second method increases it to 2, which is not the intended outcome.

## Solution

As proposed in Issue #342, one of the two methods should be removed to maintain clear logic.

The second method (using the JSON file) is removed because the first method not only addresses Merge/ControlMerge/Mux units but also enforces an OEHB or TEHB on channels near an LSQ. This enforcement is unrelated to the internal TEHB, so nothing should be deducted after the MILP solution. 

Additionally, one-input Merge/ControlMerge/Mux units can be optimized out, so there are no internal TEHBs. In the C++ code, we can intelligently avoid such conditions, but the JSON file cannot do this.

## Implementation

Specifically, internal TEHBs in Mux, Merge, and ControlMerge units are removed from their HDL files. In the Buffer Placement MILP part, the functions `addInternalBuffers` and `deductInternalBuffers` along with their associated data structure are removed (this part is already solved in #358). We now rely solely on setting minimum ~~transparent~~ buffer slot numbers on their successive channels to ensure the MILP places the necessary buffers. Previously, this configuration was applied only to MergeOp. Now, the minimum slot requirement is also applied to ~~ControlMergeOp and~~ MuxOp.