EPFL-LAP/dynamatic #354: [Backend][Refactoring] Improve Op operand/result Name Conversion to RTL Port Names

shundroid, 2025-03-19:
This issue mainly highlights a problem—it suggests a solution, but I'm unsure if it's the right approach, especially with the development of the new backend (particularly the netlist generator).

## Introduction of `NamedIOInterface`, `getOperandName`, and `getResultName`

First, I'll explain the concepts related to this issue—`NamedIOInterface`, `getOperandName`, and `getResultName`.

The port names in exported RTL files often differ from the original operand/result names at the Handshake level. For instance, `MuxOp` is defined in Handshake as:

https://github.com/EPFL-LAP/dynamatic/blob/91286b0674347d3bd97912f98abf4f10630fadb6/include/dynamatic/Dialect/Handshake/HandshakeOps.td#L396-L398

However, the corresponding RTL entity looks like this:

```vhdl
entity handshake_mux_0 is
  port (
    clk, rst : in std_logic;
    -- data input channels
    ins       : in  data_array(2 - 1 downto 0)(32 - 1 downto 0);
    ins_valid : in  std_logic_vector(2 - 1 downto 0);
    ins_ready : out std_logic_vector(2 - 1 downto 0);
    -- index input channel
    index       : in  std_logic_vector(1 - 1 downto 0);
    index_valid : in  std_logic;
    index_ready : out std_logic;
    -- output channel
    outs       : out std_logic_vector(32 - 1 downto 0);
    outs_valid : out std_logic;
    outs_ready : in  std_logic
  );
end entity;
```

Every port name differs from the original Handshake operation. Why?

The reason is `NamedIOInterface`. It’s a Handshake-level interface applied to all Handshake operations through the top-level HandshakeOp class. This interface provides `getOperandName` and `getResultName` methods, which assign names based on operand/result indices. For example, `MuxOp` returns `"index"` for `selectOperand`, so the RTL port is named `index`.

https://github.com/EPFL-LAP/dynamatic/blob/91286b0674347d3bd97912f98abf4f10630fadb6/lib/Dialect/Handshake/HandshakeInterfaces.cpp#L119-L122

## Why Are There Two Names Instead of One?

In my understanding, the reason comes down to both necessity and design decisions (though none of them seem particularly strong to me).

- **Necessity**: MLIR doesn’t provide a way to access operand/result names as strings. It manages them purely by index order, and these names are only used during preprocessing to generate utility methods like `getSelectOperand` or `getResult`.
  - But this alone doesn’t justify maintaining two different names—simply converting `selectOperand` to `"selectOperand"` would be enough.
- **Necessity**: Since multiple backend targets are supported, naming conventions vary. For example, VHDL is case-insensitive, so MLIR’s naming conventions shouldn’t be restricted by a specific target.
  - However, the current implementation still provides only one shared name across all targets, which doesn’t make much sense.
- **Design**: Each RTL unit hardcodes port names within its logic, so it should be responsible for (or at least aware of) its own port names.
  - That said, passing all port names from the C++ side and avoiding hardcoding them in RTL is still possible and reasonable.

## Problem

Currently, the implementation of getOperandName and getResultName is scattered throughout the codebase. These methods are sometimes implemented in the Op definition itself:

https://github.com/EPFL-LAP/dynamatic/blob/91286b0674347d3bd97912f98abf4f10630fadb6/include/dynamatic/Dialect/Handshake/HandshakeOps.td#L855-L865

However, other ops define them in separate files:

https://github.com/EPFL-LAP/dynamatic/blob/91286b0674347d3bd97912f98abf4f10630fadb6/lib/Dialect/Handshake/HandshakeInterfaces.cpp#L134-L137

Some ops don’t implement these methods at all, relying on the default implementation provided by the interface, which is unintuitive—operand/result names are specific to each op, so how can a generic default implementation even work...

Additionally, determining an operation's name often requires following a complex chain of 4-5 function calls, making it difficult to track how a name is actually assigned. You can try tracing the naming of `MuxOp` or `ForkOp`—it’s not straightforward.

This fragmentation creates issues when modifying MLIR ops, updating RTL, or introducing new units, as it's unclear which parts of the codebase are affected by a change.

## Expected Solution

I propose explicitly implementing `getOperandName` and `getResultName` in the Op definition (or at least in the Op subclass definition).

As long as we maintain two separate names, we need a clear way to match them. Unless someone comes up with a completely new approach, `NamedIOInterface` will likely be used in the new backend as well. The code should make it obvious where and how this matching happens. All relevant information should be consolidated in one or two places—the Op definition and the RTL unit definition.

An alternative idea (discussed with @murphe67) was to match Op names and RTL names at the netlist generator, primarily based on indices. I'm open to this approach as long as the matching is intuitive and the information is kept in a well-defined location, which is crucial.

murphe67, 2025-03-19:
Tagging everyone, since to me this is a Dynamatic-meeting-level discussion: @AyaElAkhras @Jiahui17 @Carmine50 @lana555 @paolo-ienne 

@shundroid's description is very nice, but is a code-level-description, which is maybe too much detail, so I will describe it at a higher level (@shundroid is absolutely correct that even if we make no theoretical changes, the implementation needs to be redone, but I think we need a separate very-high-level discussion about this)

## Logic Inside An Operation

When you are generating the code for an operation, for example, for an adder we have:

```
join_inputs : entity work.join(arch) generic map(2)
    port map(
      -- inputs
      ins_valid(0) => lhs_valid,
      ins_valid(1) => rhs_valid,
      outs_ready   => result_ready,
      -- outputs
      outs_valid   => result_valid,
      ins_ready(0) => lhs_ready,
      ins_ready(1) => rhs_ready
  );

result <= std_logic_vector(unsigned(lhs) + unsigned(rhs));
```

we need to know the names of the top level signals, which are defined in the operation's "interface" to the rest of the circuit, like so for an adder:

```
port (
  -- inputs
  clk          : in std_logic;
  rst          : in std_logic;
  lhs          : in std_logic_vector(DATA_TYPE - 1 downto 0);
  lhs_valid    : in std_logic;
  rhs          : in std_logic_vector(DATA_TYPE - 1 downto 0);
  rhs_valid    : in std_logic;
  result_ready : in std_logic;
  -- outputs
  result       : out std_logic_vector(DATA_TYPE - 1 downto 0);
  result_valid : out std_logic;
  lhs_ready    : out std_logic;
  rhs_ready    : out std_logic
);
```

**So inside the operation needs to know exactly what the interface looks like.**

## Building the netlist

In order to convert from a MLIR to an RTL netlist, we must turn each operand into a wire, and each operation into a component.

Naming the components is easy: we simply use the operation type, with a numbered suffix for name sanitation, e.g. addi_0, addi_1, etc. (the details of this may be different in Dynamatic, but the philosophy is the same).

**Naming the wires is extremely difficult.**

A value in MLIR is inherently nameless. Each value is produced as a result, and used as an operand. **Neither result nor operand has a name either.** 

MLIR is inherently an index-based connection system, and any "name" used in the handshake-level code is meta-processing: the name is converted to an index for actual execution.  

### Naming connections

As @shundroid has described above, Dynamatic uses an extremely complex and opaque system to convert from MLIR indices to RTL wire names. 

It is inherently positional, and makes and unmakes naming decisions several times: 

This obscures even the philosophical foundation of how we generate wire names. I think there is a pervasive assumption that the names in the tablegen declaration (which for an adder looks like this:

```
let arguments = (ins ChannelType:$lhs, ChannelType:$rhs);
let results = (outs ChannelType:$result);
``` 
[... 60 lines cut ...]

The MLIR type system perfectly describes what the interface of the operation should look like. But it doesn't have a clean naming system, and so we would have to convert from the type system representation to names using hard-coded indices.

The inside of the operation would have to additionally hard-code those names.

## How variable is the interface of an operation?

It is almost completely fixed: a single list of extra signals 100% defines the interface of an operation (for every single port). 
For units which add and remove extra signals, they need a single list of __other__ extra signals, and then the name of their specific extra signal.

If we inform the unit generators of this single list of signals, the interface is entirely deterministic (but different between different operations). So it is not unrealistic for units to still generate their own interfaces, since they are reliant on a correct interface for their logic to work.

## What if the netlist doesn't know the inner signal names?

We could make the netlist generators be fully positional, to match how MLIR works. The netlist could ask the unit generator how to name the wires of its outputs.

But this probably doesn't work for SMV, where the netlist directly uses the inner variables of the operation.

## So is there anything we can do?

I think we should formalize 3 or 4 ways of handling this from start to finish, and directly compare them. 

It is unavoidable that both the netlist and the unit generators need to share variable names, and that MLIR is not equipped to provide them in an elegant way.

Once we have a few nice proposals, we should discuss in a developers meeting, and knowingly choose the downsides of one of the approaches.

murphe67, 2025-03-19:
Concrete proposal 2, spiritually inspired by @Carmine50 and @gioelegott:

The tablegen arguments declaration:
```
let arguments = (ins ChannelType:$lhs, ChannelType:$rhs);
let results = (outs ChannelType:$result);
``` 
is an order-based definition of the operations ports.
MLIR generates helper code from it, so that any change to this order is propagated automatically.

----

The tablegen of an operation should also (seperately) define a direct mapping from index order to port names, using a similiar mechanism as (if not exactly the same as)  getOperandName and getResultName.

The order in this mapping must match the order of the arguments declaration, so they should be directly next to each other.

-----

The type verification system separately enforces strict rules about the type system. 

-----

The netlist generator receives a **non-serialized MLIR operation**. That is: the netlist generator must be written in C++ and rely on the MLIR framework.

When converting from index-based connecting to name-based connecting, it calls the operation-specific functions mapping from order to port name.

----

The netlist generator ALSO prints the entity declaration of each operation, leaving a section blank for internal logic, and leaving a section blank for dependencies.

Since it has access to the non-serialized MLIR op, it can use a generic printer to convert from the type system description to a module entity.

An example for an adder would be:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

REGEX_MATCH_INSERT_DEPENDENCY

entity addi is
  port (
    -- inputs
    clk          : in std_logic;
    rst          : in std_logic;
    lhs          : in std_logic_vector(32 downto 0);
    lhs_valid    : in std_logic;
    rhs          : in std_logic_vector(32 downto 0);
    rhs_valid    : in std_logic;
    result_ready : in std_logic;
    -- outputs
    result       : out std_logic_vector(32 downto 0);
    result_valid : out std_logic;
    lhs_ready    : out std_logic;
    rhs_ready    : out std_logic
  );
end entity;

architecture arch of addi is

REGEX_MATCH_INSERT_ARCH

end architecture;
```

A unit generator is then called on this file to replace the regex matches with the real generated code.

It uses hardcoded port names to generate its internal logic, which must match with the names in the tablegen file.

It does not receive the MLIR operation (serialized or not), but only the minimum information about extra signals required to generate their forwarding logic.

----

If an operation has a dependency which is also a top-level operation (for example, a control merge is dependant on a fork), these are now two fully separate types of components: 1) a top level fork and 2) a dependency fork.

Inside of the control merge, the unit generator needs to print the interface of the dependency fork, since there is no corresponding MLIR. This inner dependency fork would have no support for extra signals.

This dependency fork would be generated in an entirely different way to a top-level fork, as the starting point is simply completely different.

Unit generators would not support any kind of recursion or encapsulation, and inner components are completely ideologically different to top level operations.

-----

What has to match?

1. The two orders present in the tablegen declaration must match. 
2. What is enforced by the type verification system must match the inner logic that the unit generator deterministically builds.
3. The set of port names present in the unit generator must match the set of names in the tablegen declaration. This is very transparent: the two (order-unimportant) set of names must match, there is no complicated logic. 

-----
What do we remove?

MLIR currently has 3 (4?) different systems for generating a set of ordered names. We should reduce this to 2
1. The tablegen declaration, since the helper functions it generates are obviously amazing. These names are irrelevant for generating RTL and do not have to match anything
2. An additional ordered list of string-valued names, declared right underneath the previous order.

------
What is done blindly?

The netlist generator has **no idea what the names inside the unit generator** are, even though it prints the interface. If there is a mis-match here, synthesis will fail.

-----

What are the disadvantages?

An example of the nice encapsulation we currently support is for the signal wrapper:

[image]

Since this system would not support instantiating a muli as a dependency, since we have no corresponding MLIR to generate its entity from, this would no longer be allowed.

Instead, we would have to use more complicated generation to have a single-level description of the multiplier + signal forwarding, which results in more difficult to read RTL.

murphe67, 2025-03-19:
We could also go all the way to @Carmine50's proposal, but is fundamentally opposed to @paolo-ienne's comment [here](https://github.com/EPFL-LAP/dynamatic/discussions/348#discussioncomment-12509472)-

> I think a large part of the perception on how lightweight these disadvantages are is based on the assumption that we know a couple of cases of extra bits (speculation and tagging) and we assume that nobody will ever invent something fundamentally different. I find this not to be future proof (who knows what metadata tomorrow someone will want to invent?) and this seems to me terrible for an experimental compiler that we would want as open and flexible as possible.

So as well as the code example I wrote above-
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

REGEX_MATCH_INSERT_DEPENDENCY

entity addi is
  port (
    -- inputs
    clk          : in std_logic;
    rst          : in std_logic;
    lhs          : in std_logic_vector(32 - 1  downto 0);
    lhs_valid    : in std_logic;
    rhs          : in std_logic_vector(32 - 1 downto 0);
    rhs_valid    : in std_logic;
    result_ready : in std_logic;
    -- outputs
    result       : out std_logic_vector(32 -1 downto 0);
    result_valid : out std_logic;
    lhs_ready    : out std_logic;
    rhs_ready    : out std_logic
  );
end entity;

architecture arch of addi is

REGEX_MATCH_INSERT_ARCH

end architecture;
```

If there were extra signals, the C++ MLIR side would also generate **all** of this:

```vhdl
-- Note: Headers and entity are generated by `generate_entity`.
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.types.all;

-- Entity of signal manager
entity handshake_muli_0 is
  port(
    clk : in std_logic;
    rst : in std_logic;
    lhs : in std_logic_vector(32 - 1 downto 0);
    lhs_valid : in std_logic;
    lhs_ready : out std_logic;
    lhs_spec : in std_logic_vector(1 - 1 downto 0);
    rhs : in std_logic_vector(32 - 1 downto 0);
    rhs_valid : in std_logic;
    rhs_ready : out std_logic;
    rhs_spec : in std_logic_vector(1 - 1 downto 0);
    result : out std_logic_vector(32 - 1 downto 0);
    result_valid : out std_logic;
    result_ready : in std_logic;
    result_spec : out std_logic_vector(1 - 1 downto 0)
  );
end entity;

-- Architecture of signal manager (buffered)
architecture arch of handshake_muli_0 is
  signal buff_in, buff_out : std_logic_vector(1 - 1 downto 0);
  signal transfer_in, transfer_out : std_logic;
begin
  -- Transfer signal assignments
  transfer_in <= lhs_valid and lhs_ready;
  transfer_out <= result_valid and result_ready;

  -- Note: `forwarded_extra_signals` is {"spec": "lhs_spec or rhs_spec"}.
  -- Concat/split extra signals for buffer input/output
  buff_in(0 downto 0) <= lhs_spec or rhs_spec;
  result_spec <= buff_out(0 downto 0);

  inner : entity work.handshake_muli_0_inner(arch)
    port map(
      clk => clk,
      rst => rst,
      -- Note: these port forwardings are generated by `generate_inner_port_forwarding`.
      lhs => lhs,
      lhs_valid => lhs_valid,
      lhs_ready => lhs_ready,
      rhs => rhs,
      rhs_valid => rhs_valid,
      rhs_ready => rhs_ready,
      result => result,
      result_valid => result_valid,
      result_ready => result_ready
    );

  -- Generate ofifo to store extra signals
  -- num_slots = 4, bitwidth = 1
  buff : entity work.handshake_muli_0_buff(arch)
    port map(
      clk => clk,
      rst => rst,
      ins => buff_in,
      ins_valid => transfer_in,
      ins_ready => open,
      outs => buff_out,
      outs_valid => open,
      outs_ready => transfer_out
    );
end architecture;
```

And so unit generators would only ever be called on extra-signal-less operations. 
(We could also just return to the generic model and keep the library of RTL files)

The generated RTL looks exactly the same (once we pay the cost of reimplementing the signal forwarding wrapper generation in C++). However, we lose the conceptual system it was built on:

(To quote [my own reply](https://github.com/EPFL-LAP/dynamatic/discussions/348#discussioncomment-12509634)):

> The goal of the RTL generation process is to implement something simple, comprehendible, and extensible. We pre-implement common use-cases, to reduce code duplication, but there are no rules for how a component should be generated. You could pass an implementation parameter called "tags-paolo" to the backend (per operation) and selectively handle meta-data completely differently on two different add operations.

This would no longer be true: There would be a single, canonical way of forwarding extra signals. 

We use hierarchy for it, but we do not enable hierarchy as a general concept for building operations.

It is by fair the cleanest solution, but also the rigidest.

(It also has quite a large technical challenge, of how to manage dependencies, since our C++ dependency manager is bad. You would probably need to completely rebuild how the C++ dependency management system works to be able to do this).

shundroid, 2025-03-19:
Quick reply: I don't want the C++ side to generate entity and signal manager. 

> so unit generators would only ever be called on extra-signal-less operations.

This is not possible in every case: speculative units and tagger need to handle extra signals.

My update is that I no longer think it's responsibility for each unit generator to verify the arguments. I now think unit generator can trust MLIR type verification. (after talking with @gioelegott)

Rather than generating the entity by C++ side, I'd prefer to call a general helper function `generate_entity` in the unit generator, just as I did in [signal manager generation](https://github.com/EPFL-LAP/dynamatic/blob/f0b00fbb3d3656e65dd4021799334e931bf82760/experimental/tools/unit-generators/vhdl/generators/support/signal_manager.py#L228).

And pass the full `ports` arguments to each unit generator, as @gioelegott does for testbench generator:

https://github.com/EPFL-LAP/dynamatic/blob/a74aea7ff46b3fb58caa27528ca12117edd8bc47/experimental/test/tools/unit-generators/smv/test_yosys.sh#L56

Since it trusts MLIR type verification, each unit generator can pass the ports argument blindly to the `generate_entity` utility function.

So my current view of `addi` (without extra signals) is-

```python
def _generate_addi(name, params):
  ports = params["ports"]

  # Blindly generate entity from argument (the responsibility is on MLIR type verification side)
  entity = generate_entity(name, ports)

  join_name = f"{name}_join"
  join = generate_join(join_name, {"size": 2})

  architecture = f"""
-- Architecture of addi
architecture arch of {name} is
begin
  join_inputs : entity work.{join_name}(arch)
    port map(
      -- inputs
      ins_valid(0) => lhs_valid,
      ins_valid(1) => rhs_valid,
      outs_ready   => result_ready,
      -- outputs
      outs_valid   => result_valid,
      ins_ready(0) => lhs_ready,
      ins_ready(1) => lhs_ready
    );

  result <= std_logic_vector(unsigned(lhs) + unsigned(rhs));
end architecture;
"""

  return join + entity + architecture
```

Since it trusts the ports arguments, it just hardcodes `lhs`, `rhs`, and `result`.

gioelegott, 2025-03-20:
Thank you for the proposals @murphe67.

So if I understand them correctly:
- the first proposal is our current implementation but with a better naming system that is consistent from tablegen down to RTL
- the second proposal restructures the naming system in the same way. In addition now the unit-generation is split between the MLIR and python side. This system makes it hard to create wrappers on the python side, because from python we have no idea how to generate dependencies, if the entity section is generated in C++.
- the third proposal from @shundroid would be to keep everything on the python side, but now passing all the port information (which in my opinion is equivalent to passing the entire MLIR Op to python). This I think has the same issue as the second proposal, i.e. it makes it difficult to manage dependencies.

(Please correct me if I missed something important in these brief summaries so that I can edit the comment and reduce the noise in the issue)

The only difference I concretely see among these proposals is the amount of information that goes from MLIR to the unit generation. This I think creates a tradeoff we can't really escape: the more information you provide to a unit generator the better the generator works (less hardcoding), but the harder it becomes to generate units as dependencies.

So if I can add a proposal of my own:

The naming stays consistent throughout the generation flow starting from tablegen as suggested before

---

The netlist generator receives a non-serialized MLIR operation. That is: the netlist generator must be written in C++ and rely on the MLIR framework. (Same as we currently have)

---

The unit-generator in python receives an **MLIR Op** and generates the unit for that Op. All the information that we will ever need is there, so I don't expect any changes to the interface of the generator in the future.

How do we handle **dependencies**? We create internal functions that generate units from a subset of the information contained in a MLIR Op.

Example:
Mux as a Op will receive this:
```
%mux1.outs = hw.instance "mux1" @handshake_mux_1(index: %fork3.outs_1: !handshake.channel<i1>, ins_0: %extsi6.outs: !handshake.channel<i32>, ins_1: %cond_br2.trueOut: !handshake.channel<i32>, clk: %clk: i1, rst: %rst: i1) -> (outs: !handshake.channel<i32>)
```

Mux as a dependency will likely receive:
```
INDEX_BITWIDTH = 1, DATA_BITWIDTH = 32, SIZE = 2
```

Now it will be responsibility of the developer on trying to reuse the code when it can be done, so that we don't have too many implementations for the same components. For example we can have some default names and values, so that if they are not provided they don't need to be hardcoded. Note that we are quite free with the default names as they will always be used only internally and never by the netlist generator. Anyways I suggest using the same names for the sake of being consistent.

murphe67, 2025-03-21:
Hey @gioelegott !

I was hoping you would write up a proposal, so thank you very much!

> receives an MLIR Op

A handshake op or a HW op? What happens if we change how the op is serialized (@shundroid change them a lot when he built the extra signals system)? What happens if someone changes the order of the operands in the tablegen file?

A serialized MLIR op is both 1) a complex, fragile, string that encodes a huge amount of information and 2) changes relatively easily, especially if some front-end developer changes things without any knowledge of the backend.

I kind of actually hate the serialized string in general? I think it distracts people from the fact that MLIR actually stores the IR as a complex C++ data structure, and the serialized IR serves a very distinct low-level engineering purpose.

It also provides no abstraction between different generators. If the serialized IR changes, we might have to change every single generation function in every language we support. (like if we get rid of HW, for example, or later on, if we add a signal-level representation and the serialized IR that the backend receives is completely different)

> We create internal functions that generate units from a subset of the information contained in a MLIR Op.

So we will have complex string editing functions to convey 2 parameters? Why not just have the C++ pass us the 2 parameters.... 

> Now it will be responsibility of the developer on trying to reuse the code when it can be done, so that we don't have too many implementations for the same components. 

I'm not a big fan of having 2 different ways to generate the same unit, think it is not that scalable- we don't know what someone will do with the operations next.

> we are quite free with the default names as they will always be used only internally and never by the netlist generator.

This I don't understand- passing naming is the most difficult part of all of this. The names the netlist generator uses must match with the names hardcoded inside of the internal logic.

Finally-
> The unit-generator in python receives an MLIR Op and generates the unit for that Op

The unit generator has very strict rules about what interfaces it supports. This to me is a "fake" parameter- there are many values that can be passed which are not actually allowed, and really only 1 option that is accepted.

I really don't like that the generator could be called "wrong"- it should produce a valid unit for any combination of its parameter space, otherwise you have a poorly designed parameter space.

gioelegott, 2025-03-21:
Thank you @murphe67 for the feedback.

> > receives an MLIR Op
> 
> A handshake op or a HW op? What happens if we change how the op is serialized ([@shundroid](https://github.com/shundroid) change them a lot when he built the extra signals system)? What happens if someone changes the order of the operands in the tablegen file?

In this proposal I'm suggesting to pass a HW Op and it is based on the fact that the naming (including order) will be consistent from the tablegen file of Handshake down to RTL (I think we are taking this direction in all proposals).

What I'm proposing is to propagate the Op as an object until the last function that actually implements the RTL. In this way the generator can access whatever information it needs at any time.
Parsing the Op into python may be annoying, but I think this is just a side effect of using python, and not really a conceptual problem. In any case with a quick search on the Internet I found a package that may help us with parsing (https://github.com/spcl/pymlir, but I bet there are more).

This may be a drastic solution, that doesn't really comply with software engineering rules of hiding information that shouldn't be used, but I think it is the most future proof option that we have.

> I'm not a big fan of having 2 different ways to generate the same unit, think it is not that scalable- we don't know what someone will do with the operations next.

I see this as one major issue of this proposal. I though that it can be solved by having default values for parameters that are not found. But maybe having a separate implementation for the dependencies is just easier, even if not scalable.
On the other hand if the dependency "inherits" the same parameters, then the Op object can be propagated. The signal manager or any other kind of wrapper is an example of that (e.g. if you have a muli with spec bit you generate the signal manager and then call the _generate_base_muli with the same object and it will ignore the extra info).

> I really don't like that the generator could be called "wrong"- it should produce a valid unit for any combination of its parameter space, otherwise you have a poorly designed parameter space.

If a HW Op is wrong then in all cases the generated component will be wrong. Maybe with the current generator we will still have a valid unit, but the circuit will be wrong.
I'm not fully aware of the type checks we currently do, but in any case I don't think it should be the responsibility of the unit-generator.

murphe67, 2025-03-21:
> naming (including order) will be consistent from the tablegen file of Handshake down to RTL (I think we are taking this direction in all proposals).

We can't use the names from the MLIR to generate the internal logic, so the potential inconsistency remains. 

> If a HW Op is wrong then in all cases the generated component will be wrong. 

With your proposal, the netlist will look fully correct, and the generated unit will look wrong. 

In my proposal, the netlist will look wrong, and the generated unit will be correct.

If the HW op is declared wrong, conceptually, it is the netlist that should raise the error. I think your proposal shunts the error into quite a random place?

> Parsing the Op into python may be annoying, but I think this is just a side effect of using python, and not really a conceptual problem. In any case with a quick search on the Internet I found a package that may help us with parsing (https://github.com/spcl/pymlir, but I bet there are more)

We use completely custom printers and parsers for the serialized IR? 

And I disagree on the conceptuality: the unit generators should never be passed an interface, because the **interface is not a parameter.** Each unit supports only one interface- they don't care what the MLIR op looks like because they don't react to it.

> On the other hand if the dependency "inherits" the same parameters, then the Op object can be propagated. 

This is the part of the proposal that gives me nightmares the most: encoding a set of super simple parameters using convoluted string functions.

> In this way the generator can access whatever information it needs at any time.

Why not just give the generator the information it actually needs? (Never mind that there will be like 5 generators all "parsing" these awful strings....)

gioelegott, 2025-03-21:
> Why not just give the generator the information it actually needs?

The problem for this is that it seems we always need more and more information. So my proposal was a reaction (or maybe overreaction) to this trend: just pass all the information. I can see that conceptually this is becoming more a unit-translator than a unit-generator. But I'd like to take this possibility into consideration.

murphe67, 2025-03-21:
I do see that trend, but for me evil string parsing will never be the solution ahahahahahaha

If we need to do more analysis of the data structure on the c++ side to pass more info, doing the analysis on the actual data structure for me always wins out