EPFL-LAP/dynamatic #360: [unit-generator] Enforcing Python 3.12

shundroid, 2025-03-25:
Currently, `experimental/tools/unit-generators` relies on Python 3.12-specific features, such as nested f-string ([ref](https://stackoverflow.com/questions/41215365/what-are-use-cases-for-nested-f-strings/76947481#76947481)):

https://github.com/EPFL-LAP/dynamatic/blob/d0886ea6f28dcee63f302d31040e7b8eee14bbab/experimental/tools/unit-generators/smv/generators/handshake/cond_br.py#L15-L31

https://github.com/EPFL-LAP/dynamatic/blob/2b2b628f01478381faeb0da68039d1caab520ee6/experimental/tools/unit-generators/vhdl/generators/support/signal_manager.py#L572

Additionally, we use case statements, which require Python 3.10+:

https://github.com/EPFL-LAP/dynamatic/blob/2b2b628f01478381faeb0da68039d1caab520ee6/experimental/tools/unit-generators/vhdl/vhdl-unit-generator.py#L26-L30

Since Python 3.12 is not yet widespread, we should enforce (or at least explicitly document) this requirement.

### Proposal: Managing Python Version

I suggest using [Poetry](https://github.com/python-poetry/poetry) for dependency and version management. Poetry allows us to enforce Python 3.12 in `pyproject.toml`:

```toml
[tool.poetry.dependencies]
python = "^3.12"
```

With this setup, activating the Poetry environment ensures the correct Python version.

If Poetry is too much overhead, another option is using `pyenv`'s `.python-version` file specifying `3.12`, which also works well for version enforcement.

What do you think? @murphe67 @gioelegott @Jiahui17

murphe67, 2025-03-25:
We had a big discussion about this ages ago with the LSQ generator- it didn't end super productively.

I agree something like this is required, though.

Surely there is some other project out there we can take inspiration from as an example?

shundroid, 2025-03-25:
I think poetry is well-known and I previously encountered some projects using poetry for this usage like:

https://github.com/commaai/openpilot/blob/master/pyproject.toml
https://github.com/numpy/numpy/blob/main/pyproject.toml

Even PyTorch uses poetry:

https://github.com/pytorch/pytorch/blob/main/pyproject.toml

shundroid, 2025-03-27:
I also want to enforce the formatter (autopep8)

I remembered the discussion about the indent level 2 or 4 (official python style guideline (PEP) suggests 4), but this is a separate thing.

schilkp, 2025-03-30:
I have used poetry in the past and liked it.

Lately I have read a lot of good things about using `uv` instead - but have not actually tried it.

Thought I would throw it into the ring. Maybe it is worth a look?

https://github.com/astral-sh/uv

murphe67, 2025-03-31:
You guys definitely know better than I do on this, am curious what @shundroid thinks of uv? This issue also reminds me of #304- I don't really have the project management experience to apply this things confidently.

@ebosnjak, do you have time to look at this and #304? I think the solutions are pretty clear, but the question is who has time to do it?

shundroid, 2025-04-01:
I didn't know about uv, but after looking it up, it seems worth trying. Besides its speed, I like that it replaces pyenv and that it can also manage different Python versions. Poetry can enforce a specific Python version, but as far as I know, installing different versions still requires pyenv. If we can just ask people to install uv, that would be great.

schilkp, 2025-04-02:
> ...  If we can just ask people to install uv, that would be great.

I guess no matter what you pick, a set of people will have it installed but a larger set will need to install it. So I don't see a big difference between poetry and uv there.

murphe67, 2025-04-02:
Sounds we are more agreed on uv then?

schilkp, 2025-04-02:
> Sounds we are more agreed on uv then?

I have not used it and I make no guarantees - but I would try that first. We can always go back to poetry if it does not work out?