EPFL-LAP/dynamatic #361: [Buffers][TimingInfo] Bug during parameter registration

Jiahui17, 2025-03-25:
## Example:

```mlir
module {
  handshake.func @simple(%arg0: !handshake.channel<i10>) -> (!handshake.channel<i10>) {
    %1 = merge %arg0, %loopback : <i10>
    %2 = buffer %1 { hw.parameters = {NUM_SLOTS=1 : ui32, TIMING = #handshake<timing {D: 1, V: 1}>} } : <i10>
    %3 = buffer %2 { hw.parameters = {NUM_SLOTS=1 : ui32, TIMING = #handshake<timing {R: 1}>} } : <i10>
    %4:2 = fork [2] %3 : <i10>
    %5 = source : <>
    %6 = constant %5 {value = 10 : i10 } : <>, <i10>
    %7 = cmpi ult, %4#0, %6 : <i10>
    %loopback, %output = cond_br %7, %4#1 : <i1>, <i10>
    end %output : <i10>
  }
}

```

This line:
```
    %2 = buffer %1 { hw.parameters = {NUM_SLOTS=1 : ui32, TIMING = #handshake<timing {D: 1, V: 1}>} } : <i10>
```
 crashes the [logic for registering a transparent parameter](https://github.com/EPFL-LAP/dynamatic/blob/cb661b8537cc6496c7a72bbb09ee3887c70c6dab/lib/Support/RTL/RTL.cpp#L403) when calling `--lower-handshake-to-hw`.

## Workaround:

The line below works
```
    %2 = buffer %1 { hw.parameters = {NUM_SLOTS=1 : ui32, TIMING = #handshake<timing {D: 1, V: 1, R: 0}>} } : <i10>
```

murphe67, 2025-03-28:
This is the serialized timing info, but what does this look like inside MLIR?

We check against the pre-determined timing objects (from the looks of it, I didn't pay too much attention to the details)-
TimingInfo TimingInfo::oehb() {
  return TimingInfo()
      .setLatency(SignalType::DATA, 1)
      .setLatency(SignalType::VALID, 1)
      .setLatency(SignalType::READY, 0);
}

TimingInfo TimingInfo::tehb() {
  return TimingInfo().setLatency(SignalType::READY, 1);
}

If you are no longer using these objects to set the timing info, maybe they should be removed? 
Or otherwise maybe you need to add a new one?

Jiahui17, 2025-03-28:
>If you are no longer using these objects to set the timing info, maybe they should be removed?
Or otherwise maybe you need to add a new one?

According to the plan, it should be both? The timingInfo will be removed, and we will register the buffer type names instead.

shundroid, 2025-03-30:
I think the type of `std::optional<unsigned>` is a problem? I guess `nullptr` and `0` are confused. Maybe just `unsigned` is suitable?

Anyway if it will be abolished in the near future we can forget about it for now?

https://github.com/EPFL-LAP/dynamatic/blob/bc67f9975ef44839cc25730ea66bff95b1756fa5/include/dynamatic/Dialect/Handshake/HandshakeAttributes.h#L43-L49