EPFL-LAP/dynamatic #365: [integration-test/correlation_float]compile error: failed to legalize operation 'math.sqrt'

lyl-seu, 2025-03-27:
When I run the integration-test on Dynamatic, an error occurs. What's the reason? Is it because the math library here lacks the sqrt function?

dynamatic> set-src integration-test/correlation_float/correlation_float.c
dynamatic> compile
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
/home/lyl/dynamatic/integration-test/correlation_float/out/comp/cf_dyn_transformed.mlir:34:11: error: failed to legalize operation 'math.sqrt'
    %15 = math.sqrt %14 {handshake.name = "sqrt0"} : f64
          ^
/home/lyl/dynamatic/integration-test/correlation_float/out/comp/cf_dyn_transformed.mlir:34:11: note: see current operation: %135 = "math.sqrt"(%134) <{fastmath = #arith.fastmath<none>}> {handshake.bb = 4 : ui32, handshake.name = "sqrt0"} : (f64) -> f64
[FATAL] Failed to compile cf to handshake

murphe67, 2025-03-28:
@killianmccourt

Jiahui17, 2025-03-29:
This is related to #102.

lyl-seu, 2025-03-29:
> This is related to [#102](https://github.com/EPFL-LAP/dynamatic/issues/102).

oh，i see.thank u bro

closed 2025-03-29 by lyl-seu